merbanan/rtl_433 #2219: Schrader Electronics EG53MA4 update data record

jmfloyd, 2022-10-28:
The flag component is only 3 bytes long and the ID is 4 bytes long based on the latest sensors.  This is easy to confirm as recent sensors have the ID printed on them.  A recent sensor has an id = 80FDBEA9.

Everything else works ok.  There is no change to the data packet, just the breakup of it.

jmfloyd, 2022-10-28:
Also ....This does not affect how old sensors work just that the truncated id will not be unique.

gdt, 2023-09-29:
This sounds like a bug report but I don't really understand it.  Can you re-test with current git master and provide a detailed explanation of what happens vs what should happen, with which specific decoder, and point to where the code is wrong?  Better yet, a PR!