merbanan/rtl_433 #2221: Strange non-json blocks from rtl_433 published messages shown by mosquitto_sub

petterreinholdtsen, 2022-10-29:
When telling rtl_433 to publish to MQTT like this

```
rtl_433 -F "mqtt://localhost:1883,user=username,pass=password"
```

And then fetching the resulting publications, get some strange non-JSON output after each JSON block:

```
% mosquitto_sub -u username -P password -t "rtl_433/#"
{"time":"2022-10-29 18:42:29","model":"Fineoffset-WH2","id":37,"temperature_C":17.4,"mic":"CRC"}
2022-10-29 18:42:29
37
17.4
CRC
{"time":"2022-10-29 18:42:32","model":"Fineoffset-WH2","id":173,"temperature_C":11.3,"mic":"CRC"}
2022-10-29 18:42:32
173
11.3
CRC
^C
%
```

Any idea what is going on here?  Is rtl_433 publishing non-JSON strings too?

zuckschwerdt, 2022-10-29:
Add `-v` to the command to see the topics. The wildcard `"rtl_433/#"` subscribes to many topics and most are the single fields, e.g. `sensor/rtl_433/LaCrosse-TX141THBv2/0/temperature` : `19.7`

petterreinholdtsen, 2022-10-29:
[Christian W. Zuckschwerdt]
> Add `-v` to the command to see the topics. The wildcard `"rtl_433/#"`
> subscribes to many topics and most are the single fields,
> e.g. `sensor/rtl_433/LaCrosse-TX141THBv2/0/temperature` : `19.7`

Aha.  Did not realise that.

No idea why the same info is published several times, but ok, now I know
how to filter away all the non-JSON stuff.  Thank you very much.

-- 
Happy hacking
Petter Reinholdtsen

knarrff, 2022-11-11:
I saw the same and then found in the docs:

```
Supported MQTT formats: (default is all)
	  events: posts JSON event data
	  states: posts JSON state data
	  devices: posts device and sensor info in nested topics
```
Since default is "all" and this includes "devices" and those are not json, those might be what you see. If you don't need them, you can specify the mqtt format to not include all. E.g., I specify only 'events', which will not send the devices messages.

gdt, 2023-09-27:
looks fixed.  feel free to open a discussion for questions/support if needed still.