merbanan/rtl_433 #2224: Bresser 5in1 signal outages

lettore, 2022-11-01:
I own a Bresser 5in1 Weather station from many years and I not really using it because I can't receive data continuously.
I know that there are many open and closed issues about this model, but I'm trying to get it working correctly and cannot diagnose where the problem resides.
The distance between the receiver and the weather station is about 3 meters through a wood ceiling, the original receiver station doesn't have any problems receiving the signal, and the signal strength is maximum.
With rtl_433 I always had a strange behaviour, I can receive data for weeks without any holes, then for days or months, I'm unable to get any data.
This is the command I use.
`rtl_433 -f 868.30M -f 433.92M -s 1024k -H 30 -E hop -F mqtt://127.0.0.1:1883****`
The 433mhz frequency works well with an Oregon sensor.
I tried all the settings I found here, I tried many different kind of antennas. What makes me mad is that the original station receive correctly the signal also at really long distance.
It's also weird because I can receive signal from tires pressure sensors of cars passing on a road at 50 meters from my house and not from my sensor at 3 meters.

zuckschwerdt, 2022-11-02:
You would need to save and then check the signal to find out what's wrong. https://triq.org/rtl_433/ANALYZE.html
At 3 meters it could be overloading a good antenna, it could be jammed by another signal (rtl_433 will look at the whole 1024k window at once). Upload a sample that doesn't work as zip here.

lettore, 2022-11-02:
Thank you for the response, here the zip file. I see a signal at 5.2 seconds at 868.3 but it seems not much powerful. It's possible that I have weak reception?
I'm using a normal TV antenna, but I tried every kind of antenna, if the problem is the antenna can you suggest the best one to receive this frequency?
[file_868.30M_1024k.zip](https://github.com/merbanan/rtl_433/files/9919293/file_868.30M_1024k.zip)

franki29, 2022-11-02:
Hi @lettore , this sounds more a problem of your rtl dongle. You wrote you can receive data for weeks, also the one from your Bresser 5in1?

zuckschwerdt, 2022-11-02:
At around -24 dB that signal looks rather weak. I would have expected something close to 0 dB. Is the auto gain on the SDR dongle not working?
You might be able to get some reception using `-Y autolevel -M level -M noise` (the `-M` to watch levels and noise).

lettore, 2022-11-02:
I tried many dongles and many antennas and the behaviour is the same, I always receive well 433.92M data, but Bresser is not reliable. I can receive data for weeks without any single hole (every 12 seconds) then one day without any reasons I can't receive a single message for weeks or months. During this periods I have no problems receiving 433M signals.
I tried various positions of the antenna and I found that it receives better horizontally aligned with the weather station.
[image]

[image]
Now it's working but I think as it already happened in the past one day will stop working again.

[file_868.30M_1024k-2.zip](https://github.com/merbanan/rtl_433/files/9920874/file_868.30M_1024k-2.zip)

Trying with `-Y autolevel -M level -M noise`
[image]
Results it's quite the same for me

[file_868.30M_1024k-2.zip](https://github.com/merbanan/rtl_433/files/9920980/file_868.30M_1024k-2.zip)

zuckschwerdt, 2022-11-02:
> I tried many dongles and many antennas and the behaviour is the same, I always receive well 433.92M data, [...] then one day without any reasons I can't receive a single message for weeks or months. During this periods I have no problems receiving 433M signals.

That sounds a lot like interference. You should look at the band when that happens, perhaps a strong signal is shadowing everything else. If you known where that signal is, then you can tune away from it, e.g. `-f 868.4M` or `-f 868.2M` or maybe narrow the band with `-s 250k`.

gdt, 2023-09-29:
@lettore Can you recheck with current git master, and also respond to @zuckschwerdt 's comments?  This seems like not a code bug but an issue in your environment.   I'll close in 7 days unless it points to code bug, and you are welcome to ask for help in a discussion or the mailinglist.

closed 2023-09-29 by gdt