merbanan/rtl_433 #2225: Flex decoder where short, long, sync reset and tolerance values are not always the same

Andrei-Iosifescu123, 2022-11-01:
I'm trying to make a flex decoder for my wireless remote, but most values change by 2-4, and aren't always the same. The only value that remains the same is match, but I can't make a decoder only with match parameter. Sometimes it's like this: `-X 'n=name,m=OOK_PWM,s=384,l=1184,r=1236,g=0,t=320,y=0'`, and sometimes like this: `-X 'n=name,m=OOK_PWM,s=380,l=1168,r=1252,g=0,t=315,y=0'`. Is there a way I can get around this and only check the match value?

zuckschwerdt, 2022-11-02:
Those two are practically the same, but the reset could be too narrow. Use
`'n=my-remote,m=OOK_PWM,s=400,l=1200,r=1800,match=a345fe'`

closed 2022-11-02 by zuckschwerdt

Andrei-Iosifescu123, 2022-11-02:
Thank you, this saved my time.

a scris: