merbanan/rtl_433 #2228: KTNNKG Button (Akhan 100F14) Duplicate messages suppression?

erew123, 2022-11-03:
I'm using this `https://uk.banggood.com/KTNNKG-3V-433Mhz-Wireless-Remote-Control-1-Button-Round-Remote-Control-Switch-EV1527-Chip-Learning-Type-p-1516981.html?cur_warehouse=CN`

It matched **protocol 48 Akhan 100F14 remote keyless entry**. I'm sending the data to MQTT for processing home automation routines e.g. Lights On or Lights Off. So duplicated MQTT messages are an issue. I just need the 1x MQTT message to arrive as there is only 1 type of code/data this button sends out e.g. data: "1234567", which I can use to identify the button was pressed (aka, cannot differentiate on/off from the button code).

The problem is, that no matter how quickly I tap the button, even after building a new flex decoder, I always get a **minimum** of 2x messages e.g.

[image]

My current flex decoder looks like this:

`decoder name="Akhan-100F14",modulation=OOK_PWM,short=360,long=1075,reset=13108,gap=1305,sync=0,tolerance=80,unique,repeats<=1,`

I have tried different amount of repeats with less and greater values (the **<=** and **>=** values), tried with unique on/off (unique does help suppress SOME of the messages).

But I cannot get it down to just the 1x message. Here's a **rtl_433 -w file_433.92M_250k.cu -T 60** capture:

[image]

I'm pretty new to rtl_433. I have done a lot of reading and tried to make a best effort to make something work. I wouldn't claim I understand everything, by any stretch. I am however, now a little frustrated and hoped someone may be able to either help guide me, or tell me what I'm asking for isn't possible.

Any suggestions? Or what info can I provide to try figure this out? Do I have to try figure out a way to deal with this downstream in my home automation software (which may not be possible)?

Thanks

zuckschwerdt, 2022-11-03:
With luck you might be able to get the detection down to one single event, but that is not a stable result. There might be issues like interference or bad reception and then you'd get a split and again multiple messages.

rtl_433 is actually doing the defined goal here: report the data stream as it is on the air. Requirements how to filter that down differ over quite a range. Someone might want to identify short vs long presses or maybe record start and end of a button press (think e.g. smoke alarm or leak detector).

The best solution is to filter with your own optimal logic, either receive rtl_433 output exclusively, filter, and forward: https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_custom.py and https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_relay.py
or listen to MQTT, filter, and resend on a different topic (as https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py does)

If you want a low-code solution then NodeRed would be a good way to implement such logic.

erew123, 2022-11-03:
Thanks for the reply on that. Ill look into those and see where I get to!

Really appreciated!