merbanan/rtl_433 #2229: No data from ESIC WT450H

petterreinholdtsen, 2022-11-05:
I got a temperature sensor from ESIC Technologies Ltd. type WT450H and Clas Ohlson article numer 36-1797.  The way I read issues https://github.com/merbanan/rtl_433/issues/1038 and https://github.com/merbanan/rtl_433/issues/105  as well as commit https://github.com/merbanan/rtl_433/pull/174 , this device should be decoded by rtl_433.  But sadly, no information show up in the MQTT stream of detected messages in my installation.

I would be happy to help debug this, but am unsure how to continue.

petterreinholdtsen, 2022-11-05:
I notice https://github.com/nlrb/com.weather-sensors/issues/58 talk about different units using different frequencies, where 433.850 MHz is ok but 433.700 and 433.600 is out of range.  Could this be the case here too?

I also came across https://community.zipato.com/topic/esic-wt450h with someone else struggling to get the sensor working.

zuckschwerdt, 2022-11-05:
Yes, the frequencies in devices and RTL-SDR are not accurate, best to pin down the signal first: https://triq.org/rtl_433/ANALYZE.html#verify-a-transmission
If you get a good sample and it still doesn't decode then post as zip here.

gdt, 2023-09-29:
Closing as long inactive.  Please feel free to start an issue if you get some samples, or a discussion  or the mailinglist if you need help.

closed 2023-09-29 by gdt