merbanan/rtl_433 #2233: rtl_433 fails after a while

fmuntean, 2022-11-10:
it works fine for about a day then stops working and I get the following logs:
Nov 10 17:06:00 rpi rtl_433[6256]: rtl_433 version 21.12-160-geee869e7 branch  at 202210252023 inputs file rtl_tcp RTL-SDR
Nov 10 17:06:00 rpi rtl_433[6256]: Use -h for usage help and see https://triq.org/ for documentation.
Nov 10 17:06:00 rpi rtl_433[6256]: New defaults active, use "-Y classic -s 250k" for the old defaults!
Nov 10 17:06:00 rpi rtl_433[6256]: Publishing MQTT data to localhost port 1883
Nov 10 17:06:00 rpi rtl_433[6256]: Publishing events info to MQTT topic "rtl_433".
Nov 10 17:06:00 rpi rtl_433[6256]: Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
Nov 10 17:06:00 rpi rtl_433[6256]: Detached kernel driver
Nov 10 17:06:00 rpi rtl_433[6256]: Found Fitipower FC0012 tuner
Nov 10 17:06:00 rpi rtl_433[6256]: Sample rate set to 1024000 S/s.
Nov 10 17:06:00 rpi rtl_433[6256]: Tuner gain set to Auto.
Nov 10 17:06:01 rpi rtl_433[6256]: Tuned to 433.920MHz.
Nov 10 17:06:01 rpi rtl_433[6256]: Allocating 15 zero-copy buffers
Nov 10 17:06:01 rpi rtl_433[6256]: MQTT Connected...
Nov 10 17:06:01 rpi rtl_433[6256]: baseband_demod_FM: low pass filter for 1024000 Hz at cutoff 102400 Hz, 9.8 us
Nov 10 17:06:01 rpi rtl_433[6256]: MQTT Connection established.
Nov 10 17:06:49 rpi rtl_433[6256]: {"time" : "2022-11-10 17:06:49.295478", "protocol" : 163, "model" : "Acurite-590TX", "id" : 226, "battery_ok" : 0, "channel" : 1, "humidity" : 48, "mic" : "PARITY", "mod" : "ASK", "freq" : 433.991, "rssi" : -0.126, "snr" : 11.553, "noise" : -11.679}
Nov 10 17:08:02 rpi rtl_433[6256]: {"time" : "2022-11-10 17:08:02.296298", "protocol" : 163, "model" : "Acurite-590TX", "id" : 226, "battery_ok" : 0, "channel" : 1, "temperature_C" : 18.200, "mic" : "PARITY", "mod" : "ASK", "freq" : 434.003, "rssi" : -0.162, "snr" : 11.451, "noise" : -11.613}
Nov 10 17:09:15 rpi rtl_433[6256]: {"time" : "2022-11-10 17:09:15.300206", "protocol" : 163, "model" : "Acurite-590TX", "id" : 226, "battery_ok" : 0, "channel" : 1, "humidity" : 48, "mic" : "PARITY", "mod" : "ASK", "freq" : 434.001, "rssi" : -0.138, "snr" : 11.664, "noise" : -11.802}
Nov 10 17:10:28 rpi rtl_433[6256]: {"time" : "2022-11-10 17:10:28.299545", "protocol" : 163, "model" : "Acurite-590TX", "id" : 226, "battery_ok" : 0, "channel" : 1, "temperature_C" : 18.200, "mic" : "PARITY", "mod" : "ASK", "freq" : 434.003, "rssi" : -0.158, "snr" : 11.862, "noise" : -12.020}
Nov 10 17:11:01 rpi rtl_433[6256]: LIBUSB_ERROR_NOT_FOUND: Entity not found!
Nov 10 17:11:01 rpi rtl_433[6256]: Check your RTL-SDR dongle, USB cables, and power supply.
Nov 10 17:11:01 rpi rtl_433[6256]: Tuned to 915.000MHz.
Nov 10 17:11:01 rpi rtl_433[6256]: {"center_frequency" : 915000000, "frequencies" : [433920000, 915000000], "hop_times" : [300]}
Nov 10 17:11:01 rpi rtl_433[6256]: WARNING: async read failed (-5).
Nov 10 17:11:01 rpi rtl_433[6256]: {"time" : "2022-11-10 17:11:00.046981", "enabled" : 191, "since" : "2022-11-10T17:06:00", "frames" : {"count" : 1951, "fsk" : 0, "events" : 4}, "stats" : [{"device" : 163, "name" : "Acurite 590TX Temperature with optional Humidity", "events" : 1946, "ok" : 4, "messages" : 4, "abort_early" : 1942}]}
Nov 10 17:11:01 rpi rtl_433[6256]: Library error -5, exiting...
Nov 10 17:11:01 rpi rtl_433[6256]: Reattached kernel driver
Nov 10 17:11:01 rpi rtl_433[6383]: rtl_433 version 21.12-160-geee869e7 branch  at 202210252023 inputs file rtl_tcp RTL-SDR

if I stop the rtl_433 and start the rtl_tcp then connect from external laptop to it works with no issues.

The only way to make it work again for 1 day is to restart the entire board.

Please advise.

knarrff, 2022-11-11:
Hi. Could you please go over #854 and see if, like there, this is a USB problem of your system? Maybe you see also USB-related entries in the system logs, maybe you also have other devices on the same bus and could try without those attached to see if that helps.

fmuntean, 2022-12-18:
did not seem to be a USB problem. I recompile it and now is has been working fine for few week now.

closed 2022-12-18 by fmuntean