merbanan/rtl_433 #2237: TFA 30.3221.02 temperature weather sensor recognized as Oil tank

samuelhermann, 2022-11-16:
Hello,

Without device specification TFA 30.3221.02 is recognized as a [Oil tank sensor](https://community.home-assistant.io/t/oil-tank-level-monitor-watchman-sonic-rtl-433-integration-for-rf-sensors-or-level-sensors/223986)

time      : 2022-11-16 21:15:41
model     : Oil-SonicStd id        : ffff
flags     : f5           alarm     : 1             binding_countdown: 254    depth_cm  : 0

I tried -R option with device protocol id 73 but it is not working.
https://github.com/merbanan/rtl_433/blob/master/src/devices/tfa_30_3221.c

Whereas with the flipper zero it is fine. Could it be my hackRF that capture too much noise ?

Thanks

samuelhermann, 2022-11-16:
Here is the command I have used. This should indeed use the protocol even without specifying it with -R 73 option.

rtl_433 -d driver=hackrf -g LNA=40,AMP=14,VGA=30
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
[INFO] Opening HackRF One #0 81c69dc325f4f1b...
Sample rate set to 250000 S/s.
Setting gain element LNA: 40.000000 dB
Setting gain element AMP: 14.000000 dB
Setting gain element VGA: 30.000000 dB
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-11-16 21:30:42
model     : Oil-SonicStd id        : ffff
flags     : f5           alarm     : 1             binding_countdown: 254    depth_cm  : 0

closed 2022-11-16 by samuelhermann

zuckschwerdt, 2022-11-17:
Interessting, so a TFA 30.3221.02 is misinterpreted as Oil-SonicStd when usig 250k sample rate?
Can you grab a sample at 250k that shows this behaviour?
https://triq.org/rtl_433/ANALYZE.html