merbanan/rtl_433 #2239: FineOffset WH57 not decoding

craiggenner, 2022-11-17:
I'm having some difficulties getting data from a FineOffset WH57 (actually a ecowitt WH57). The signal is being received, but it's not being decoded for some reason.

This is what I have so far:
```
$ ./rtl_433 -d :00000301 -f 868M -c /dev/null -F json -R 0 -X 'n=WH57,m=FSK_PCM,s=60,l=60,r=1800,preamble=AA2DD4'
rtl_433 version 21.12-167-g1aa49c04 branch master at 202211141746 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.

New defaults active, use "-Y classic -s 250k" for the old defaults!

Disabling all device decoders.
Registered 1 out of 223 device decoding protocols [ ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.000MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
{"time" : "2022-11-17 21:14:01", "model" : "WH57", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 80, "data" : "5700760a01cfc0143a52"}], "codes" : ["{80}5700760a01cfc0143a52"]}
{"time" : "2022-11-17 21:16:39", "model" : "WH57", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 78, "data" : "5700ec14033f0143a604"}], "codes" : ["{78}5700e
[SignalDump.zip](https://github.com/merbanan/rtl_433/files/10036113/SignalDump.zip)
c14033f0143a604"]}
{"time" : "2022-11-17 21:19:17", "model" : "WH57", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 79, "data" : "5701d814019f802874c2"}], "codes" : ["{79}5701d814019f802874c2"]}
{"time" : "2022-11-17 21:20:36", "model" : "WH57", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 79, "data" : "5700ec14033f0050e944"}], "codes" : ["{79}5700ec14033f0050e944"]}
^CSignal caught, exiting!
```

The WH57 is covered by the 190 device protocol.

I have the same hardware receiving data from another weather station (Bresser Weather Center 7-in-1 , protocol 172), and this is working perfectly well and has done for some months.

I've tried to follow the guidance at https://triq.org/rtl_433/ANALYZE.html#grab-a-sample, but I'll admit I'm struggling. I ran a `./rtl_433 -d :00000301 -f 868M -c /dev/null -F json -S unknown` and have attached the results in a zip.

zuckschwerdt, 2022-11-18:
The demod gets confused here. You may have more luck if you get closer to the signal, e.g. `-f 868.1M` or center the signal, e.g. `-f  868.25M`
The `classic` demod does work: `rtl_433 -f 868M -Y classic -c 0 -R 0 -A g006_868M_1000k.cu8`

But note, that the PCM there (if the sample rate was correct) has 120 µs bit-width, see
https://triq.org/pdv/#AAB02708010168007900F301F80316057800A1233486919191919191919191919191919191919191919055+AAB01708010168007900F301F80316057800A1233491A181919055+AAB01308010168007900F301F80316057800A123349355+AAB01608010168007900F301F80316057800A12334A1B1929455+AAB01408010168007900F301F80316057800A12334919555+AAB01408010168007900F301F80316057800A1233482B055+AAB01308010168007900F301F80316057800A12334A055+AAB01608010168007900F301F80316057800A12334C291B1C355+AAB01708010168007900F301F80316057800A12334919292919455+AAB01508010168007900F301F80316057800A1233492919355+AAB01708010168007900F301F80316057800A12334919291A1D355+AAB01408010168007900F301F80316057800A12334A19055+AAB01308010168007900F301F80316057800A123349755

and use `-X 'n=WH57,m=FSK_PCM,s=120,l=120,r=1800,preamble=AA2DD4'`

zuckschwerdt, 2022-11-18:
@merbanan might have some interest in this sample as it breaks the minmax demod.

merbanan, 2022-11-18:
I would need to look at the signals components and there is no debug mode for that out of the box IIRC but most likely the heuristics go bad. And changing the heuristics has the risk of affecting other decoders.

Ideally the demod framework should be reworked so that both modes can be active at the same time. But I will not have the time to do that in the near future.

craiggenner, 2022-11-18:
> The demod gets confused here. You may have more luck if you get closer to the signal, e.g. `-f 868.1M` or center the signal, e.g. `-f 868.25M` The `classic` demod does work: `rtl_433 -f 868M -Y [...]

For other having this issue, running with `-f 868.25M` worked.

Thanks for you time and suggestion @zuckschwerdt

zuckschwerdt, 2022-11-18:
> Thanks for you time and suggestion @zuckschwerdt

Back to you, thanks for an interesting test case. Best I can see it's the ripple on the signal combined with the freq offset that might get the demod (level discriminator really) stuck. Perhaps we can someday improve that.