merbanan/rtl_433 #2242: Add decoders for AC114-xxB and BeSmart_Sx

TheChatty, 2022-11-21:
SignalDuino already has support for [AC114-xxB](https://github.com/RFD-FHEM/RFFHEM/blob/master/FHEM/lib/SD_ProtocolData.pm#L1656) and [BeSmart_Sx](https://github.com/RFD-FHEM/RFFHEM/blob/master/FHEM/lib/SD_ProtocolData.pm#L2177).

Using rtl_433_ESP both devices are reported as "[unknown protocol](https://github.com/NorthernMan54/rtl_433_ESP/issues/45)".

Raw message from AC114-xxB:
```
rtl_433_ESP(6): Signal length: 417996, Gap length: 21995386, Signal RSSI: -35, train: 0, messageCount: 6, pulses: 394
rtl_433_ESP(6): RAW (417996): +589-197(-38)+189-610(-36)+594-197(-36)+193-591(-36)+199-596(-36)+198-595(-37)+598-196(-36)+588-198(-36)+196-595(-36)+600-196(-36)+194-596(-36)+200-588(-36)+199-595(-36)+574-220(-36)+590-197(-36)+589-198(-36)+595-198(-36)+197-595(-36)+200-595(-36)+596-198(-37)+192-590(-37)+575-221(-37)+591-197(-37)+193-596(-37)+598-198(-37)+195-594(-36)+199-597(-37)+594-197(-37)+193-595(-37)+571-221(-36)+591-197(-36)+193-595(-37)+199-595(-37)+199-595(-37)+199-595(-36)+198-591(-37)+198-591(-37)+198-594(-37)+199-595(-37)+595-196(-36)+195-591(-37)+200-596(-37)+198-595(-37)+197-593(-37)+198-591(-37)+199-594(-37)+198-594(-38)+197-592(-38)+199-593(-38)+199-594(-37)+573-221(-37)+191-593(-37)+199-594(-37)+198-594(-37)+573-221(-37)+589-197(-37)+595-198(-37)+196-595(-37)+198-596(-37)+573-221(-37)+192-590(-37)+602-196(-37)+567-222(-37)+591-197(-37)+590-247(-37)+5166-601(-37)+599-197(-37)+191-594(-37)+598-197(-37)+195-592(-37)+199-595(-37)+197-593(-37)+597-197(-38)+590-196(-37)+194-598(-37)+595-198(-37)+194-593(-37)+198-596(-37)+199-595(-37)+594-196(-37)+592-196(-37)+594-198(-37)+592-197(-37)+195-597(-37)+198-594(-37)+574-219(-37)+193-597(-37)+595-198(-37)+592-198(-37)+194-593(-38)+600-196(-38)+192-594(-38)+199-593(-37)+597-198(-37)+195-593(-37)+595-197(-37)+590-198(-37)+195-594(-37)+200-598(-37)+196-594(-37)+198-594(-37)+196-594(-37)+198-595(-37)+197-592(-37)+198-591(-37)+597-197(-37)+196-596(-37)+200-595(-37)+197-589(-37)+199-594(-37)+198-596(-37)+198-594(-37)+200-586(-37)+198-593(-37)+200-595(-37)+199-596(-38)+593-197(-37)+192-591(-37)+199-597(-37)+199-594(-37)+594-196(-38)+590-198(-37)+594-197(-37)+195-597(-37)+197-598(-37)+593-197(-37)+196-590(-37)+598-196(-37)+590-198(-37)+593-200(-37)+592-244(-37)+5177-597(-37)+570-222(-37)+191-588(-37)+598-197(-37)+191-593(-37)+199-593(-37)+199-593(-37)+574-221(-37)+588-197(-37)+196-595(-37)+600-198(-37)+195-593(-37)+197-595(-37)+197-590(-36)+599-198(-36)+592-196(-37)+589-197(-37)+594-198(-37)+197-596(-37)+199-594(-37)+599-171(-37)+216-589(-37)+599-199(-37)+592-196(-37)+196-594(-37)+597-197(-37)+195-594(-37)+199-597(-37)+570-221(-37)+192-590(-37)+598-198(-37)+594-196(-37)+194-595(-37)+198-598(-37)+197-593(-37)+198-594(-37)+199-595(-37)+197-592(-37)+199-589(-36)+198-594(-37)+601-194(-37)+191-596(-37)+198-593(-37)+201-592(-37)+198-595(-37)+197-593(-37)+197-592(-37)+197-594(-37)+199-594(-37)+197-595(-37)+198-592(-37)+572-221(-37)+193-594(-37)+198-595(-37)+198-591(-38)+572-221(-38)+591-197(-37)+592-198(-37)+196-594(-37)+199-597(-37)+598-197(-37)+190-593(-37)+597-197(-37)+594-198(-36)+595-170(-36)+612-248(-37)+5173-596(-37)+571-221(-37)+192-594(-37)+597-197(-37)+193-594(-37)+197-597(-37)+199-594(-37)+594-196(-37)+590-197(-37)+196-596(-37)+598-196(-37)+192-593(-37)+198-593(-37)+199-594(-37)+599-173(-37)+610-197(-36)+593-197(-36)+598-194(-36)+195-597(-36)+198-596(-36)+595-197(-36)+193-594(-36)+599-172(-36)+611-198(-36)+194-593(-36)+601-197(-36)+194-595(-36)+198-594(-36)+596-199(-37)+192-592(-36)+600-197(-37)+587-198(-37)+195-593(-37)+199-598(-38)+199-595(-36)+196-593(-36)+198-591(-36)+198-594(-36)+198-596(-37)+197-593(-36)+595-197(-36)+193-593(-36)+200-597(-36)+198-592(-36)+199-589(-36)+199-593(-36)+199-594(-36)+196-593(-36)+197-591(-36)+200-596(-36)+198-594(-36)+598-172(-36)+215-590(-36)+199-594(-36)+199-595(-36)+599-172(-36)+611-197(-37)+595-199(-37)+197-593(-36)+199-597(-36)+594-196(-36)+193-590(-36)+599-195(-36)+592-197(-36)+591-199(-36)+594-15782(-43)+5155-669(-30)+519-270(-30)+119-686(-31)+520-247(-31)+144-639(-31)+148-640(-31)+149-665(-32)+521-247(-32)+566-222(-33)+169-640(-32)+547-223(-33)+171-643(-34)+150-620(-34)+173-614(-33)+571-222(-34)+567-222(-35)+566-222(-35)+591-198(-35)+195-594(-35)+200-594(-36)+574-220(-36)+192-589(-36)+598-199(-36)+592-196(-36)+196-594(-36)+597-198(-36)+194-593(-36)+199-596(-37)+597-196(-36)+192-592(-36)+597-198(-36)+592-197(-36)+194-594(-36)+199-598(-36)+198-591(-36)+199-594(-36)+198-594(-36)+197-593(-37)+200-588(-36)+199-593(-36)+599-194(-36)+192-596(-36)+198-593(-36)+199-594(-36)+199-595(-36)+197-592(-36)+197-592(-36)+198-593(-36)+201-592(-36)+197-595(-36)+197-591(-36)+597-198(-36)+195-593(-36)+197-594(-36)+198-591(-36)+597-198(-36)+593-196(-36)+589-198(-36)+196-594(-36)+199-595(-36)+599-173(-36)+215-593(-36)+596-198(-36)+592-198(-36)+594-197(-36)+591-222(-36)+5193-596(-36)+572-222(-36)+193-593(-36)+597-195(-36)+193-593(-35)+198-595(-35)+200-593(-36)+594-196(-36)+591-198(-36)+196-595(-36)+599-195(-35)+192-594(-35)+198-594(-35)+198-595(-35)+600-172(-36)+609-198(-35)+592-198(-35)+596-197(-36)+194-598(-36)+197-596(-36)+595-197(-36)+193-593(-36)+601-171(-36)+611-199(-35)+193-592(-35)+600-196(-35)+195-597(-35)+198-595(-36)+596-198(-36)+192-591(-35)+599-196(-35)+587-198(-35)+195-593(-35)+200-598(-36)+199-595(-35)+197-593(-35)+197-591(-35)+199-593(-35)+199-594(-36)+197-592(-35)+596-197(-35)+194-594(-35)+199-600(-35)+197-594(-35)+199-586(-36)+198-593(-36)+199-594(-36)+199-597(-35)+195-588(-35)+201-593(-35)+199-594(-35)+599-171(-35)+215-590(-35)+200-595(-35)+198-595(-35)+598-172(-35)+611-197(-35)+595-198(-35)+197-595(-35)+199-595(-35)+594-196(-35)+193-594(-35)+597-196(-35)+591-196(-35)+591-198(-35)
```

Raw message from BeSmart_Sx:
```
rtl_433_ESP(6): Signal length: 186996, Gap length: 37097380, Signal RSSI: -18, train: 0, messageCount: 3, pulses: 98
rtl_433_ESP(6): RAW (186996): +494-246(-18)+214-529(-18)+512-245(-18)+216-530(-18)+221-523(-18)+488-267(-18)+488-242(-18)+223-516(-18)+513-249(-18)+212-515(-18)+229-527(-18)+509-246(-18)+210-526(-18)+217-533(-18)+510-241(-18)+217-526(-18)+217-528(-18)+220-533(-18)+209-4104(-20)+242-502(-21)+511-245(-18)+216-532(-18)+498-244(-18)+242-503(-18)+242-498(-18)+513-245(-18)+507-219(-18)+232-525(-18)+513-245(-18)+210-526(-18)+244-506(-18)+512-239(-18)+218-525(-18)+218-528(-18)+515-247(-18)+215-508(-18)+236-516(-18)+235-498(-18)+243-4107(-20)+219-533(-18)+508-244(-18)+218-502(-18)+518-246(-18)+215-531(-18)+212-524(-18)+512-245(-18)+508-219(-18)+229-523(-18)+512-245(-18)+216-514(-18)+227-531(-18)+514-244(-18)+209-527(-18)+218-534(-19)+503-237(-18)+212-535(-18)+236-500(-18)+243-503(-18)+223-4114(-19)+216-523(-21)+505-257(-18)+218-528(-18)+488-271(-18)+215-506(-18)+225-531(-18)+487-268(-18)+488-242(-18)+224-519(-18)+513-245(-18)+216-530(-18)+197-546(-18)+489-268(-18)+218-502(-18)+221-540(-18)+507-245(-18)+217-504(-18)+223-541(-18)+217-524(-18)+216-4126(-20)+210-526(-18)+516-219(-18)+244-513(-18)+513-244(-18)+217-515(-18)+225-533(-18)+489-268(-18)+488-219(-18)+245-517(-18)+512-245(-18)+216-530(-18)+222-522(-18)+489-268(-18)+219-502(-18)+222-541(-18)+489-239(-18)+246-517(-18)+222-522(-18)+236-499(-18)
```

zuckschwerdt, 2022-11-21:
From the RAW pulse data we get for the BeSmart_Sx: 20 bit, PWM short: 230 long: 510 reset: 800, e.g.

https://triq.org/pdv/#AAB10300E801FC1008819081908181909081908181908181908181818255

or `BeSmart_Sx.ook`:
```
242 502
511 245
216 532
498 244
242 503
242 498
513 245
507 219
232 525
513 245
210 526
244 506
512 239
218 525
218 528
515 247
215 508
236 516
235 498
243 4107
```

For the AC114-xxB we get: 88 bit, PWM short: 200 long: 500 reset: 800, e.g.

https://triq.org/pdv/#AAB1041430025000C000F48192A192A1A1A19292A192A1A1A192929292A1A192A19292A192A1A192A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A1A1929292A1A192A1929292938155

or `AC114-xxB.ook`:
```
5166 601
599 197
191 594
598 197
195 592
199 595
197 593
597 197
590 196
194 598
595 198
194 593
198 596
199 595
594 196
592 196
594 198
592 197
195 597
198 594
574 219
193 597
595 198
592 198
194 593
600 196
192 594
199 593
597 198
195 593
595 197
590 198
195 594
200 598
196 594
198 594
196 594
198 595
197 592
198 591
597 197
196 596
200 595
197 589
199 594
198 596
198 594
200 586
198 593
200 595
199 596
593 197
192 591
199 597
199 594
594 196
590 198
594 197
195 597
197 598
593 197
196 590
598 196
590 198
593 200
592 244
5177 597
```

Not sure what the data fields are though? Any info on that?

TheChatty, 2022-11-21:
Top two links describe the SignalDuino RF protocol. I am able to send the follwing commands:
[AC114-xxB](https://github.com/RFD-FHEM/RFFHEM/issues/906#issuecomment-731840351):
```
canvas up   --> set sduino sendMsg u56#0xA347969601000B7F8#R3
canvas stop --> set sduino sendMsg u56#0xA3479696010023978#R3
canvas down --> set sduino sendMsg u56#0xA3479696010043B78#R3
```

[BeSmart_Sx](https://github.com/RFD-FHEM/RFFHEM/issues/909#issuecomment-740160716):
```
fan light  --> set sduino sendMsg P78#0x53490#R4
fan 5min   --> set sduino sendMsg P78#0x53481#R4
fan higher --> set sduino sendMsg P78#0x53488#R4
fan lower  --> set sduino sendMsg P78#0x53484#R4
```

TheChatty, 2022-11-21:
@zuckschwerdt What can I help you with? Do you need multiple samples from the same key presses or samples from each key? Is the code from SignalDuino any help to you?

zuckschwerdt, 2022-11-21:
You can try a flex decoder to make sure we got things right:
`rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=200,l=500,r=800,bits>=20'`
This should output codes for both senders.

If that works out we need to identify more fields, there should at least be an ID field. Can you change a "house code" on these? Or do IDs reset on battery change? What changes in the output if you do?

TheChatty, 2022-11-22:
I don't have a SDR handy atm. I was hoping that the already existing implementations for both protocols could help here.

Neither do I have an Arduino atm. What I meant to offer was further samples of the debug output of rtl_433_ESP like the raw messages in the first post. Can they also be analyzed at triq.org?

TheChatty, 2023-02-23:
I took samples from AC114 [down] and [stop] and from BeSmart [-], [+], [lamp] and [5min]:
[Samples.zip](https://github.com/merbanan/rtl_433/files/10818469/Samples.zip)
Each button was pressed 4 times (and saved to independent files).

TheChatty, 2023-03-02:
@zuckschwerdt Can I help more?

TheChatty, 2023-03-25:
Is it possible to adjust sync length in source code thus non-flex decoders will pick up the signal? I want to use [OMG](https://github.com/1technophile/OpenMQTTGateway/) using [rtl_433_ESP](https://github.com/NorthernMan54/rtl_433_ESP/).

zuckschwerdt, 2023-03-26:
These are very simple signals. You can just write a decoder for them. If you are lazy and want something fast just change one of the existing decoders (e.g. generic_motion.c).

We can't merge these simple "doorbell"-style decoders, but you can write/change your own one for ESP.

TheChatty, 2023-03-26:
```
# rtl_433 -R 0 -X 'n=AC114,m=OOK_PWM,s=200,l=500,y=5250,r=800,bits>=20'
time      : 2023-03-26 13:37:30
model     : AC114         count     : 3             num_rows  : 3             rows      : 
len       : 65           data      : 5cb86969feffdc680, 
len       : 65           data      : 5cb86969feffdc680, 
len       : 65           data      : 5cb86969feffdc680
codes     : {65}5cb86969feffdc680, {65}5cb86969feffdc680, {65}5cb86969feffdc680
```

With adjusted [generic_motion.c]:
```
 # tail -n 11 generic_motion.c
r_device const generic_motion = {
        .name        = "AC114-01B",
        .modulation  = OOK_PULSE_PWM,
        .short_width = 200,
        .long_width  = 500,
        .sync_width  = 5250,
        .gap_limit   = 1200,
        .reset_limit = 800,
        .decode_fn   = &generic_motion_callback,
        .fields      = output_fields,
};
# rtl_433 -R 87
Registered 1 out of 243 device decoding protocols [ 87 ]
... no output when button pushed ...
```

zuckschwerdt, 2023-03-26:
You do need to change the check also:
https://github.com/merbanan/rtl_433/blob/3264d01d65f386529619eecadf1471c4cf42ee5a/src/devices/generic_motion.c#L42-L47

TheChatty, 2023-03-26:
I modified line 43 to check for either 20 (BeSmart) or 65 (AC114) bits.

On my mac with RTL2832U it works.

Ony my ESP with CC1101:
- BeSmart
  - it works with every button push if they are not to fast together (otherwise it gets not detected)
- AC114
  - it works for a few button pushes - mostly signal parsing fails

How to output all bits of bitbuffer? Like `code = ...`?

zuckschwerdt, 2023-03-26:
Change `char code_str[6];` into `char code_str[20];` then change `sprintf` to
```
sprintf(code_str, "%02x%02x%02x%02x%02x%02x%02x%02x", b[0], b[1], b[2], b[3], b[4], b[5], b[6], b[7]);
```

TheChatty, 2023-03-26:
Thank you very much for your strong support. It now works very well with RTL2832U. I need to tackle CC1101 further...

TheChatty, 2024-05-22:
One more question: Is there a simple way to convert above codes to [esphome raw codes](https://github.com/dbuezas/esphome-cc1101/blob/main/cc1101-esp32.yaml#L55)?

Dumping received signals in raw format looks similar to this:
```
AC114 stop press : {65}5cb86969feffdc680

-->

code: [-599, 596, -199, 196, -571, 618, -174, 219, -594, 197, -595, 198, -596, 596, -173, 615, -199, 195, -595, 596, -198, 195, -571, 222, -593, 198, -594, 596, -197, 593, -173, 616, -199, 594, -199, 197, -595, 198, -572, 618, -174, 219, -593, 
             596, -197, 592, -174, 219, -596, 596, -199, 196, -594, 198, -571, 619, -173, 219, -593, 595, -198, 592, -174, 218, -596, 198, -595, 198, -596, 198, -570, 222, -570, 222, -593, 198, -594, 197, -571, 618, -174, 219, -595, 197, -596, 198, -571, 222, -570, 
             222, -592, 197, -595, 197, -571, 221, -595, 198, -594, 197, -595, 596, -173, 218, -570, 222, -594, 197, -595, 595, -173, 615, -174, 618, -199, 196, -596, 197, -596, 597, -173, 218, -593, 595, -199, 593, -197, 593, -173, 616, -223]
```

This is close enough for matching further signals but it looks like it's not close enough if I send this via esphome. It's not picked up by the orginal receiver.

zuckschwerdt, 2024-05-22:
Since this is PWM coded a single 0-bit should be `600,-200` and the 1-bit `200,-600`. The +/- is pulse/gap format. Just out output each bit that way.

zuckschwerdt, 2024-05-23:
Also you can open https://triq.org/pdv/ then click the plus symbol and choose "Raw pulse data" and paste your example above to see the raw esphome pulses:

[image]

TheChatty, 2024-05-23:
I was looking for the opposite. Like having a .cu8 sample and convert it to the above raw pulse format.

zuckschwerdt, 2024-05-23:
After converting (transform all bits to 600,-200 / 200,-600) use the resulting codes in pdv to compare if you got it right :)

TheChatty, 2024-05-23:
I needed to get the bits then converted them with Excel formulas to pulses. For AC114 it is important to send the high preamble (but it was not shown in esphome raw dump... luckily rtl_433 and my SDR did).

For sake of documentation - here are the working codes in esphome raw format:
```
button:
  - platform: template
    name: AC114 stop
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [-1,5200,-600,600,-200,200,-600,600,-200,200,-600,200,-600,200,-600,600,-200,600,-200,200,-600,600,-200,200,-600,200,-600,200,-600,600,-200,600,-200,600,-200,600,-200,200,-600,200,-600,600,-200,200,-600,600,-200,600,-200,200,-600,600,-200,200,-600,200,-600,600,-200,200,-600,600,-200,600,-200,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,600,-200,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,600,-200,200,-600,200,-600,200,-600,600,-200,600,-200,600,-200,200,-600,200,-600,600,-200,200,-600,600,-200,600,-200,600,-200,600,-200]
          repeat:
            times: 4
            wait_time: 0ms
      - lambda: get_cc1101(transciver).endTransmission();
  - platform: template
    name: AC114 up
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [-1,5200,-600,600,-200,200,-600,600,-200,200,-600,200,-600,200,-600,600,-200,600,-200,200,-600,600,-200,200,-600,200,-600,200,-600,600,-200,600,-200,600,-200,600,-200,200,-600,200,-600,600,-200,200,-600,600,-200,600,-200,200,-600,600,-200,200,-600,200,-600,600,-200,200,-600,600,-200,600,-200,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,600,-200,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,600,-200,200,-600,600,-200,600,-200,200,-600,600,-200,600,-200,600,-200,600,-200,600,-200,600,-200,600,-200,600,-200]
          repeat:
            times: 4
            wait_time: 0ms
      - lambda: get_cc1101(transciver).endTransmission();
  - platform: template
    name: AC114 down
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [-1,5200,-600,600,-200,200,-600,600,-200,200,-600,200,-600,200,-600,600,-200,600,-200,200,-600,600,-200,200,-600,200,-600,200,-600,600,-200,600,-200,600,-200,600,-200,200,-600,200,-600,600,-200,200,-600,600,-200,600,-200,200,-600,600,-200,200,-600,200,-600,600,-200,200,-600,600,-200,600,-200,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,600,-200,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,200,-600,600,-200,200,-600,200,-600,200,-600,200,-600,600,-200,600,-200,600,-200,200,-600,600,-200,600,-200,200,-600,600,-200,600,-200,600,-200,600,-200]
          repeat:
            times: 4
            wait_time: 0ms
      - lambda: get_cc1101(transciver).endTransmission();

  - platform: template
    name: BeSmart minus
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [200,-500,500,-200,200,-500,500,-200,200,-500,200,-500,500,-200,500,-200,200,-500,500,-200,200,-500,200,-500,500,-200,200,-500,200,-500,200,-500,200,-500,500,-200,200,-500,200,-500]
          repeat:
            times: 5
            wait_time: 4ms
      - lambda: get_cc1101(transciver).endTransmission();
  - platform: template
    name: BeSmart plus
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [200,-500,500,-200,200,-500,500,-200,200,-500,200,-500,500,-200,500,-200,200,-500,500,-200,200,-500,200,-500,500,-200,200,-500,200,-500,200,-500,500,-200,200,-500,200,-500,200,-500]
          repeat:
            times: 5
            wait_time: 4ms
      - lambda: get_cc1101(transciver).endTransmission();
  - platform: template
    name: BeSmart light
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [200,-500,500,-200,200,-500,500,-200,200,-500,200,-500,500,-200,500,-200,200,-500,500,-200,200,-500,200,-500,500,-200,200,-500,200,-500,500,-200,200,-500,200,-500,200,-500,200,-500]
          repeat:
            times: 5
            wait_time: 4ms
      - lambda: get_cc1101(transciver).endTransmission();
  - platform: template
    name: BeSmart 5min
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [200,-500,500,-200,200,-500,500,-200,200,-500,200,-500,500,-200,500,-200,200,-500,500,-200,200,-500,200,-500,500,-200,200,-500,200,-500,200,-500,200,-500,200,-500,200,-500,500,-200]
          repeat:
            times: 5
            wait_time: 4ms
      - lambda: get_cc1101(transciver).endTransmission();

``` 

@zuckschwerdt Thank you for your support and all the great tools you made.