merbanan/rtl_433 #2247: Support for Costco Christmas Tree RF Remote

kspearrin, 2022-11-27:
Recently purchased a Christmas tree from Costco that have integrated LED lighting. It comes with a little 3 button RF remote that lets you cycle through lighting modes (colored lights, white lights, fading, twinkling, etc) with one button and change brightness of the lights up and down with the other two buttons. I wanted to sniff the signals so that I could create my own remote to interface with Home Assistant and replay the known signals.

Here is the remote: https://fccid.io/2APJZ-CW002

Using GQRX I was able to confirm the remote sends its signals at around 433.9M. I have not been able to get any feedback from rtl_433, however.

Following [the guide here](https://triq.org/rtl_433/ANALYZE.html) I tried to collect some of the data using `-S unknown` while pressing some buttons on the remote. It produces a lot of cu8 files when running this for just ~30 seconds. Here is one of the files for which I believe it contains some button presses on the remote: [g003_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/10097359/g003_433.92M_250k.cu8.zip)

I dropped this into IQS and can see some data pulses, I think.

Finally, I ran `-A` on the file. I am a bit new to this, but saw some mentions of a flex decoder with value `-X 'n=name,m=OOK_PWM,s=1040,l=14940,r=21316,g=2024,t=5560,y=0'`, so I tried using that with:

```
rtl_433 -X 'n=name,m=OOK_PWM,s=1040,l=14940,r=21316,g=2024,t=5560,y=0'
```

Among other things, I am now seeing this output whenever I press any of the remote buttons:

```
time      : 2022-11-26 17:58:22
model     : name
count     : 12
num_rows  : 12
rows      :

len       : 1
data      : 0,

len       : 33
data      : ffffffff8,

len       : 1
data      : 0,

len       : 33
data      : ffffffff8,

len       : 1
data      : 0,

len       : 33
data      : ffffffff8,

len       : 1
data      : 0,

len       : 33
data      : ffffffff8,

len       : 1
data      : 0,

len       : 33
data      : ffffffff8,

len       : 1
data      : 0,

len       : 33
data      : ffffffff8
codes     : {1}0, {33}ffffffff8, {1}0, {33}ffffffff8, {1}0, {33}ffffffff8, {1}0, {33}ffffffff8, {1}0, {33}ffffffff8, {1}0, {33}ffffffff8
```

Any tips on how to proceed?

zuckschwerdt, 2022-11-27:
Use https://triq.org/pdv/ which has a more features, like pulse decode.
The signal looks really cheap, but is a good capture. Zoom in on the pulse data and you spot it's PPM with 1000 / 2000 µs.
E.g. use `-X 'n=name,m=OOK_PPM,s=1000,l=2000,r=3000'`
You can capture the different codes with a flex [decoder conf](https://github.com/merbanan/rtl_433/tree/master/conf).

If you want to regenerate that signal for transmission, try [tx_tools](https://github.com/triq-org/tx_tools/) maybe.

kspearrin, 2022-11-27:
@zuckschwerdt Thank you. Passing in `-X 'n=name,m=OOK_PPM,s=1000,l=2000,r=3000'` I was able to capture pulses of the following:

Brightness up

```
time      : 2022-11-27 12:29:23
model     : name
count     : 1
num_rows  : 1
rows      :

len       : 32
data      : 07380bf4
codes     : {32}07380bf4
```

Brightness down

```
time      : 2022-11-27 12:31:47
model     : name
count     : 1
num_rows  : 1
rows      :

len       : 32
data      : 07380af5
codes     : {32}07380af5
```

Change mode

```
time      : 2022-11-27 12:32:19
model     : name
count     : 1
num_rows  : 1
rows      :

len       : 32
data      : 07380cf3
codes     : {32}07380cf3
```

A few questions. Can you point me to some guides/info for:

1. How you determined the `-X` decoder from that PDV analysis? That way I don't have to bug you for future. Is there some getting started guide for how to analyze this? I am relatively new to analyzing RF.
2. How I could contribute a decoder for this remote/data?

zuckschwerdt, 2022-11-27:
There is a short introduction to identify pulse formats here: https://triq.org/rtl_433/PULSE_FORMATS.html#structure-of-fsk-and-ook-transmissions
And `rtl_433 -X` also has some short help summary.

Then figure out what the 1-bit and 0-bit symbols are, usually a short/long pulse or gap, that's the s(hort)/l(ong) values in the flex spec.

The format above might be inverted (0/1 flipped) with 24 bit ID first, then action. E.g. try also:
`rtl_433 -X 'n=name,m=OOK_PPM,s=1000,l=2000,r=3000,invert,bits=32,get=id:@0:{24}:%06x,get=action:@24:{8},unique'`

Write a conf file with your findings documented and PR that if you like.

closed 2022-11-28 by zuckschwerdt