merbanan/rtl_433 #2251: 22.11 not statically compiled on Windows

neversphere, 2022-11-28:
Out of the box it doesn't run due to a complaint about libwinpthread-1.dll being missing. On the same box version 21.12 runs fine.

zuckschwerdt, 2022-11-28:
Thanks for the report! We don't actually use pthreads on win32, so something else must have broken.

cizzop, 2022-12-09:
Same issue here. After putting libwinpthread-1.dll directly in the folder I then get "The application was unable to start correctly (0xc000007b)"

zuckschwerdt, 2022-12-09:
Ah ok, we only use native threads with MSC (Visual Studio), but pthreads on WIN32 otherwise (GCC).
I changed that to always native threads in dc95819 now. Still need to do a new build though.

zuckschwerdt, 2022-12-09:
If you want to test this right away here are the new binaries:
[rtl_433-win-x32.zip](https://github.com/merbanan/rtl_433/files/10198678/rtl_433-win-x32.zip)
[rtl_433-win-x64.zip](https://github.com/merbanan/rtl_433/files/10198679/rtl_433-win-x64.zip)

gdt, 2023-09-26:
This seems likely fixed, and we're 9 months after binaries were provided for testing, with no reply.

closed 2023-09-26 by gdt

neversphere, 2023-09-27:
Sorry I didn't update. This fixed it. Thanks!

gdt, 2023-09-27:
Thanks for confirming.