merbanan/rtl_433 #2252: `-M bits` are not included to `-F json` output

beta-tester, 2022-11-29:
hi,
when i run rtl_433 like this:
```bash
rtl_433 -F json -M time:iso:utc -M protocol -M level -M bits
...
{"time" : "2022-11-29T11:39:57", "protocol" : 19, "model" : "Nexus-T", "id" : 3, "channel" : 1, "battery_ok" : 0, "temperature_C" : 6.800, "mod" : "ASK", "freq" : 433.915, "rssi" : -10.579, "snr" : 28.555, "noise" : -39.134}
```
i can see the requested ` -M time:iso:utc`, `-M protocol` and `-M level` output, but i am missing the requested `-M bits` in the json output.

the rtl_433 version i use is:
```bash
~$ rtl_433 -V
rtl_433 version 22.11-5-gd4c0e626 branch master at 202211211916
```

zuckschwerdt, 2022-11-29:
That only takes effect on `decoder_output_bitbuffer*()` calls in the decoders. Currently only Digitech-XC0324 and Ikea-Sparsnas use that and only in verbose mode.

The `-M bits` option is only for debug when developing new decoders. I guess we should remove it from user-facing documentation and help.

closed 2022-11-29 by zuckschwerdt

gdt, 2022-11-29:
Perhaps you should not remove -M bits, but instead caution that most drivers do not support it.  The real question is would it be better if support were added to the rest, and if not, why is it in some?

zuckschwerdt, 2022-11-29:
The `decoder_output_*()` functions actually send data to the outputs (JSON, CSV, MQTT…) and are not used much. I guess we'll be using the newer `decoder_log*()` functions instead at some point. The logging just needs to be more configurable and should optionally also go to the outputs.

beta-tester, 2022-11-29:
> The -M bits option is only for debug when developing new decoders. I guess we should remove it from user-facing documentation and help.

i made a python script that collects temperature sensor data of my neighborhood, to have the outside temperature. i hard coded the model,id,channel of sensors i am interested in to give my the outside temperature close to my appartement.
it happens that some sensors disappear and new sensors appear - e.g., when the battery is changed, some sensors seems to get a new id. so extend my collecting script by adding unknworn sensors automatically.
there i found out, that some of the sensor `temperature_C` values seems to give incorrect values.

to see whats going on there i tried to get additional information in the json output.

i want to get this `bits` output for debugging and i want it in the json output, because i have only one rtl-sdr device and i don't want to interrupt the normal work of my script. so it would be nice to have the `-M bits` and other `-M` stuff in the normal json output to not break the normal behavoir if my script.

zuckschwerdt, 2022-11-29:
Well it does say "debug" and "code outputs": `Use "bits" to add bit representation to code outputs (for debug).`

But, yes, that's the use-case we want to support. The biggest chunks of refactoring are already done, fprintfs in decoders are now decoder_log calls. The final piece now is to add verbosity levels to outputs so log messages can be included.

I'll set up a branch with these changes shortly. The "bits" option and (mostly unused now) decoder_output functions will be removed. You can then raise the verbosity globally (`-v`) or per decoder (`-R 99,v`) and configure outputs to display those (`-F json,v`).

beta-tester, 2022-11-29:
you mean with option `-v -F json,v` i receive the additional extra data that are of interest for debugging in the json output... without the need of enabling `-M bits` option...

that's nice. thank you.

zuckschwerdt, 2022-11-29:
Not right now though. The code needs to be written.

zuckschwerdt, 2022-11-29:
Preliminary test code is in ##2254.