merbanan/rtl_433 #2256: WHE542 data seems not decoded

algorni, 2022-12-01:
Hi,
this is about a Siemens Heating cost allocator device using W M-BUS.

I'm able to receive the package but it seems the data packet is not fully decoded as i never got "inst_volume_flow_min_1" valued

Look:

`{"time":"2022-12-01 10:05:58","model":"Wireless-MBus","mode":"S","M":"LSE","id":93442859,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532592844933408780b6e0000000dff5f2f0001a0807f3208b06effffde2967100000de2b000000007501f700a9003a00080000000000000000000000d5fb0000046d130bc12c0648","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}`

And i receive a lot of them! 

[image]

How can i help to understand if this is something doable or not?

algorni, 2022-12-01:
Here is some other JSON output 

{"time":"2022-12-01 10:06:03","model":"Wireless-MBus","mode":"S","M":"LSE","id":92838933,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532338983923408780b6e1901000dff5f2f0001b2807e3408b06effffde2990160000de2b150100009b012301ed00fe0075000000000000000000000066f97300046d1f0bc12cb240","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:08:19","model":"Wireless-MBus","mode":"S","M":"LSE","id":93451103,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532031145933408780b6e0000000dff5f2f0001b180ff1008b06effffde2901000000de2b000000000000000000000000000000000000000000000000ffff0000046d150bc12cf3e2","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:06:27","model":"Wireless-MBus","mode":"S","M":"LSE","id":93451113,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532131145933408780b6e5700000dff5f2f0001b1807f2308b06effffde2929040000de2b5400000074005a007200380000000000000000000000000053fe3600046d130bc12c003b","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:06:58","model":"Wireless-MBus","mode":"S","M":"LSE","id":92838934,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532348983923408780b6e6600000dff5f2f0001b280fe2a08b06effffde2963140000de2b630000005d0156012e01360143000000000000000000000049fa3f00046d1a0bc12c99e0","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:07:36","model":"Wireless-MBus","mode":"S","M":"LSE","id":93451085,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532851045933408780b6e0000000dff5f2f0001b180000308b06effffde2973000000de2b000000001c00210002000000010000000000000000000000b7ff0000046d160bc12ce29c","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:08:56","model":"Wireless-MBus","mode":"S","M":"LSE","id":92838938,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532388983923408780b6e0400000dff5f2f0001b2807e3508b06effffde2909000000de2b040000000400010000000100000000000000000000000000f7ff0400046d1a0bc12c1a70","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:08:58","model":"Wireless-MBus","mode":"S","M":"LSE","id":93442857,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532572844933408780b6e0300000dff5f2f0001a0807e3008b06effffde2912030000de2b02000000610068003f000c00030000000000000000000000c8fe0200046d1b0bc12ca7c7","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

merbanan, 2022-12-01:
Ask for support in the https://github.com/weetmuts/wmbusmeters project.

merbanan, 2022-12-01:
But it looks like you got the inst_volume_flow_min_1 key here.

merbanan, 2022-12-01:
I guess you mean that it is never any other value then 0.

algorni, 2022-12-06:
@merbanan  you are correct, i mean it's never anything else than 0, anyway i'm going to keep that under monitor to see eventually if the value is just sent once per day or per few hours...   till now i saw always 0  (even for radiators that is actually hot) 

Will ask also in the project you mentioned.  Thank you very much!

algorni, 2022-12-06:
@merbanan  it sounds with the driver qcaloric i'm able to receive some value with wmbusmeters as current_consumption_hca -> 13

`{"media":"heat cost allocation","meter":"qcaloric","name":"MyHCA","id":"93442865","status":"OK","current_consumption_hca":13,"set_date":null,"consumption_at_set_date_hca":null,"set_date_1":null,"consumption_at_set_date_1_hca":null,"device_date_time":"2022-12-06 14:57","timestamp":"2022-12-06T13:40:12Z","device":"rtlwmbus[00000001]","rssi_dbm":59}`

algorni, 2022-12-06:
OOKk ish

I can receive my heat cost allocator:

`{"media":"heat cost allocation","meter":"qcaloric","name":"Studio","id":"93450758","status":"OK","current_consumption_hca":24,"set_date":null,"consumption_at_set_date_hca":null,"set_date_1":null,"consumption_at_set_date_1_hca":null,"device_date_time":"2022-12-06 15:05","timestamp":"2022-12-06T13:50:57Z","device":"rtlwmbus[00000001]","rssi_dbm":60}`

I see the data is overwritten multiple time on file:

`{"media":"heat cost allocation","meter":"qcaloric","name":"Studio","id":"93450758","status":"OK","current_consumption_hca":24,"set_date":null,"consumption_at_set_date_hca":null,"set_date_1":null,"consumption_at_set_date_1_hca":null,"device_date_time":"2022-12-06 15:07","timestamp":"2022-12-06T13:53:06Z","device":"rtlwmbus[00000001]","rssi_dbm":68}
`
`{"media":"heat cost allocation","meter":"qcaloric","name":"Studio","id":"93450758","status":"OK","current_consumption_hca":24,"set_date":null,"consumption_at_set_date_hca":null,"set_date_1":null,"consumption_at_set_date_1_hca":null,"device_date_time":"2022-12-06 15:12","timestamp":"2022-12-06T13:57:19Z","device":"rtlwmbus[00000001]","rssi_dbm":54}`

Interesting..

algorni, 2022-12-06:
Ok my last understanding is that inst_volume_flow_min_1 is not valid as value for such kind of telegram

Looking here:

https://wmbusmeters.org/analyze/48446532281145933408780B6e1600000dff5f2f0001Be80811B08B06effffde2945000
000de2B050000000d000B0005000400010000000000000000000000d3ff0500046d2c0ec62c65

This kind of device send out a telegram which contains current_consumption_hca and device_date_time as proper info.

@merbanan do you know if rtl_433 can decode such telegram properly?

merbanan, 2022-12-06:
AFAIK this part of the payload is not OMS-compliant and thus fails to decode properly with rtl_433. rtl_433 can receive the telegram properly but can not decode the payload in the telegram. My stance has been that decoding should be left to the wmbusmeeters project as they actively supports all kind of sensors. But if some one want to add support for some specific sensor to rtl_433 I am not stopping them.

merbanan, 2022-12-06:
The online decoder was really nice. There is a large chunk of data that is not currently decoded by wmbusmeters. Open a ticket and see if they can help out.

[46446532572844933408780b6e0300000dff5f2f0001a0807e3008b06effffde2912030000de2b02000000610068003f000c00030000000000000000000000c8fe0200046d1b0bc12ca7c7](https://wmbusmeters.org/analyze/46446532572844933408780b6e0300000dff5f2f0001a0807e3008b06effffde2912030000de2b02000000610068003f000c00030000000000000000000000c8fe0200046d1b0bc12ca7c7)

The red chunk here should decode to more data.

algorni, 2022-12-13:
Thanks both.  I learned a lot out of these threads.

It turn outs that rtl_433 decode already well the heat cost allocator telegram

`{"time":"2022-12-13 08:24:48","model":"Wireless-MBus","mode":"S","M":"LSE","id":93451119,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":39,"data":"244465321911459334087ae80000000b6e2800004b6e441000426cde29326cffff046d2809cd2c1142","mic":"CRC","CI":122,"AC":232,"ST":0,"CW":0,"inst_hca_0":"28.000 ","inst_hca_1":"1044.000 ","inst_date_1":"22-09-30","err_date_0":"invalid","inst_timedate_0":"22-12-13T09:40:00"}`

Actually i can receive the current consumption but also the historical one!      inst_hca_0   and inst_hca_1 

That's more than enought so we can close this specific ticket.

closed 2022-12-13 by algorni