merbanan/rtl_433 #2257: Bresser 5 in 1 issue

mczakk, 2022-12-01:
I have a Bresser 5 in 1 running on rtl_433 via node red.
This morning i noticed that the rainfall value  had spiked to 66616.6015mm from 503.6
Interestingly, this only happened when the temperature dipped below freezing
This is the data as output by node red, which reorts every 5 minutes:
```
{"time":"2022-12-01 07:06:39","model":"Bresser-6in1","id":939865349,"battery_ok":1,"temperature_C":0.1,"humidity":99,"wind_max_ms":0,"wind_avg_ms":0,"wind_dir_deg":96,"rain_mm":503.60001,"mic":"CRC"}
{"time":"2022-12-01 07:11:27","model":"Bresser-6in1","id":939865349,"battery_ok":1,"temperature_C":0,"humidity":99,"wind_max_ms":0,"wind_avg_ms":0,"wind_dir_deg":96,"rain_mm":503.60001,"mic":"CRC"}
{"time":"2022-12-01 07:16:27","model":"Bresser-6in1","id":939865349,"temperature_C":-0.1,"humidity":99,"wind_max_ms":0,"wind_avg_ms":0,"wind_dir_deg":96,"rain_mm":66616.60156,"mic":"CRC"}
{"time":"2022-12-01 07:21:27","model":"Bresser-6in1","id":939865349,"battery_ok":1,"temperature_C":0.1,"humidity":99,"wind_max_ms":0,"wind_avg_ms":0,"wind_dir_deg":96,"rain_mm":503.60001,"mic":"CRC"}
```
here is the raw data for the same period:
```
{"time":"2022-12-01 07:11:39","model":"Bresser-6in1","id":939865349,"channel":0,"battery_ok":1,"temperature_C":0,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:11:51","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:12:03","model":"Bresser-6in1","id":939865349,"channel":0,"battery_ok":1,"temperature_C":0,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:12:15","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:12:27","model":"Bresser-6in1","id":939865349,"channel":0,"battery_ok":1,"temperature_C":0,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:12:39","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:12:51","model":"Bresser-6in1","id":939865349,"channel":0,"battery_ok":1,"temperature_C":0,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:13:03","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:13:15","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:13:15","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:13:27","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:13:39","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:13:39","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:13:51","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:14:03","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:14:03","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:14:15","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:14:27","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:14:27","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:14:39","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:14:51","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:14:51","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:15:03","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:15:15","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:15:15","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:15:27","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:15:39","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:15:39","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:15:51","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:16:03","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:16:03","model":"Bresser-6in1","id":939865349,"channel":0,"temperature_C":-0.1,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":66616.60156,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:16:15","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
[... cut ...]
{"time":"2022-12-01 07:19:27","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
{"time":"2022-12-01 07:19:39","model":"Bresser-6in1","id":939865349,"channel":0,"battery_ok":1,"temperature_C":0,"humidity":99,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":96,"uv":0,"startup":1,"flags":0,"mic":"CRC"}
{"time":"2022-12-01 07:19:51","model":"Bresser-6in1","id":939865349,"channel":0,"sensor_type":1,"wind_max_m_s":0.6,"wind_avg_m_s":0.6,"wind_dir_deg":96,"rain_mm":503.60001,"startup":1,"flags":1,"mic":"CRC"}
```
Is this a known issue?

merbanan, 2022-12-01:
Can you provide a signal sample for this ? -S known

mczakk, 2022-12-01:
I'm not sure, do you mean a sample of the raw rtl_433 output?
I run rtl_433 on a raspberry pi, which is setup so that rtl runs on boot, so I can't open another instance. 
If I add -S known to the run command will that work?
Also to replicate the problem I would have to wait until the temperature dips below freezing again - so the signal file may become very large!

mczakk, 2022-12-01:
OK, i have looked into this and found that when running rtl via supervisor (at boot) using the bash script 
```#!/bin/bash
sudo -u testpi rtl_433 -S known
```
it doesn't seem to write a .cu8 file anywhere i can find.
However, if i run the bash script in a terminal it does save a .cu8 file for every signal into the main rtl_433 folder.
is there a way to tell rtl_433 where to save the .cu8 file (ideally a nominated folder)?
I will run overnight  and see if the temperature drops below zero and try and isolate the signal files around that time.

mczakk, 2022-12-02:
Ok, I now have signal files for the times that the error occurs, it definitely happens when the temp goes below 0C.
The first instance is at 2022-12-01 22:56
[bresser_data.zip](https://github.com/merbanan/rtl_433/files/10139924/bresser_data.zip)
the uploaded file contains the node red output and the signal files, hope this helps!

franki29, 2022-12-02:
Hi, same issue and solution like  #2184
Br. Frank

franki29, 2022-12-02:
[image]

mczakk, 2022-12-08:
@franki29 I've just done a complete re install , and changed  line 179 to:
//int rain_ok   = msg[12] <= 0x99 && msg[13] <= 0x99 && msg[14] <= 0x99;
    int rain_ok = msg[12] <= 0x65 && msg[13] <= 0x99 && msg[14] <= 0x99;
before running the cmake command sequence.
unfortunately as i was doing this the outside temp has crept up to 0.0deg !
I guess i'll have to wait until it drops again to see if it's worked!

Will report back then :)

franki29, 2022-12-08:
Hi @mczakk , you do not have to wait, you can use your cu8 files with the command rtl_433 -r *.cu8 in the folder where you have stored your cu8 files

mczakk, 2022-12-08:
[part.zip](https://github.com/merbanan/rtl_433/files/10185206/part.zip)
[image]
 hi @franki29 , I ran rtl_433 -r *.cu8 on the files in part.zip, it gave the output shown in temp.png, still reading 66k mm rainwhen below 0degC!

franki29, 2022-12-08:
HI @mczakk looks like you still using a older version of rtl_433.
Did you previously installed the rtl_433 via apt? 
If so, please first purge this version, than run "sudo make install" again. 
Or start the rtl with. /usr/local/bin/rtl_433 -r *.cu8. 
If this works, than you have two different installations on your raspberry and you should purge the old one using apt.

mczakk, 2022-12-08:
@franki29 yes it seemed that i had two installs on my pi.
removed both and reinstalled with modified line 179.
ran rtl_433 -r *.cu8 on part.zip and values were all good.
waiting to see if tonights frost is reported properly now!
Many Thanks
:) :)

mczakk, 2022-12-08:
Hey @franki29, just to let you know - outside temp has just dropped below zero, and the rain reading stayed stable!

Thanks for your help 😁

closed 2022-12-08 by mczakk

merbanan, 2022-12-08:
Unfortunately it looks like there is no clear way yet know what type a specific message is. Your samples will help us at least.