merbanan/rtl_433 #2261: RTL TUBE LIKE SCE-R1 GUNMETAL SOMFY PROTOCOL PROBLEM

filzek, 2022-12-02:
Hi Guys,

The TUBE® remoto control use a RTL motor but the rfl_433 detects it as SOMFY RTL
[image]
[image]
using the RFLINK it also detects it as a SOMFY RTS but the problem start that the remote ID has 7 positions beside 6.

Is there anything that can be done to solve this in RTS/TUBE problem?

zuckschwerdt, 2022-12-02:
The Somfy-RTS has a well defined protocol with a checksum. So I guess this really is a Somfy-RTS message.
There are 6 bytes and the last 3 are the address (6 characters / "positions"?).
https://github.com/merbanan/rtl_433/blob/24c2a2baf4773eb7df2dc67f1e3c73b5cab89908/src/devices/somfy_rts.c#L42-L46

Can you give an example what the problem is?

filzek, 2022-12-05:
Do you want me to send the data, let me know which rtl_433 command to check do you want me to run and i will do.

i have run:

Protocols: Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
SDR: Found 1 device(s)
SDR: trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T/2 tuner
SDR: Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
SDR: Sample rate set to 250000 S/s.
SDR: Tuner gain set to Auto.
SDR: Tuned to 433.920MHz.
Allocating 15 (non-zero-copy) user-space buffers
Baseband: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Detected OOK package    2022-12-05 13:00:52
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 13:00:52
model     : Somfy-RTS    id        : 2A8265
Control   : Stop (5)     Counter   : 1069          Retransmission: 0         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 13:00:52
model     : Somfy-RTS    id        : 2A8265
Control   : Stop (5)     Counter   : 1069          Retransmission: 1         Integrity : CHECKSUM
Analyzing pulses...
Total count:   90,  width: 338.25 ms            (84563 S)
Pulse width distribution:
 [ 0] count:    1,  width: 9988 us [9988;9988]  (2497 S)
 [ 1] count:    9,  width: 2524 us [2524;2528]  ( 631 S)
 [ 2] count:    2,  width: 4840 us [4840;4840]  (1210 S)
 [ 3] count:   34,  width: 1304 us [1300;1312]  ( 326 S)
 [ 4] count:   44,  width:  664 us [660;672]    ( 166 S)
Gap width distribution:
 [ 0] count:    1,  width: 96828 us [96828;96828]       (24207 S)
 [ 1] count:    9,  width: 2540 us [2540;2544]  ( 635 S)
 [ 2] count:   34,  width: 1248 us [1244;1256]  ( 312 S)
 [ 3] count:   44,  width:  612 us [608;620]    ( 153 S)
 [ 4] count:    1,  width: 32776 us [32776;32776]       (8194 S)
Pulse period distribution:
 [ 0] count:    1,  width: 106816 us [106816;106816]    (26704 S)
 [ 1] count:   11,  width: 5252 us [5068;6092]  (1313 S)
 [ 2] count:   34,  width: 1920 us [1916;1928]  ( 480 S)
 [ 3] count:   16,  width: 2556 us [2556;2564]  ( 639 S)
 [ 4] count:   26,  width: 1280 us [1272;1284]  ( 320 S)
 [ 5] count:    1,  width: 33444 us [33444;33444]       (8361 S)
Pulse timing distribution:
 [ 0] count:    1,  width: 9988 us [9988;9988]  (2497 S)
 [ 1] count:   18,  width: 2532 us [2524;2544]  ( 633 S)
 [ 2] count:    2,  width: 4840 us [4840;4840]  (1210 S)
 [ 3] count:   68,  width: 1276 us [1244;1312]  ( 319 S)
 [ 4] count:   88,  width:  640 us [608;672]    ( 160 S)
 [ 5] count:    2,  width: 98356 us [96828;99884]       (24589 S)
 [ 6] count:    1,  width: 32776 us [32776;32776]       (8194 S)
Level estimates [high, low]:  15973,     59
RSSI: -0.1 dB SNR: 24.3 dB Noise: -24.4 dB
Frequency offsets [F1, F2]:  -10030,      0     (-38.3 kHz, +0.0 kHz)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB0110701270409E412E804FC0280FFFF80088555+AAB03A0701270409E412E804FC0280FFFF80089191A3B4C3B3B3B3C4B3C4B4C4C3C4B3B4C4C4C3B4C4C3B3B4C3B4C4C4C4C3B3B4C4C3B3B4C4C3B4C4C655+AAB03F0701270409E412E804FC0280FFFF800891919191919191A3B4C3B3B3B3C4B3C4B4C4C3C4B3B4C4C4C3B4C4C3B3B4C3B4C4C4C4C3B3B4C4C3B3B4C4C3B4C4C555

The learned control when using to send a TX with the correct rolling code in RFLINK doesnt work.

This only happen with TUBE remote controls, TUBE doesnt work with SOMFY blinds. Also one interesting part is that TUBE uses a 433.92mhz frequency besides the SOMFY 433.42mhz.

Is there any extra debug or anything aditional that I could do to send the remote control information and data to you?

filzek, 2022-12-11:
@zuckschwerdt Hi bro, 

So, maybe the Encryption KEY must match to the sender / receiver? Is there anyway to print the Encryption Key used in this transmission?

filzek, 2022-12-11:
we have add  seed to be printed to check if the seed/key is changed among comands or keys pressed as show:

a     /* clang-format off */
    data = data_make(
            "model",          "",               DATA_STRING, "Somfy-RTS",
            "id",             "",               DATA_FORMAT, "%06X", DATA_INT, address,
            "control",        "Control",        DATA_STRING, control_str,
            "counter",        "Counter",        DATA_INT,    counter,
            "retransmission", "Retransmission", DATA_INT,    is_retransmission,
            "mic",            "Integrity",      DATA_STRING, "CHECKSUM",
            "seed Decimal",   "seed",           DATA_INT,    seed,
            "seed Hex",       "",               DATA_FORMAT, "0x%0X", DATA_INT, seed,
            NULL);
    /* clang-format on */

and this is quite interesting to check as, for each key (UP, DOWN, STOP, PROG) it uses an exclusive key as per:
KEY // seed/key
UP // 0x96 hex
STOP // 0x95 hex
DOWN // 0x98 hex
PROG // 0x9C hex

So, it seens that the TUBE uses a fixed seed per key pressed.

Also the timmings are a little better adjusted than the SOMFY Trellis remote. 

We will try to reproduce (clone) the remote using this new parameters so this could be the trick to use the Somfy library correctly.

filzek, 2023-02-02:
we have found all changes to use with SOMFY RTS TUBE

frame[0] = 0x95;         // is the COMMAND FRAME
frame[1] = (unsigned char)(15)<< 4;   //is fixed always

so, the CMD as off:
#define TUBE_SCE_R1_CMD_PROG 0x9C
#define TUBE_SCE_R1_CMD_STOP 0x95
#define TUBE_SCE_R1_CMD_UP 0x96
#define TUBE_SCE_R1_CMD_DOWN 0x98
#define TUBE_SCE_R1_CMD_UP_DOWN 0x9A

Repeate the FRAMES to use special Command: 
to make MY working just sent the STOP 10 times
to make the PROG to work just send it 25 times!

so all is done!!!

gdt, 2023-09-29:
What's the status and next steps?  I don't quite see how this is aligned wtih creating a decoder for rtl_433 and if not probably should be closed.

gdt, 2024-06-04:
abandoned

filzek, 2024-06-04:
It must show the correct decode not as Somfy, but as Somfy Tube protocol.