merbanan/rtl_433 #2267: Support for Quinetic 433Mhz Button

pazzernick, 2022-12-04:
Hi,

Firstly, thanks for the fantastic rtl_433 tooling. Amazing work!

I'm trying to add support for a kinetic-powered switch, which is typically 'paired' via RF to one or more receivers such as Smart Plugs, Lighting Relay, etc. Model: [Quinetic QUGDMK](https://www.tlc-direct.co.uk/Products/QUGDMK.html)

I've demodulated the samples in **Universal Radio Hacker** and think I've found the data...
* Device Model (48-bits from bit 85)
* Unique ID (32-bits from bit 133)

Signal Characteristics:
* Signal Type: FSK PCM
* Signal captured at 2048k sample-rate
* Button is designed to broadcast 3 times (mentioned in product docs) which is why 3 (sometimes 4) pulses occur.
* I've tried to capture signals at 250k and 1024k sample rate, no success with URH or rtl_433.

Command used to capture samples ( antenna attached to R820T )
```
rtl_433 -s 2048000 -R 0 -Y autolevel -Y squelch -g 20 -S unknown
```

Analysis output:
```
# rtl_433 -s 2048000 -R 0 -A button_11_g003_433.92M_2048k.cu8

rtl_433 version 22.11-9-g6e1f1202 branch master at 202211291952 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 0 out of 223 device decoding protocols [ ]
Test mode active. Reading samples from file: button_11_g003_433.92M_2048k.cu8
Detected OOK package	@0.738655s
Analyzing pulses...
Total count:    3,  width: 8.11 ms		(16604 S)
Pulse width distribution:
 [ 0] count:    3,  width: 1313 us [1308;1321]	(2690 S)
Gap width distribution:
 [ 0] count:    2,  width: 2083 us [2083;2083]	(4266 S)
Pulse period distribution:
 [ 0] count:    2,  width: 3399 us [3395;3404]	(6962 S)
Pulse timing distribution:
 [ 0] count:    3,  width: 1313 us [1308;1321]	(2690 S)
 [ 1] count:    2,  width: 2083 us [2083;2083]	(4266 S)
 [ 2] count:    1,  width: 13213 us [13213;13213]	(27061 S)
Level estimates [high, low]:   1000,      9
RSSI: -12.1 dB SNR: 20.5 dB Noise: -32.6 dB
Frequency offsets [F1, F2]:     941,      0	(+29.4 kHz, +0.0 kHz)
Guessing modulation: Un-modulated signal. Maybe a preamble...
view at https://triq.org/pdv/#AAB10305210823339D81818255
```

Failing flex command:
```
# rtl_433 -s 2048000 -R 0 -Y minmax -X "name=Quinetic,m=FSK_PCM,s=10,l=10,r=600,t=1" button_11_g003_433.92M_2048k.cu8

rtl_433 version 22.11-9-g6e1f1202 branch master at 202211291952 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 223 device decoding protocols [ ]
Test mode active. Reading samples from file: button_11_g003_433.92M_2048k.cu8
baseband_demod_FM: low pass filter for 2048000 Hz at cutoff 409600 Hz, 2.4 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.738655s
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 124          data      : 02aaaaaaaaa10d2118dc380985826b4
codes     : {124}02aaaaaaaaa10d2118dc380985826b4
```

[quinetic_button11_button12.zip](https://github.com/merbanan/rtl_433/files/10148538/quinetic_button11_button12.zip)

Any help in identifying the signal / flex-config would be greatly appreciated....

zuckschwerdt, 2022-12-04:
That looks like a pretty fast signal, naturally as there is not much time/energy to transmit.
The bit width is 10 µs (100 kBps), which is why you need a high sample rate.

This should work: ` rtl_433 -Y minmax -R 0 -X 'n=name,m=FSK_PCM,s=10,l=10,r=90'`

Unlikely the correct preamble/sync, but you can for now sync on `5554` to get aligned codes, e.g. `,preamble=5554`
Looks like about 80 bits payload.

pazzernick, 2022-12-04:
Thanks for the suggestion. In URH, my payload looks to be 70 bits in length.

With the suggested flex config I get 2 different lengths of data. **Button-11** is showing 79 and **Button-12** is showing 45.

Button 11
```
rtl_433 -s 2048000 -R 0 -Y minmax -X "name=Quinetic,m=FSK_PCM,s=10,l=10,r=90,preamble=5554" button_11_g003_433.92M_2048k.cu8

rtl_433 version 22.11-9-g6e1f1202 branch master at 202211291952 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 223 device decoding protocols [ ]
Test mode active. Reading samples from file: button_11_g003_433.92M_2048k.cu8
baseband_demod_FM: low pass filter for 2048000 Hz at cutoff 409600 Hz, 2.4 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.738655s
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 79           data      : 21a4231b870130b04d68
codes     : {79}21a4231b870130b04d68
```

Button 12
```
rtl_433 -s 2048000 -R 0 -Y minmax -X "name=Quinetic,m=FSK_PCM,s=10,l=10,r=90,preamble=5554" button_12_g003_433.92M_2048k.cu8 

rtl_433 version 22.11-9-g6e1f1202 branch master at 202211291952 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 223 device decoding protocols [ ]
Test mode active. Reading samples from file: button_12_g003_433.92M_2048k.cu8
baseband_demod_FM: low pass filter for 2048000 Hz at cutoff 409600 Hz, 2.4 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.651895s
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 45           data      : 21a423b69000
codes     : {45}21a423b69000
```

I'm not sure how the 'reset=90' is derived, can you give me a hint on how to tune it, or should I leave it alone?

If I'm not barking up the wrong tree, I believe this is the Hex data from Button 11. I guess that's what I'm expecting to see in rtl_433

```
fffe6666666666666666666018079860181e079f807e00061e019e0061e67986080 [Pause: 4224 samples]
f7f333333333333333333300c03cc300c0f03cfc03f00030f00cf0030f33cc3060 [Pause: 4224 samples]
ffe8ccccccccccccccccccc0300f30c0303c0f3f00fc000c3c033c00c3ccf30c1a [Pause: 43478 samples]
```

zuckschwerdt, 2022-12-04:
> With the suggested flex config I get 2 different lengths of data. **Button-11** is showing 79 and **Button-12** is showing 45.

I didn't check how many consecutive 0-bits are expected, use `r=200` then. Basically you want to go slightly bigger than the longest expected gap, e.g. a byte of 0's and some margin gives 90. Reasonable protocols use bit-stuffing, whitening, or such to limit the number of consecutive symbols -- but that's no used here apparently.

> Hex data from Button 11. I guess that's what I'm expecting to see in rtl_433

That does not look valid, all your symbols are multiples of length 2. Maybe you mismatched sample rate and bit length?

pazzernick, 2022-12-04:
I may have answered my own question... using reset=120, I have similar data lengths now. Hopefully I can extract some differences to detect the Switch UID.

Button 11
```
rtl_433 -s 2048000 -R 0 -Y minmax -X 'name=Quinetic,m=FSK_PCM,s=10,l=10,r=120,preamble=5554' button_11_g003_433.92M_2048k.cu8 
rtl_433 version 22.11-9-g6e1f1202 branch master at 202211291952 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 223 device decoding protocols [ ]
Test mode active. Reading samples from file: button_11_g003_433.92M_2048k.cu8
baseband_demod_FM: low pass filter for 2048000 Hz at cutoff 409600 Hz, 2.4 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.738655s
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 79           data      : 21a4231b870130b04d68
codes     : {79}21a4231b870130b04d68
```

Button 12
```
rtl_433 -s 2048000 -R 0 -Y minmax -X 'name=Quinetic,m=FSK_PCM,s=10,l=10,r=120,preamble=5554' button_12_g003_433.92M_2048k.cu8 
rtl_433 version 22.11-9-g6e1f1202 branch master at 202211291952 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/nickparrott/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 223 device decoding protocols [ ]
Test mode active. Reading samples from file: button_12_g003_433.92M_2048k.cu8
baseband_demod_FM: low pass filter for 2048000 Hz at cutoff 409600 Hz, 2.4 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.651895s
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 83           data      : 21a423b69000aacca6b48
codes     : {83}21a423b69000aacca6b48
```

merbanan, 2022-12-05:
You can use dd to split out relevant sections. Use -W to let rtl_433 to do the demodulation of FSK data.

zuckschwerdt, 2022-12-05:
There is one more option to split/crop cu8 files soon: pdv will get a "save as" feature. There is a (bare bones) preview at https://triq.org/spectrogram-next/

merbanan, 2022-12-05:
The PV stuff is just for visualization. To get the bits you need to use rtl_433. I think you have to share a sample to get any better advice from me.

pazzernick, 2022-12-05:
OK, thanks for confirming...

There are 4 zipped samples on the first post, if you can shed any light on how I might get the online PDV to work with these PCM samples.

The flex-config I'm testing with is below. It's 95% working, but cannot handle the occasional payload which is "too short", e.g. 4th pulse from switch, which is frequently useless and just needs ignoring.

That is why I'm hoping to start on **./src/devices/quinetic.c** shortly, so the payload can be checked.

```
# Sample Data:
#
#       Button X
# 21 a4 23 1b 87 01 30 b0 4d 68 (PUSH)
# 21 a4 23 1b 87 c0 7c ee a6 b4 80 (RELEASE)
#
#       Button Y
# 21 a4 23 b6 90 00 aa cc a6 b4 8 (PUSH)
# 21 a4 23 b6 90 c0 9c f4 4d 69 0 (RELEASE)
#
#

decoder {
    name=Quinetic,
    modulation=FSK_PCM,
    short=10,
    long=10,
    reset=120,
    tolerance=2,
    match={24}0x21a423,
    preamble=5554,
    get=id:@24:{16},
    get=action:@40:{4}:[0:Push 12:Release],
}
```

zuckschwerdt, 2022-12-05:
Are the IDs printed somewhere on the device or known from some receiver, and shouldn't that be 32 bit?
Do you know which part of `21 a4 23` is a sync word /type /model code?
The changes from push to release seem random, is it encrypted? Is there a 2-way pairing?
Are all push messages the same or do some bits always keep changing?

pazzernick, 2022-12-05:
> Are the IDs printed somewhere on the device or known from some receiver, and shouldn't that be 32 bit?

I presumed it would be 32-bit, but really not sure. Each device has a numeric value on a barcode-label. It may include the ID, but I cannot be certain...

Value mappings:
```
Button 11
codes     : {79}21a4231b870130b04d68
label     : EBM1202206181272

Button 12
codes     : {83}21a423b69000aacca6b48
label     : EBM1202206181263
```

> Do you know which part of `21 a4 23` is a sync word /type /model code?

I haven't got a clue, but I have 2 physically separate switches, both sharing the leading 24-bits `21 a4 23`. I have 10 other switches at another site, so can surely check for further variations. However, they are all the same model.

My objective of the day is to demodulate this accurately so that I can answer your question about the sync word... ;-)

> The changes from push to release seem random, is it encrypted?

I'm stabbing in the dark, but made a guess that bit 40 is relating to the PUSH vs RELEASE.

When doing repeated physical testing, the changes on bit 40 seem to be consistent, e.g. when bit40 is 0, it seems to be push rather than release.

If there are repeated packets ( which occurs sometimes ) they always seem to be in the correct order ( e.g. 1 or more PUSH messages, then 1 or more RELEASE messages )

> Is there a 2-way pairing? Are all push messages the same or do some bits always keep changing?

I don't think it is 2-way pairing... RF receivers (up to 10) can be paired to a single button. It seems that push and release messages are consistent (with the exception of the data length which varies)

Example - 3 push and release cycles on Button 11

```
time      : 2022-12-05 14:01:04
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 83           data      : 21a4231b870130b04d690                   id        : 7047
action    : Push
codes     : {83}21a4231b870130b04d690
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 14:01:04
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 82           data      : 21a4231b87c0f9dd4d690                   id        : 7047
action    : Release
codes     : {82}21a4231b87c0f9dd4d690

<GAP>

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 14:01:14
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 82           data      : 21a4231b870130b04d690                   id        : 7047
action    : Push
codes     : {82}21a4231b870130b04d690
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 14:01:14
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 82           data      : 21a4231b87c0f9dd4d690                   id        : 7047
action    : Release
codes     : {82}21a4231b87c0f9dd4d690

<GAP>

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 14:01:20
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 83           data      : 21a4231b870130b04d690                   id        : 7047
action    : Push
codes     : {83}21a4231b870130b04d690
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 14:01:20
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 82           data      : 21a4231b87c0f9dd4d690                   id        : 7047
action    : Release
codes     : {82}21a4231b87c0f9dd4d690
```

I'm totally open to suggestions... Radio is new territory for me so apologies for any confusion generated!

zuckschwerdt, 2022-12-05:
If you invert the bits then your lower ID byte has the same delta of 9, [see BitBench](https://triq.net/bitbench#c=21a423b69000aacca6b4%5BEBM%201202206181%20263%5D&c=21a4231b870130b04d68%5BEBM%201202206181%20272%5D&f=8h%208h%208h%208d%208d%208h8h%20%208h8h%208h&i=true&cw=4).
Hard to tell from just two sensors, but that should be part of the ID, yes.

What I meant about push or release messages: are different presses always the same code? I.e. do you only ever get the same two codes for one device?

This looks too strange:
```
21 a4 23 1b 87 c0 7c ee a6 b4 (RELEASE Button X)
21 a4 23 b6 90 00 aa cc a6 b4 (PUSH Button Y)
                        ^^^^^ why?
21 a4 23 1b 87 01 30 b0 4d 68 (PUSH Button X)
21 a4 23 b6 90 c0 9c f4 4d 69 (RELEASE Button Y)
                        ^^^^^ why?
```

pazzernick, 2022-12-05:
I'd be more comfortable removing the **action** because it's potentially inaccurate.

Anyway, it looks like the PUSH message does vary. Here is another sample showing changes.

```
time      : 2022-12-05 15:05:54
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 83           data      : 21a4231b870130b04d690                   id        : 7047
action    : Push
codes     : {83}21a4231b870130b04d690
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 15:05:55
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 79           data      : 21a4231b87c0f9dd4d68                    id        : 7047
action    : Release
codes     : {79}21a4231b87c0f9dd4d68

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 15:06:21
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 79           data      : 21a4231b870130b04d68                    id        : 7047
action    : Push
codes     : {79}21a4231b870130b04d68
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 15:06:22
model     : Quinetic     count     : 1             num_rows  : 1             rows      : 
len       : 79           data      : 21a4231b87c0f9dd4d68                    id        : 7047
action    : Release
codes     : {79}21a4231b87c0f9dd4d68

```

zuckschwerdt, 2022-12-05:
There are some hints on the web that ENER-J might be same rebrand as Quinetic and manufactured by Baydo IOT Technology Shenzhen Co. Ltd. They state to use some EBL 1.1 communication protocol, but also no details on that.

pazzernick, 2022-12-05:
Cheers, I will have a dig.

I'm tempted to use a 32bit value from offset 8 for the ID, and then get the other 10 buttons tested to verify behaviour.

As long as the ID is consistent, my requirements are met. Push vs Release is not really important.

I just need to de-bounce the multiple packages, and discard the short ones.

Does this sound like the right plan?

zuckschwerdt, 2022-12-05:
Sounds good. As long as we are missing most of the protocol and can't figure out a checksum this is best left as flex decoder conf -- but PR that conf if you like.

re debounce: https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_custom.py can be a starting point.

pazzernick, 2022-12-05:
I'm happy to keep as flex-decoder, based on the limited info about the protocol.

But... my reasoning for writing up a device file ( **./src/devices/quinetic.c** ) was so that I could inspect the payload. Sometimes it's too short and doesn't contain the full ID, e.g. on the dirty 4th burst from button.

Perhaps I'm missing something: using `bits>=75` within the flex config does not discard packages with data lengths less than 75. Because... that would solve the issue of small/partial packages.

Thanks for all the help guys.

pazzernick, 2022-12-05:
I've tested a few flex config options to try to reduce duplicates.

I'm starting to get the feeling that if the 2nd/3rd/4th burst are picked up (based on gain setting), that the de-bouncing needs to be done outside of rtl_433, because they are separate inbound signals.

pazzernick, 2022-12-08:
Yep, I follow.

Now that I've got a steady gain/pulse configuration, I've been able to properly inspect the package data.

* `bits>=120` is now being used in my flex-config with success.
* There is always 1 row of data in the FSK package. Thus, this setting helps to filter out partials.

I won't file a PR yet, because I would like to try some further improvements... e.g. checksum

Either way, this is the working configuration: [Quinetic-QUGDMK.conf](https://github.com/pazzernick/rtl_433/blob/master/conf/Quinetic-QUGDMK.conf)

I don't need any specific help at this stage, unless you can offer any improvements, suggestions, etc.

Many thanks!

pazzernick, 2022-12-08:
This is the latest [bitbench data](https://triq.net/bitbench#c=%7B125%7D555555555421a423b6900155994d6900%5BPUSH%20%3A%20EBM%201%2020220618%201%20263%5D&c=%7B126%7D555555555421a423b690c09cf44d6900%5BRELEASE%20%3A%20EBM%201%2020220618%201%20263%5D&c=%7B126%7D555555555421a4231b870130b04d6900%5BPUSH%20%3A%20EBM%201%2020220618%201%20272%5D&c=%7B125%7D555555555421a4231b87c0f9dd4d6900%5BRELEASE%20%3A%20EBM%201%2020220618%201%20272%5D&c=%7B125%7D555555555421a4231e8701db404d6900%5BPUSH%20%3A%20EBM%201%2020220618%201%20284%5D&c=%7B123%7D555555555421a4231e87c0122d4d690%5BRELEASE%20%3A%20EBM%201%2020220618%201%20284%5D&c=%7B125%7D555555555421a423718800be1226b480%5BPUSH%20%3A%20EBM%201%2020220618%201%20360%5D&c=%7B124%7D555555555421a4237188c0b5494d692%5BRELEASE%20%3A%20EBM%201%2020220618%201%20360%5D&f=8h%208h%208h%208d%208d%208h8h%208h8h%208h&a=Preamble&m=555421&cw=4)

There are one or two bad-eggs within the Quinetic-QUGDMK.conf samples.

craigstory93, 2023-01-02:
Hi, I've tried using your config as above but I'm using a slightly different model than you, I have 1 / 2 / 3 button switches as link (https://www.tlc-direct.co.uk/Products/QUWS2B.html) it is receiving the signals and the ID of the switch (most of the time, sometimes throws a random ID) which is fine for the single switches as you mentioned above but was wondering did you get any further with the decoding data so i could try to use it for multiple buttons on the switch?

Thanks

pazzernick, 2023-01-03:
Hey...

I got testing again.... and also captured codes coming from other Quinetic hardware like smart-plugs.

I need more packet samples - so the three button switch data would be great if you can supply it!

* Please use the decoder below in the existing config file, and comment-out the 'Quinetic-GUGDMK' decoder.
* If you run live into the shell you will hopefully see 1-3 packets for a button push, and 1-3 packets for a release.
* I usually hit enter a few times before releasing the button, to separate the release packets easily.

Please dump the data into this [bit-bucket](https://triq.net/bitbench#c=%5BCODE%20LINES%20DO%20NOT%20INCLUDE%20preamble%3Daaaaaaaaabde%20%3A%20already%20inverted%20%3A%20min-bit-length%20120%5D&c=%5BGROUP%3A%20MULTIPLE%20BUTTON%20QUINETIC%20SWITCHES%5D&c=&c=5bdce3b4fe19e6b296fc%5BB1%20PUSH%5D&c=5bdce3b4fe19e6b296fc%5BB1%20PUSH%5D&c=5bdce3b43fd08bb296e%5BB1%20RELEASE%5D&c=5bdce3b43fd08bb296e%5BB1%20RELEASE%5D&f=COMMON%3A%208h%208h%20ID%3A%208h%208h%20ACTION%3A%208h%20OTHER%3A%208h%208h%208h%208h%208h&cw=4), then copy the new link and provide it here. Hopefully the codes appear when you push and release on each button.

```
decoder {
    name=QDebug,
    modulation=FSK_PCM,
    short=10,
    long=10,
    reset=200,
    bits>=120,
    invert,
    unique,
    preamble=aaaaaaaaabde 
}
```

craigs21, 2023-01-03:
Hi,

Below is the 2 button switch, just working on the 3 button switch now.

[Bit-bucket](https://triq.net/bitbench#c=%5BCODE%20LINES%20DO%20NOT%20INCLUDE%20preamble%3Daaaaaaaaabde%20%3A%20already%20inverted%20%3A%20min-bit-length%20120%5D&c=%5BGROUP%3A%20MULTIPLE%20BUTTON%20QUINETIC%20SWITCHES%5D&c=&c=5bdcdbb1fe8a17af0ae%20%5BB1%20PUSH%5D&c=5bdcdbb1fe8a17af0ae%20%5BB1%20PUSH%5D&c=5bdcdbb1fe8a17af0ac%20%5BB1%20PUSH%5D&c=5bdcdbb13f437aaf0ae%20%5BB1%20RELEASE%5D&c=5bdcdbb13f437aaf0ac%20%5BB1%20RELEASE%5D&c=5bdcdbb13f437aaf0ac%20%5BB1%20RELEASE%5D&c=&c=5bdcdbb1fdba74af0ac%20%5BB2%20PUSH%5D&c=5bdcdbb1fdba74af0ae%20%5BB2%20PUSH%5D&c=5bdcdbb13f437aaf0ac%20%5BB1%20RELEASE%5D&c=5bdcdbb13f437aaf0ac%20%5BB1%20RELEASE%5D&c=5bdcdbb13f437aaf0ae%20%5BB1%20RELEASE%5D&f=COMMON%3A%208h%208h%20ID%3A%208h%208h%20ACTION%3A%208h%20OTHER%3A%208h%208h%208h%208h%208h&cw=4)

pazzernick, 2023-01-03:
> Hi,
>
> Below is the 2 button switch, just working on the 3 button switch now. [...]

Can you double check it, because **B1 RELEASE** shows up more than expected.

craigs21, 2023-01-03:
Hi, 

below is the 3 button switch, [Bit-bucket](https://triq.net/bitbench#c=%5BCODE%20LINES%20DO%20NOT%20INCLUDE%20preamble%3Daaaaaaaaabde%20%3A%20already%20inverted%20%3A%20min-bit-length%20120%5D&c=%5BGROUP%3A%20MULTIPLE%20BUTTON%20QUINETIC%20SWITCHES%5D&c=&c=5bdc2a07feeadeb4503%20%5BB1%20PUSH%5D&c=5bdc2a07feeadeb4502%20%5BB1%20PUSH%5D&c=5bdc2a07feeadeb4502%20%5BB1%20PUSH%5D&c=5bdc2a073f23b3b4503%20%5BB1%20RELEASE%5D&c=5bdc2a073f23b3b4500%20%5BB1%20RELEASE%5D&c=5bdc2a073f23b3b4502%20%5BB1%20RELEASE%5D&c=5bdc2a073f23b3b4500%20%5BB1%20RELEASE%5D&c=5bdc2a073f23b3b4503%20%5BB1%20RELEASE%5D&c=5bdc2a073f23b3b4500%20%5BB1%20RELEASE%5D&c=&c=&c=5bdc2a07fbba7bb4503%20%5BB2%20PUSH%5D&c=5bdc2a07fbba7bb4502%20%5BB2%20PUSH%5D&c=5bdc2a07fbba7bb4503%20%5BB2%20PUSH%5D&c=5bdc2a073f23b3b4502%20%5BB2%20RELEASE%5D&c=5bdc2a073f23b3b4500%20%5BB2%20RELEASE%5D&c=5bdc2a073f23b3b4502%20%5BB2%20RELEASE%5D&c=5bdc2a073f23b3b4502%20%5BB2%20RELEASE%5D&c=5bdc2a073f23b3b4503%20%5BB2%20RELEASE%5D&c=&c=&c=5bdc2a07fddabdb4502%20%5BB3%20PUSH%5D&c=5bdc2a07fddabdb4503%20%5BB3%20PUSH%5D&c=5bdc2a07fddabdb4503%20%5BB3%20PUSH%5D&c=5bdc2a073f23b3b4500%20%5BB3%20RELEASE%5D&c=5bdc2a073f23b3b4503%20%5BB3%20RELEASE%5D&c=5bdc2a073f23b3b4503%20%5BB3%20RELEASE%5D&c=5bdc2a073f23b3b4500%20%5BB3%20RELEASE%5D&c=5bdc2a073f23b3b4500%20%5BB3%20RELEASE%5D&c=5bdc2a073f23b3b4500%20%5BB3%20RELEASE%5D&f=COMMON%3A%208h%208h%20ID%3A%208h%208h%20ACTION%3A%208h%20OTHER%3A%208h%208h%208h%208h%208h&cw=4)

Sorry labels was wrong on the 2 button switch, see corrected [bit-bucket](https://triq.net/bitbench#c=%5BCODE%20LINES%20DO%20NOT%20INCLUDE%20preamble%3Daaaaaaaaabde%20%3A%20already%20inverted%20%3A%20min-bit-length%20120%5D&c=%5BGROUP%3A%20MULTIPLE%20BUTTON%20QUINETIC%20SWITCHES%5D&c=&c=5bdcdbb1fe8a17af0ae%20%5BB1%20PUSH%5D&c=5bdcdbb1fe8a17af0ae%20%5BB1%20PUSH%5D&c=5bdcdbb1fe8a17af0ac%20%5BB1%20PUSH%5D&c=5bdcdbb13f437aaf0ae%20%5BB1%20RELEASE%5D&c=5bdcdbb13f437aaf0ac%20%5BB1%20RELEASE%5D&c=5bdcdbb13f437aaf0ac%20%5BB1%20RELEASE%5D&c=&c=5bdcdbb1fdba74af0ac%20%5BB2%20PUSH%5D&c=5bdcdbb1fdba74af0ae%20%5BB2%20PUSH%5D&c=5bdcdbb13f437aaf0ac%20%5BB2%20RELEASE%5D&c=5bdcdbb13f437aaf0ac%20%5BB2%20RELEASE%5D&c=5bdcdbb13f437aaf0ae%20%5BB2%20RELEASE%5D&f=COMMON%3A%208h%208h%20ID%3A%208h%208h%20ACTION%3A%208h%20OTHER%3A%208h%208h%208h%208h%208h&cw=4)

pazzernick, 2023-01-03:
Awesome, thanks for these. They help a lot.

Do you have any more switches, because I want to confirm if the model is baked into the codes. Or, just 2 switches on hand?

I've been including the last 4 digits of the switch serial in my notes, to separate the switches out.

TIA

craigs21, 2023-01-03:
I only have the same model of switches, I have about 10 of them split across 1 / 2 / 3 button variety 

The 3 button switch is on the wall but the back of the two button has the following info on the barcode sticker

ES2K2202207200477    244E000202

Then printed onto the plastic itself is 

QUWS2B CW473648

The one button switch data is as below if that helps as well

a423c72e04d70650f500 PUSH
a423c72e04d70650f500 PUSH
a423c72e04d70650f500 PUSH
a423c72ec04ece50f50 release
a423c72ec04ece50ea0 release

Let me know if you need any more.

Thanks

pazzernick, 2023-01-03:
Super, I could do with duplicates for the 1, 2 and 3 button variety... which almost means every switch (sorry!)

If you want to start with samples for the 1-button switches you have, let's confirm if they all have a similar code and model string. Guessing you have something like this? [QU WS1B](https://www.tlc-direct.co.uk/Products/QUWS1B.html)

At this juncture, I think I've got the method for extracting the individual button code, so I will probably place that into a channel-number which will enable you to determine which button is pushed down.

craigs21, 2023-01-03:
Hi,

Yes the 1 button is QU WS1B

Here is another 1 button switch data

a423152c04ba9050f50 - PUSH
a423152c04ba9050f500 - PUSH
a423152c04ba9050f50 - PUSH
a423152cc0235850f50 - RELEASE
a423152cc0235850f500 - RELEASE

2 Button  which is a QU WS2B

a423a65701993950f50 - B1 PUSH
a423a65701993950f50 - B1 PUSH
a423a65701993950f50 - B1 PUSH
a423a657c0505450f500 - B1 RELEASE
a423a657c0505450f500 - B1 RELEASE
a423a657c0505450f500 - B1 RELEASE
a423a65702a95a50f50 - B2 PUSH
a423a657c0505450f50 - B2 RELEASE
a423a657c0505450f50 - B2 RELEASE
a423a657c0505450f500 - B2 RELEASE

for the 3 button switch I only have one of these.

hope this helps.

craigs21, 2023-01-03:
I just noticed on MQTT that I was getting a few other ID's while gathering data for these pushes. Most of the button pushes seem to decode the ID correctly but there was a few random one's they was appearing in the data on MQTT, unsure if this is just occurring from pressing / releasing too quickly etc?

a423fd97c0e83f4c480
a423fd97c0e83f4c4800
a423fd97c0e83f4c480
a423fd9703012098800
a423fd970301104c480
a423fd970301104c4800
a4222a598046b0a1ea0
a4238e5c09ae0ca1d40
88474caf80a0a8a1ea0

pazzernick, 2023-01-03:
Yep, there are a few possible reasons for the extra ID's showing up.

* Reason 1: I don't know how to calc the checksum (if there is one) so some packets may be malformed.
* Reason 2: If a switch is already 'paired' with some hardware, e.g. Smart Plug, Lighting Transformer, the end-device is likely to transmit some RF too. I know that Smart Plugs transmit RF when initially booting.

I've updated the [config file](https://github.com/pazzernick/rtl_433/blob/master/conf/Quinetic-QUGDMK.conf) - it should now allow you to handle the MQTT 'actioncode' of 251 - 254, for each button within the 1/2/3 switches.

My MQTT devices tree is pretty clean, and there are very few extra IDs showing up. That said, I have no paired hardware in the house right now, just 12 single switches. So very little noise...

Let me know if that config file works for your setup.

Are you planning to use the RF buttons only with rtl_433, MQTT and some Automation platform. Or, do you still have RF pairing direct with Quinetic HW?

craigs21, 2023-01-03:
Hi,

I'll give that config a test and let you know the results, I'm using Sonoff switches with Node-Red on Home Assistant fed by MQTT from the RTL_443.

I don't actually have any of the Smart Plugs or Transformers.

pazzernick, 2023-01-04:
I've renamed the file and improved the decoder. [Quinetic-Electrical.conf](https://github.com/pazzernick/rtl_433/blob/master/conf/Quinetic-Electrical.conf)

There are still a few false-fires, e.g. the wrong ID scraped sometimes. I'm not sure how to checksum the data, so I'm just relying upon the consistent ID that is seen, combined with 'actioncode' value.

When I hold a button, my **MQTT Explorer** allows me to see the 'events' item from rtl_433, showing the button (actioncode) value being pushed. Hopefully that state can be used in HASS. That was my plan, based on several MQTT sensors targeting specific topics (numeric ID of switch)

blutack, 2023-06-30:
If it's of any use, this video documents the checksum details and possibly has some more information on the protocol:
https://www.youtube.com/watch?v=tz_F4Tjhap0

gdt, 2023-09-29:
Where are we on this issue?  It seems like one could construct PRs that would add a decoder and then we could close it, but would appreciate someone summarizing status and next steps.

craigs21, 2023-10-29:
I have multiple of these switches so can help anyone who needs test data.

pazzernick, 2024-11-22:
> I have multiple of these switches so can help anyone who needs test data.

Sorry for the (long) delay. I've used the CRC method detailed (thanks Cameron + others) and have new code under test using that CRC check.

If still viable, can you grab fresh codes for 3-Button Switch using the flex-config file [here](https://github.com/pazzernick/rtl_433_tests/tree/master/tests/quinetic/switch) ?

The only sample needed is the 3-Button QUWS3x, to verify the action codes on various buttons.

1. Download **quinetic_switch.conf** to your box
2. Make a copy to new file: quinetic_switch_minimal.conf
3. Remove lines 18-23 ( remove preamble + all the ***get=*** settings )
4. Capture live transmissions: ```rtl_433 -g 37 -R 0 -c ./quinetic_switch_minimal.conf```
5. Copy the full codes for a press and release (please keep duplicates) into [bit-bucket](https://triq.net/bitbench#c=%5BModel%3A%20QUGDMK%2C%20Button%20Count%3A%201%5D&c=%7B127%7D2aaaaaaaaa10d2118dc380985826b480%20%5BPUSH%20B1%5D&c=%7B126%7D2aaaaaaaaa10d2118dc380985826b484%20%5BPUSH%20B1%5D&c=%7B128%7D1555555555086908c6e1c04c2c135a40%20%5BPUSH%20B1%5D&c=%7B125%7D555555555421a4231b870130b04d6900%20%5BPUSH%20B1%5D&c=%7B128%7D0aaaaaaaaa8434846370f81f3ba9ad20%20%5BREL%20B1%5D&c=%7B128%7D1555555555086908c6e1f03e77535a40%20%5BREL%20B1%5D&c=%7B118%7Dd555555555086908c6e1f03e775358%20%20%20%5BREL%20B1%5D&c=&c=%5BModel%3A%20X%2C%20Button%20Count%3A%20Y%5D&f=SYNC%3A16h%20ID%3A16h%20ACTION%3Ad%20CRC%3A16h%20OTHER%3A%20hh%20hh%20hh%20hh&z=1&a=Sync&m=a423&cw=4)

Reasoning for getting new full-length codes: changed frequency, changed sample_rate, no longer using 'invert' option, etc.

closed 2024-12-13 by zuckschwerdt