merbanan/rtl_433 #2274: Support for WH32 Indoor coded

sdalu, 2022-12-12:
Device[1] is not decoded by the `fineoffset.c` it is certainly a variation of the Fineoffset WH32 model.

Packet retrieved using `rtl_433 -f 868.35M -A  -X 'n=Ecowitt,prio=9,m=FSK_PCM,s=58,l=58,r=3000,preamble=aaaa2dd4'`
~~~
{98}e6f24c3626961604100000000
{98}e6f2503426961804100000000
{98}e6f2503426961804100000000
{98}e6f2503426951704100000000
{98}e6f2503426951704100000000
{98}e6f2513426971a04100000000
{98}e6f2553226820704100000000
~~~

Seems to decode well (at least for temperature, humidiy, pressure) with: `TYPE:4h ID:8d FLAGS:2b TEMP_C:10d HUM:8d HPA:16d CHK:8h`

[1] https://www.ecowitt.com/shop/goodsDetail/193#

zuckschwerdt, 2022-12-12:
What is the packet length without trimming the `preamble`?
There are some checks on that here, you might get it to work by tweaking that
https://github.com/merbanan/rtl_433/blob/dc95819b3d59aa73ebd59d6e38f1df5cec8e3500/src/devices/fineoffset.c#L476-L480

sdalu, 2022-12-12:
Looks like 169 bits :
~~~
{169}ab5555555555545ba9cde4ac624d000a08200000000
{169}ab5555555555545ba9cde4ac624d020c08200000000
~~~

If I remove package validation and force `type = 32`
https://github.com/merbanan/rtl_433/blob/dc95819b3d59aa73ebd59d6e38f1df5cec8e3500/src/devices/fineoffset.c#L475-L484

and bitsearch at index 0 instead of 100
https://github.com/merbanan/rtl_433/blob/dc95819b3d59aa73ebd59d6e38f1df5cec8e3500/src/devices/fineoffset.c#L489

I got:

~~~
time      : 2022-12-12 15:43:43
model     : Fineoffset-WH32B                       ID        : 111
Battery   : 1            Temperature: 19.7 C       Humidity  : 49 %          Pressure  : 985.3 hPa
Integrity : CRC
~~~

Which is correct (well, no idea about the ID field). Model on the backside of the device says `WN32AP`

zuckschwerdt, 2022-12-12:
Thanks. The branching there isn't ideal anyway, I'll need to check if we can just run all 3 decoders always.

sdalu, 2023-09-30:
@gdt kind of the  `WH32 indoor`[1] is the `WH32` with pressure sensor added.
1: https://shop.ecowitt.com/products/wh32-indoor

gdt, 2023-10-01:
Can you test with current git master?   It seems like someone needs to extend the decoder, assuming the format is longer.  (I am trying to push people to create PRs from issues so we can have improved code and close the issues.)

Mart124, 2024-01-24:
Also bought this [WH32P](https://shop.ecowitt.com/products/wh32-indoor), which is not recognized (model at the backside also says `WN32AP`).
All my other [WH31](https://shop.ecowitt.com/products/wh31) work flawlessly though.
Any progress perhaps on this fix ?
May we help somehow ?
Thank you very much 👍

zuckschwerdt, 2024-01-24:
This was partially addressed with #2303. We need a sample to finally add WH32P support. https://triq.org/rtl_433/ANALYZE.html
Check with https://triq.org/pdv/ and upload as zip here.

Mart124, 2024-01-24:
Many thanks for your reply Christian :+1:
Let me then check the analysis process an report back !

Mart124, 2024-01-25:
So !
Here are 4 samples : [2274.zip](https://github.com/merbanan/rtl_433/files/14057836/2274.zip)

At this time, sensor values were :
20.1°C / 60% / 1029.0hPa
20.1°C / 66% / 1028.9hPa
20.2°C / 67% / 1028.9hPa
20.3°C / 73% / 1029.0hPa

Of course feel free if needed !
Thank you 👍

Mart124, 2024-01-29:
Just to be sure, are my samples above clean enough to go further ?
Of course do not hesitate if you need additional ones !
Thank you again 👍

zuckschwerdt, 2024-01-29:
Thanks! The samples work. If you look at the spectrogram you can see that two samples (001 & 002) have a weaker signal nearly colliding. There is a rogue carrier at 868 MHz exactly -- you might want to tune up (`-f 868.5M`) to get rid of that.
Overall the signals are clipping somewhat ("too loud"), but rtl_433 can still handle that.

Mart124, 2024-01-29:
Many thanks for your feedback @zuckschwerdt, good news 👍 
I had the sensor at about 1 or 2cm from my receiver (without its antenna) to get these samples, certainly the cause of the "too loud" signals :)
Feel free if you want / need additional samples with `-f 868.5M`.

Mart124, 2024-01-30:
Did not initially gave a try, but I confirm that last nightly build gives proper values :

```
C:\...>rtl_433.exe -S known -f 868.34M -s 250k
rtl_433 version -128-NOTFOUND branch  at 202312291305 inputs file rtl_tcp RTL-SDR SoapySDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-30 21:40:29
model     : Fineoffset-WH32B                       ID        : 157
Battery   : 1            Temperature: 19.7 C       Humidity  : 56 %          Pressure  : 1032.2 hPa
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-30 21:40:29
model     : Fineoffset-WH32B                       ID        : 157
Battery   : 1            Temperature: 19.7 C       Humidity  : 56 %          Pressure  : 1032.2 hPa
Integrity : CRC
*** Saving signal to file g001_868.34M_250k.cu8 (46280 samples, 131072 bytes)
```
[g001_868.34M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/14104034/g001_868.34M_250k.cu8.zip)

Perhaps we could then close this issue, not sure if there's still something to do here 🤔 (but the sensor name, `WH32B` vs `WH32P`, not sure it is important though).

Thank you again !

gdt, 2024-06-04:
This seems ready to close.   Yell if not.

closed 2024-06-04 by gdt