merbanan/rtl_433 #2276: bresser-6-in-1 reports wrong rain_mm

gaaf, 2022-12-14:
Since a few weeks I get occasional messages with a completely wrong (way too high) rain_mm. These are all in a frame with temperature_C and humidity, which shouldn't happen IIRC.

The presence of the rain_mm field depends on the value of temperature_C. Also, when it is present,  rain_mm is varying in step with temperature_C. The rain_mm value in the "wind" frames stay constant (as it should, there is no rain).

When a wrong rain_mm field is present, the battery_ok field is missing.

I suspect some bits of the temperature_C and/or battery_ok field(s) are incorrectly interpreted as rain_mm.

Wrong rain_mm field present when temperature_C = -8.300, -8.400, -9.100, -9.200, -9.300, -9.400
No rain_mm field when temperature_C = -8.500, -8.600, -8.700, -8.800, -8.900, -9.000, -9.500, -9.600

In the lines below, 1233.200 is the correct value for rain_mm.
```
{"time" : "2022-12-14 08:38:11", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "temperature_C" : -8.300, "humidity" : 95, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "rain_mm" : 74857.000, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}

{"time" : "2022-12-14 08:46:11", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "temperature_C" : -8.400, "humidity" : 95, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "rain_mm" : 74957.000, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}

{"time" : "2022-12-14 08:47:47", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "battery_ok" : 1, "temperature_C" : -8.500, "humidity" : 95, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}

{"time" : "2022-12-14 08:49:23", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "battery_ok" : 1, "temperature_C" : -8.600, "humidity" : 95, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}

{"time" : "2022-12-14 08:51:23", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "battery_ok" : 1, "temperature_C" : -8.700, "humidity" : 94, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}

{"time" : "2022-12-14 08:52:35", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "battery_ok" : 1, "temperature_C" : -8.800, "humidity" : 94, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}

{"time" : "2022-12-14 08:54:35", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "battery_ok" : 1, "temperature_C" : -8.900, "humidity" : 94, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}

{"time" : "2022-12-14 08:55:23", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "battery_ok" : 1, "temperature_C" : -9.000, "humidity" : 94, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}

{"time" : "2022-12-14 09:06:23", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "rain_mm" : 1233.200, "startup" : 1, "flags" : 1, "mic" : "CRC"}
{"time" : "2022-12-14 09:06:35", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "temperature_C" : -9.400, "humidity" : 94, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "rain_mm" : 75957.102, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}
{"time" : "2022-12-14 09:06:47", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "rain_mm" : 1233.200, "startup" : 1, "flags" : 1, "mic" : "CRC"}
{"time" : "2022-12-14 09:06:59", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "temperature_C" : -9.400, "humidity" : 94, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "rain_mm" : 75957.102, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}
{"time" : "2022-12-14 09:07:11", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "rain_mm" : 1233.200, "startup" : 1, "flags" : 1, "mic" : "CRC"}
{"time" : "2022-12-14 09:07:23", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "battery_ok" : 1, "temperature_C" : -9.500, "humidity" : 94, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}

{"time" : "2022-12-14 09:32:59", "model" : "Bresser-6in1", "id" : 312482654, "channel" : 0, "battery_ok" : 1, "temperature_C" : -9.600, "humidity" : 94, "sensor_type" : 1, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 112, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}
```

Version:
```
$ rtl_433 -V
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
```

Invocation:
```
$ rtl_433 -f 868.3M -s 256k -R 172 -F json
```

zuckschwerdt, 2022-12-14:
Thanks! Duplicate of #2184? Can you verify?

gaaf, 2022-12-14:
Could be, it seems similar. 

My Bresser is also actually a 5-in-1, but only work(ed) with the 6-in-1 decoder.

I did not have this problem with the previous cold weather a year ago, so it is a regression.

MacH-21, 2022-12-14:
@zuckschwerdt 
Can you please check my post on #1214
This old chestnut, Rain is not valid when flags = 0

MadMacMad, 2023-01-18:
hello,
have the same issue...
rain has wrong value every second message....

gdt, 2023-09-26:
Dup of #1214.  Please retest with current git master and post there.

closed 2023-09-26 by gdt