merbanan/rtl_433 #2277: Support for Reolink doorbell

sdalu, 2022-12-14:
Decoding is done with: m=OOK_PWM,s=352,l=956,r=9360,g=872,t=242,y=0 on the 433MHz
Pressing the doorbell seems to send 2 packets, one with 49 rows, another with 33 rows.

Data seems to be decoded with akhan_100F14, but I don't know if it is a hasard.

My doorbell send  payload: 59 7f 4d
If other can share there payload, perhaps we can identified fields

Model is: https://reolink.com/fr/product/reolink-video-doorbell/

~~~
Detected OOK package	2022-12-14 15:52:27
Analyzing pulses...
Total count: 1200,  width: 1862.08 ms		(465519 S)
Pulse width distribution:
 [ 0] count:  768,  width:  352 us [344;364]	(  88 S)
 [ 1] count:  432,  width:  956 us [948;968]	( 239 S)
Gap width distribution:
 [ 0] count:   48,  width: 9348 us [9348;9356]	(2337 S)
 [ 1] count:  432,  width:  252 us [244;264]	(  63 S)
 [ 2] count:  719,  width:  860 us [856;868]	( 215 S)
Pulse period distribution:
 [ 0] count:   48,  width: 9700 us [9696;9716]	(2425 S)
 [ 1] count: 1151,  width: 1212 us [1204;1216]	( 303 S)
Pulse timing distribution:
 [ 0] count:  768,  width:  352 us [344;364]	(  88 S)
 [ 1] count: 1152,  width:  896 us [856;968]	( 224 S)
 [ 2] count:   48,  width: 9348 us [9348;9356]	(2337 S)
 [ 3] count:  432,  width:  252 us [244;264]	(  63 S)
Level estimates [high, low]:  16003,      1
RSSI: -0.1 dB SNR: 42.0 dB Noise: -42.1 dB
Frequency offsets [F1, F2]:     710,      0	(+2.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB00B040101600380248400FC8255+AAB023042F01600380248400FC9381938181939381938181818181818193819393818193818255+AAB022040101600380248400FC93819381819393819381818181818181938193938181938155
Attempting demodulation... short_width: 352, long_width: 956, reset_limit: 9360, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=352,l=956,r=9360,g=872,t=242,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 49 
[00] { 1} 80          : 1
[01] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[02] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[03] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[04] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[05] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[06] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[07] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[08] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[09] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[10] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[11] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[12] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[13] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[14] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[15] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[16] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[17] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[18] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[19] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[20] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[21] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[22] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[23] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[24] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[25] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[26] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[27] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[28] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[29] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[30] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[31] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[32] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[33] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[34] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[35] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[36] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[37] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[38] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[39] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[40] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[41] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[42] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[43] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[44] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[45] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[46] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[47] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[48] {24} 59 7f 4d    : 01011001 01111111 01001101 

Detected OOK package	2022-12-14 15:52:29
Analyzing pulses...
Total count:  800,  width: 1241.09 ms		(310273 S)
Pulse width distribution:
 [ 0] count:  512,  width:  348 us [344;360]	(  87 S)
 [ 1] count:  288,  width:  956 us [948;968]	( 239 S)
Gap width distribution:
 [ 0] count:   32,  width: 9348 us [9344;9360]	(2337 S)
 [ 1] count:  288,  width:  252 us [244;264]	(  63 S)
 [ 2] count:  479,  width:  860 us [856;864]	( 215 S)
Pulse period distribution:
 [ 0] count:   32,  width: 9700 us [9696;9712]	(2425 S)
 [ 1] count:  767,  width: 1212 us [1204;1220]	( 303 S)
Pulse timing distribution:
 [ 0] count:  512,  width:  348 us [344;360]	(  87 S)
 [ 1] count:  767,  width:  896 us [856;968]	( 224 S)
 [ 2] count:   33,  width: 9368 us [9344;10004]	(2342 S)
 [ 3] count:  288,  width:  252 us [244;264]	(  63 S)
Level estimates [high, low]:  15947,      1
RSSI: -0.1 dB SNR: 42.0 dB Noise: -42.1 dB
Frequency offsets [F1, F2]:     619,      0	(+2.4 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB00B0401015C0380249800FC8255+AAB023041F015C0380249800FC9381938181939381938181818181818193819393818193818255+AAB0220401015C0380249800FC93819381819393819381818181818181938193938181938255
Attempting demodulation... short_width: 348, long_width: 956, reset_limit: 9364, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=348,l=956,r=9364,g=868,t=243,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 33 
[00] { 1} 80          : 1
[01] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[02] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[03] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[04] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[05] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[06] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[07] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[08] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[09] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[10] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[11] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[12] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[13] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[14] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[15] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[16] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[17] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[18] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[19] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[20] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[21] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[22] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[23] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[24] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[25] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[26] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[27] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[28] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[29] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[30] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[31] {25} 59 7f 4d 80 : 01011001 01111111 01001101 1
[32] {24} 59 7f 4d    : 01011001 01111111 01001101 
~~~

zuckschwerdt, 2022-12-14:
That's a EV1527 transmission. 24 bit plus 1 sync.
A packet is those 24(25) bit, and all the 82 packet make a transmission.

Very likely it's always just one fixed random number per bell. No scheme or protocol.

sdalu, 2022-12-14:
Proposed decoder:
~~~
decoder {
    name        = Reolink-Doorbell,
    modulation  = OOK_PWM,
    short       = 352,
    long        = 956,
    reset       = 9360,
    gap         = 872,
    tolerance   = 242,
    sync        = 0,
    bits       >= 24,
    rows       >= 30,
    unique,
    get         = @0:{24}:id,
}
~~~

zuckschwerdt, 2022-12-14:
```
Gap distribution:
 [ 0] count:   32,  width: 9348 us
...
 [ 2] count:  479,  width:  860 us
```

860 us -> +~10% -> ~950 us -> gap = 950
9348 us -> +~5% -> ~9800 us -> reset = 9800

sdalu, 2022-12-14:
* Is there something to add, not to trigger Akhan-100F14 and Smoke-GS558, if this decoder match ?
* should I create a pull request ?

zuckschwerdt, 2022-12-14:
Both Akhan-100F14 (P. 48) and Smoke-GS558 (P. 86) should be disabled really. We might exclude them in the future.
For now use: `-R -48 -R -86` or if you only want your Reolink anyway: `-R 0`

segdy, 2023-03-10:
I would be extremely excited to receive my Reolink doorbell signal as well.

What is the latest status of the discussion? Seems fairly straightforward to integrate ?

zuckschwerdt, 2023-03-10:
The conf has been added now: https://github.com/merbanan/rtl_433/blob/master/conf/Reolink-doorbell.conf

closed 2023-03-10 by zuckschwerdt

sdalu, 2023-03-10:
@zuckschwerdt, following your advice the following have been changed.
~~~
    reset       = 9800,
    gap         = 950,
~~~
Which is not reflected in the file  you commited.

JanLoebel, 2023-04-21:
Hi there,

sorry I'm new to this project, followed the guides and got everything up and running. If I just start `rtl433` I don't see any events happening if I press the "Reolink Wifi Doorbell" but the chime is ringing.

If I start `rtl433 -A` I get the following messages (a lot):

```
pi@raspberrypi:~/rtl_433 $ rtl_433 -A
rtl_433 version nightly-26-g408d2fdf branch master at 202304191141 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
[Protocols] Registered 211 out of 244 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243-244 ]
Found Fitipower FC0012 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
Detected OOK package	2023-04-21 12:15:08
Analyzing pulses...
Total count:   88,  width: 224.98 ms		(56244 S)
Pulse width distribution:
 [ 0] count:   39,  width:  616 us [528;712]	( 154 S)
 [ 1] count:   34,  width: 1796 us [1684;1920]	( 449 S)
 [ 2] count:   12,  width: 2984 us [2896;3120]	( 746 S)
 [ 3] count:    1,  width:  488 us [488;488]	( 122 S)
 [ 4] count:    1,  width:   88 us [88;88]	(  22 S)
 [ 5] count:    1,  width:    0 us [0;0]	(   0 S)
Gap width distribution:
 [ 0] count:    6,  width: 9096 us [9012;9152]	(2274 S)
 [ 1] count:   79,  width:  616 us [504;736]	( 154 S)
 [ 2] count:    1,  width:   44 us [44;44]	(  11 S)
 [ 3] count:    1,  width:   64 us [64;64]	(  16 S)
Pulse period distribution:
 [ 0] count:    6,  width: 9700 us [9700;9704]	(2425 S)
 [ 1] count:   32,  width: 2424 us [2420;2432]	( 606 S)
 [ 2] count:   12,  width: 3636 us [3632;3640]	( 909 S)
 [ 3] count:   34,  width: 1212 us [1208;1220]	( 303 S)
 [ 4] count:    2,  width: 1800 us [1784;1816]	( 450 S)
 [ 5] count:    1,  width:  640 us [640;640]	( 160 S)
Pulse timing distribution:
 [ 0] count:  118,  width:  616 us [504;736]	( 154 S)
 [ 1] count:   34,  width: 1796 us [1684;1920]	( 449 S)
 [ 2] count:   12,  width: 2984 us [2896;3120]	( 746 S)
 [ 3] count:    1,  width:  488 us [488;488]	( 122 S)
 [ 4] count:    1,  width:   88 us [88;88]	(  22 S)
 [ 5] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 6] count:    6,  width: 9096 us [9012;9152]	(2274 S)
 [ 7] count:    1,  width:   44 us [44;44]	(  11 S)
 [ 8] count:    1,  width:   64 us [64;64]	(  16 S)
 [ 9] count:    1,  width:   20 us [20;20]	(   5 S)
Level estimates [high, low]:  13948,   2167
RSSI: -0.7 dB SNR: 8.1 dB Noise: -8.8 dB
Frequency offsets [F1, F2]:    -373,      0	(-1.4 kHz, +0.0 kHz)
Guessing modulation: No clue...

Detected OOK package	2023-04-21 12:15:08
Analyzing pulses...
Total count:  268,  width: 682.29 ms		(170573 S)
Pulse width distribution:
 [ 0] count:  103,  width: 1788 us [1664;1956]	( 447 S)
 [ 1] count:  124,  width:  616 us [524;728]	( 154 S)
 [ 2] count:   34,  width: 2988 us [2856;3124]	( 747 S)
 [ 3] count:    1,  width: 2380 us [2380;2380]	( 595 S)
 [ 4] count:    2,  width: 1120 us [1096;1148]	( 280 S)
 [ 5] count:    2,  width:   44 us [40;48]	(  11 S)
 [ 6] count:    1,  width:  464 us [464;464]	( 116 S)
 [ 7] count:    1,  width:    0 us [0;0]	(   0 S)
Gap width distribution:
 [ 0] count:  238,  width:  616 us [496;760]	( 154 S)
[... 703 lines cut ...]
 [ 0] count:   32,  width:  636 us [524;748]	( 159 S)
 [ 1] count:    2,  width: 9108 us [9012;9204]	(2277 S)
 [ 2] count:    1,  width:  496 us [496;496]	( 124 S)
Pulse period distribution:
 [ 0] count:    5,  width: 3636 us [3636;3640]	( 909 S)
 [ 1] count:   14,  width: 2424 us [2420;2428]	( 606 S)
 [ 2] count:   14,  width: 1212 us [1208;1216]	( 303 S)
 [ 3] count:    2,  width: 9704 us [9704;9704]	(2426 S)
Pulse timing distribution:
 [ 0] count:    5,  width: 2984 us [2916;3092]	( 746 S)
 [ 1] count:   15,  width: 1772 us [1672;1848]	( 443 S)
 [ 2] count:   48,  width:  624 us [500;748]	( 156 S)
 [ 3] count:    2,  width: 9108 us [9012;9204]	(2277 S)
 [ 4] count:    1,  width:  496 us [496;496]	( 124 S)
 [ 5] count:    1,  width: 30924 us [30924;30924]	(7731 S)
Level estimates [high, low]:  15067,   3523
RSSI: -0.4 dB SNR: 6.3 dB Noise: -6.7 dB
Frequency offsets [F1, F2]:    -430,      0	(-1.6 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB01506010BA806EC0270239401F078CC829292A2A292A355+AAB01D06010BA806EC0270239401F078CC9282A2A292A2A292829292A2A292A355+AAB01C06010BA806EC0270239401F078CC9282A4A292A2A292829292A2A29555
Attempting demodulation... short_width: 600, long_width: 1772, reset_limit: 9208, sync_width: 2984
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=600,l=1772,r=9208,g=0,t=0,y=2984'
```

Is there something I did wrong or is it a different protocol for the EU / WIFI version?

zuckschwerdt, 2023-04-21:
You need to load the conf with e.g. `rtl_433 -c conf/Reolink-doorbell.conf`
But the timings look different here, grab a sample and upload as zip here. https://triq.org/rtl_433/ANALYZE.html

JanLoebel, 2023-04-21:
@zuckschwerdt Thanks, attached the zip of the captured signal.

[g001_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/11299078/g001_433.92M_250k.cu8.zip)

zuckschwerdt, 2023-04-21:
That signal is right on DC and clipping. Put more distance between antenna and sender and tune away from the signal, e.g. `-f 434.85M`

JanLoebel, 2023-04-24:
I've tested again with tuning away as suggested. That worked even with the distance from before (3 meters)
`rtl_433 -R 0 -c conf/Reolink-doorbell.conf -f 434.85M`. The timings than seems to be fine. Even if I increase the distance to around 9 meters without `-f 434.85M` it does not work. Strange..

segdy, 2023-04-30:
@sdalu and @zuckschwerdt Thank you for all the legwork in this!!

I got it up and running now but @sdalu , I am wondering, how do you translate this to Home Assistant? In MQTT I have now:

[image]

Fields are count, num_rows, len, data, id and I assume "data" is the most promising. How do you decode this and (if you use HA), to which device and how are you mapping it?

sdalu, 2023-05-02:
@segdy, I'm not using Home Assistant.  And @zuckschwerdt said: "Very likely it's always just one fixed random number per bell. No scheme or protocol."

gcormier, 2023-05-02:
Got this up and running as well, awesome work as always.

My ID is fixed as is my data when ringing.

commit 62a95188 (2023-03-10): Add decoder conf for Reolink doorbell (closes #2277)