merbanan/rtl_433 #2283: MQTT Client-ID not computer automatically

Giulianini, 2022-12-19:
[image]
[image]

gdt, 2022-12-19:
Please actually describe what you think is wrong.

Giulianini, 2022-12-19:
I have a raspberry with a radio dongle.
I'm using the configuration shown above and the client ID keeps staying unknown while I read in the doc that it's computed from the driver

merbanan, 2022-12-19:
Well we can not take responsibility for how your broker handles connections.

Giulianini, 2022-12-19:
I know, but it seems that rtl_433 is not setting the clientID. I'll continue troubleshooting. 
thanks for the fast response

Giulianini, 2022-12-19:
[image]

Fixed i'm a bit silly that I thought that `""` would not interfere with user and pass. 
I close the issue, if you want delete it.
Thanks

zuckschwerdt, 2022-12-19:
No worries. I would have expected quotes (`"..."`) to work too, but we don't support quotes :/

Careful though: `mqtt=...` is actually read as `mqtt ,=...` you likely want `mqtt://192.169.1.100:1883,user=mqtt_user,pass=...`