merbanan/rtl_433 #2284: [214]  EMOS E6016 weatherstation with DCF77

dartfrogdk, 2022-12-20:
[214]  EMOS E6016 weatherstation with DCF77

i got this weatherstation wind temp part not the display, i cant get rtl_433 to find it

RTL_433 works fine it picks TPMS sensors fine and a neighbours temperature sensor, does anyone know what could be wrong

on the sensor there is a TX button a 1-2-3 channel button and a WAVE button

dartfrogdk, 2022-12-20:
[image]

zuckschwerdt, 2022-12-20:
Try to check the band if the transmissions from the weather station are there.
https://triq.org/rtl_433/ANALYZE.html

If you find them but they are not decoded then upload a sample as zip here.

gdt, 2023-09-29:
@dartfrogdk If you want to work on this please feel free to gather some data and try to figure it out and ask for help.   I don't see any data in the issue, just a model number, so I'm closing as inactive issues just make it harder to understand what's going on.

closed 2023-09-29 by gdt

dartfrogdk, 2023-09-29:
i could not get it to work

Den fre. 29. sep. 2023 kl. 02.28 skrev Greg Troxel ***@***.***