merbanan/rtl_433 #2286: Error with Oregon UV meter UVR128

geschriever, 2022-12-21:
Hi,

Please can you investigate if something changed. I got this message not before?

RTL433: Unhandled sensor reading, please report: ({"time" : "2022-12-21 12:31:45", "model" : "Oregon-UVR128", "id" : 9, "uv" : 0, "battery_ok" : 1, "mod" : "ASK, "freq" : 433.910, "rssi" : -0.134, "snr" : 22.467, "noise" : -22.602})

Thanks in advance

zuckschwerdt, 2022-12-21:
That's a Domoticz message, right? Report it there.
For change history reference https://github.com/merbanan/rtl_433/commits/master/src/devices/oregon_scientific.c

geschriever, 2022-12-21:
Right, but is already solved by an update. Sorry to disturb you

zuckschwerdt, 2022-12-21:
No trouble at all. But we don't track other projects like Domoticz, unlikely we can help.

Other users might see this issue here and offers some help ;)