merbanan/rtl_433 #2292: battery_ok is 1 for low battery on [172]  Bresser Weather Center 6-in-1

level20peon, 2022-12-29:
I am reading aforementioned station with rtl-433 20.11 on a Raspberry PI using a RTL2832U Tuner via `sudo rtl_433 -f 868.3M`. The battery output states "1", even though the battery low indicator on the original base station is active. Another one of the attached sensors has full batteries right now, and it shows up as "1" in the rtl_433 output as well.

How can I get any information about that low battery via rtl_433?

zuckschwerdt, 2022-12-29:
IIRC there was an issue where we came up with a fix. Not sure, sorry. But I'll address the backlog over the next weeks hopefully.

level20peon, 2022-12-30:
I just realized that version 20.11 is 2 years old. So I built the current version from source, but now the battery output just states the opposite "0" in both cases (battery full, battery low).

btw: I think you were referring to [this](https://github.com/merbanan/rtl_433/issues/1214#issuecomment-1148849488), right? So this *should* have worked with the current source, right? Or do I still need to adjust something?

zuckschwerdt, 2022-12-30:
I guess I was thinking of #1993 but #2184 indicates the battery bit might be somewhere else.
Can you output the raw codes with good/low battery? This should work `-R 172:vv`

level20peon, 2022-12-30:
The sensor reading 20.5°C has the low battery indicator shown on the base station, whereas the sensor reading 8.2°C does not.

```
[bresser_6in1_decode]
codes     : {144}57b1183016b52900000000002050610000f2
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-30 10:22:38
model     : Bresser-6in1 id        : 183016b5
channel   : 1            Battery   : 0             Temperature: 20.5 C       Humidity  : 61            Sensor type: 2
Startup   : 1            Flags     : 0             Integrity : CRC
[pulse_slicer_pcm] Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A
codes     : {428}555555555516ea2bd88c180b5a948000000000102830800079000000e00000000000000000000000000000000000000000000000000
[bresser_6in1_decode]
codes     : {144}a83c18300cb618ffffff3308082478fff012
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-30 10:22:43
model     : Bresser-6in1 id        : 18300cb6
channel   : 0            Battery   : 0             Temperature: 8.2 C        Humidity  : 78            Sensor type: 1
Wind Gust : 0.0 m/s      Wind Speed: 0.0 m/s       Direction : 330           UV        : 0.0           Startup   : 1
Flags     : 0            Integrity : CRC
[pulse_slicer_pcm] Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A
codes     : {205}aaaaaaaaaa2dd4a83c18300cb618ffffff3308082478fff01200
```

zuckschwerdt, 2022-12-30:
Here is [a BitBench with those codes](https://triq.net/bitbench#c=555555555516ea2bd88c180b5a948000000000102830800079000000&c=aaaaaaaaaa2dd4a83c18300cb618ffffff3308082478fff01200&f=DIGEST%3A8h8h%20ID%3A8h8h8h8h%20STYPE%3A4h%20STARTUP%3A1b%20CH%3A3d%20WSPEED%3A~8h~4h%20~4h~8h%20WDIR%3A12h%20%3F4h%20TEMP%3A8h.4h%20%3F2b%20BATT%3A1b%20%3F1b%20HUM%3A8h%20UV%3F~12h%20%3F4h%20CHKSUM%3A8h&a=Preamble&m=aaaa2dd4&cw=4). Looks like the battery bit might be just before where we suspected it? Or maybe it's not just 0/1 but two bits?

MacH-21, 2022-12-30:
Looking at the BitBench codes, Battery readings look ok,

Battery status flag   = (msg[13] >> 2) & 1; (0=Lo, 1=Ok)
Temperature sign flag = (msg[13] >> 3) & 1; (0=Positive, 1=Negative)

Can you check your recent install on PI is correctly installed see #2257

level20peon, 2022-12-30:
> Here is [a BitBench with those [...]

I'm going to swap batteries on both stations so that the low one will be full afterwards and vice versa. Maybe this will help identifying those bits?

> Looking at the BitBench codes, Battery readings look ok,

Yet they don't show me if the battery low indicator is present or not!?

> Can you check your recent install on PI is correctly installed see #2257
```
$ rtl_433 -V
rtl_433 version 22.11-33-g083dcca4 branch master at 202212281345 
```

zuckschwerdt, 2022-12-30:
@MacH-21 currently we use (bit 0x2 not 0x4)
https://github.com/merbanan/rtl_433/blob/083dcca4b4842354d5ff6d057df0671e89b5c94e/src/devices/bresser_6in1.c#L143
Is there some change/PR still pending? I'll try to work through some backlog, but that will take weeks.

MacH-21, 2022-12-31:
@zuckschwerdt 
USING ORGrtl_433 (Original version of rtl_433/src/devices/bresser_6in1.c in the MacH-21 Bresser 6in1.zip #1214 3weeks ago)
    int battery = (msg[13] >> 1) & 1; // b[13] & 0x02 is battery_good, s.a. #1993
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-31 17:04:44
model     : Bresser-6in1 id        : 183016b5
channel   : 1            Battery   : 0             Temperature: 20.5 C       Humidity  : 61            Sensor type: 2
Startup   : 1            Flags     : 0             Integrity : CRC
pulse_slicer_pcm(): Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A
bitbuffer:: Number of rows: 1
[00] {202} aa aa aa aa aa 2d d4 57 b1 18 30 16 b5 29 00 00 00 00 00 20 50 61 00 00 f2 00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-31 17:05:02
model     : Bresser-6in1 id        : 18300cb6
channel   : 0            Battery   : 0             Temperature: 8.2 C        Humidity  : 78            Sensor type: 1
Wind Gust : 0.0 m/s      Wind Speed: 0.0 m/s       Direction : 330           UV        : 0.0           Startup   : 1
Flags     : 0            Integrity : CRC
pulse_slicer_pcm(): Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A
bitbuffer:: Number of rows: 1
[00] {202} aa aa aa aa aa 2d d4 a8 3c 18 30 0c b6 18 ff ff ff 33 08 08 24 78 ff f0 12 00

USING myrtl_433 (my modified version of rtl_433/src/devices/bresser_6in1.c in the MacH-21 Bresser 6in1.zip #1214 3weeks ago)
	battery  = (msg[13] >> 2) & 1;  //if battery == 0 Battery is Low, If battery == 1 Battery Ok	
	sign     = (msg[13] >> 3) & 1;  //if sign == 0 Temperature is +Pos, if sign == 1 Temperature is -Neg
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-31 17:07:32
model     : Bresser-6in1 id        : 183016b5
Flags     : 0            Sensor type: 2            channel   : 1             Battery   : 0             Temperature: 20.5 C
Humidity  : 61           Integrity : CRC
pulse_slicer_pcm(): Bresser Weather Center 6-in-1, Hygrometer
bitbuffer:: Number of rows: 1
[00] {202} aa aa aa aa aa 2d d4 57 b1 18 30 16 b5 29 00 00 00 00 00 20 50 61 00 00 f2 00

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-31 17:07:47
model     : Bresser-6in1 id        : 18300cb6
Flags     : 0            Sensor type: 1            channel   : 0             Battery   : 1             Temperature: 8.2 C
Humidity  : 78           Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s       Direction : 330           UV        : 0.0
Integrity : CRC
pulse_slicer_pcm(): Bresser Weather Center 6-in-1, Hygrometer
bitbuffer:: Number of rows: 1
[00] {202} aa aa aa aa aa 2d d4 a8 3c 18 30 0c b6 18 ff ff ff 33 08 08 24 78 ff f0 12 00

ESP32/SX1231/Bresser_6in1 Receiver
57B1183016B52900000000002050610000F2
CRC Ok         = 57B1
CheckSum Ok     = F2
Device ID        = 183016B5
Device Type      = 2
Device Channel   = 1
TX_flag          = 0 & Dev-Type  = 2
Battery_flag     = 0 Battery Low
Temperature_flag = 0 Temperature = +
Temperature       = 20.5C
Humidity          = 61%

ESP32/SX1231/Bresser_6in1 Receiver
A83C18300CB618FFFFFF3308082478FFF012
CRC Ok         = A83C
CheckSum Ok     = 12
Device ID        = 18300CB6
Device Type      = 1
Device Channel   = 0
TX_flag          = 0 & Dev-Type  = 1
WindGust         = 0.0  m/sec 
WindGust-mph     = 0.00 mph 
Wind Average     = 0.0  m/sec 
Wind-Avg-mph     = 0.00 mph 
Wind Direction   = 330  Degrees 
Battery_flag     = 1 Battery Ok
Temperature_flag = 0 Temperature = +
Temperature      = 8.2C
Humidity         = 78%
UV-Index         = 0.0

level20peon, 2023-01-03:
```
[bresser_6in1_decode]
codes     : {144}05d9183016b52900000000002196590000b3
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-01-03 21:18:52
model     : Bresser-6in1 id        : 183016b5
channel   : 1            Battery   : 1             Temperature: 21.9 C       Humidity  : 59            Sensor type: 2
Startup   : 1            Flags     : 0             Integrity : CRC
[pulse_slicer_pcm] Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A
codes     : {426}555555555516ea02ec8c180b5a94800000000010cb2c800059ffff83800000000000000000000000000000000000000000000000000
```

If it helps, this is the sensor which had the low battery indicator before, now with a set of fresh batteries.

gdt, 2023-09-27:
A lot has been fixed in bresser.  Please retest with up-to-date git master.

gdt, 2024-06-04:
abandoned