merbanan/rtl_433 #2294: FUZHOU ESUN ELECTRONIC CO.,LTD outdoor T21 sensor support

asdasdomar, 2022-12-31:
Just bought this weather station on aliexpress:
https://www.aliexpress.com/item/4000015320163.html

I have the station and 3 outdoor sensors and i'm currently pleased with how they work, they seem to be accurate within the specifications.
The FCC ID is printed on a label on each of the 3 sensors:
https://fccid.io/2APN2-EN8822

[image]

Here's the output form the analyzer :

```
Detected OOK package    2022-12-31 10:26:09
Analyzing pulses...
Total count:  126,  width: 430.29 ms            (107573 S)
Pulse width distribution:
 [ 0] count:  126,  width:  592 us [572;660]    ( 148 S)
Gap width distribution:
 [ 0] count:   88,  width: 2032 us [1984;2152]  ( 508 S)
 [ 1] count:   32,  width: 4100 us [4044;4216]  (1025 S)
 [ 2] count:    5,  width: 9032 us [8980;9132]  (2258 S)
Pulse period distribution:
 [ 0] count:   88,  width: 2628 us [2608;2732]  ( 657 S)
 [ 1] count:   32,  width: 4692 us [4644;4800]  (1173 S)
 [ 2] count:    5,  width: 9620 us [9576;9708]  (2405 S)
Pulse timing distribution:
 [ 0] count:  126,  width:  592 us [572;660]    ( 148 S)
 [ 1] count:   88,  width: 2032 us [1984;2152]  ( 508 S)
 [ 2] count:   32,  width: 4100 us [4044;4216]  (1025 S)
 [ 3] count:    6,  width: 9192 us [8980;10004] (2298 S)
Level estimates [high, low]:  16039,   5712
RSSI: -0.1 dB SNR: 4.5 dB Noise: -4.6 dB
Frequency offsets [F1, F2]:    3132,      0     (+11.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Position Modulation with fixed pulse width
view at https://triq.org/pdv/#AAB02D0401025007F0100423E8818181818181818182818281818182828281828181818181818181818281828181818355+AAB00C0401025007F0100423E8818355+AAB0350401025007F0100423E88182818182828182818181818181818182818281818182828281828181818181818181818281828181818355+AAB00C0401025007F0100423E8818355+AAB0350401025007F0100423E88182818182828182818181818181818182818281818182828281828181818181818181818281828181818355+AAB00B0401025007F0100423E88355
Attempting demodulation... short_width: 2032, long_width: 4100, reset_limit: 9136, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PPM,s=2032,l=4100,g=4220,r=9136'

*** signal_start = 1324800, signal_end = 1578787, signal_len = 253987, pulses_found = 269
Iteration 1. t: 139    min: 138 (153)    max: 140 (116)    delta 20
Iteration 2. t: 138    min: 137 (85)    max: 140 (184)    delta 1
Iteration 3. t: 137    min: 136 (35)    max: 139 (234)    delta 2
Iteration 4. t: 137    min: 136 (6)    max: 139 (263)    delta 0
Distance coding: Pulse length 137

Short distance: 518, long distance: 1032, packet distance: 2232

p_limit: 137
bitbuffer:: Number of rows: 13
[00] {42} 4d 00 a3 a0 0a 00 : 01001101 00000000 10100011 10100000 00001010 00
[01] { 0}                   :
[02] { 0}                   :
[03] {42} 4d 00 a3 a0 0a 00 : 01001101 00000000 10100011 10100000 00001010 00
[04] { 1} 00                : 0
[05] {42} 4d 00 a3 a0 0a 00 : 01001101 00000000 10100011 10100000 00001010 00
[06] { 1} 00                : 0
[07] {42} 4d 00 a3 a0 0a 00 : 01001101 00000000 10100011 10100000 00001010 00
[08] { 1} 00                : 0
[09] {42} 4d 00 a3 a0 0a 00 : 01001101 00000000 10100011 10100000 00001010 00
[10] { 1} 00                : 0
[11] {42} 4d 00 a3 a0 0a 00 : 01001101 00000000 10100011 10100000 00001010 00
[12] { 0}                   :
```

This has been recorded after pressing the TX button one single time. It looks like the same data is sent multiple times, do i need to worry about it? Is it just to make communication more robust?

This is where i'm at with decoding the data:

https://triq.net/bitbench#c=%7B42%7D%20a2%2020%20e1%208e%200c%20c0%20%5B22.5C%2071%25%20B_OK%5D&c=%7B42%7D%20a2%2020%20c4%208a%200c%2000%20%5B19.6C%2069%25%20B_OK%5D&c=%7B42%7D%20a2%2020%20cb%208c%200e%2040%20%5B20.3C%2070%25%20B_OK%5D&c=%7B42%7D%20a2%2020%2077%2096%200a%20c0%20%5B11.9C%2075%25%20B_OK%5D&c=%7B42%7D%20a2%202f%20f9%208c%2002%2040%20%5B-0.7C%2070%25%20B_OK%5D&c=%7B42%7D%20a2%202f%20c4%209a%2000%2000%20%5B-6.0C%2077%25%20B_OK%5D&c=%7B42%7D%2065%2000%20b1%2096%2009%2080%20%5B17.7C%2075%25%20B_OK%5D&f=ID%3F12d%20TEMP12s%20HUM7d%201b%208b%202b&cf=ADD&cl=32&cw=4

The ID in the first 12 bits is still a guess, it changes if batteries are removed. On the other hand i'm confident about temperature and humidity. 
I still need to identify the battery status information, it could be stored in the 8th bit of the byte that contains also the humidity but i will have to test that.
There are however 10 other bits that i need to identify, some of those are proabably some kind of CRC but i'm at a complete loss. Can you help me identify what this data might mean?

Thanks,

zuckschwerdt, 2022-12-31:
Yes that the PDV shows this is plain PPM data with (600 µs pulse) 2000 / 4100µs gap and sync of 2000 + 9200 µs gap.
Use a flex decoder with -X 'n=name,m=OOK_PPM,s=2032,l=4100,g=4500,r=11000'
Multiple repeats are expected for reliability.

You already got all the decoding right :) Just use `ID?12b TEMP12s HUM7d 1b 4b 6h`
(only the last 6 bits have entropy, thus are the likely checksum. 6 bit does not sound like CRC but like add with carry over nibbles -- like you guessed!)

asdasdomar, 2022-12-31:
Perfect. That last bit after humidity has to be the battery status bit at this point.
I will now attempt to add the support for the device.

Thanks

zuckschwerdt, 2022-12-31:
For battery-low tests it might work to use a variable voltage power supply, and you additionally get the exact warn- and cut-off voltages.

asdasdomar, 2022-12-31:
I don't have a variable power supply. i will have to wait for some batteries run down.
I found out the ID is acutally only the first byte and the remaining bits contain the channel information:
ID:8d 2b CH:2d TEMP:12s HUM:7d 1b 4b CHKSUM:6h

This is probably the same encoding used here: https://github.com/merbanan/rtl_433/pull/2231

loopy321, 2023-01-22:
Your device is the same as the one [just merged here](https://github.com/merbanan/rtl_433/pull/2231).
I have it and it works well.

commit 9cd20dd8 (2023-01-22): minor: Add Vauno-EN8822C rebrand note (closes #2294)
closed 2023-01-22 by zuckschwerdt