merbanan/rtl_433 #2298: Could not find procedure/function entry point "rtlsdr_cancel_async" from rtl_433-win-x64-22.11.zip

mdeweerd, 2023-01-06:
I extracted the official release rtl_433-win-x64-22.11.zip and try to run

```
rtl_433 -d ""
```
to use my HackRF One on Windows 11 with rtl_433.  The static build does not throw this exception but it does not provide soapysdr.

[image]

CubicSDR from PothosSDR finds the device:
[image]

I tried with the libraries from PothosSDR-2021.07.25-vc16-x64.exe as well with no success (I checked that those were loaded by tracking the files used by rtl_433.exe during runtime).

mdeweerd, 2023-01-06:
I tried with [rtl_433-win-msvc-x64-22.11.zip](https://github.com/merbanan/rtl_433/releases/download/22.11/rtl_433-win-msvc-x64-22.11.zip) which does not produce this error.  So I'll stick with that.

zuckschwerdt, 2023-01-06:
Thanks for the report. There was similar trouble in #2251. So far not sure what the actual problem is, sorry.

The verbose message problem is addressed in the next release.

mdeweerd, 2023-01-06:
FYI: I tried the new binaries posted in #2251 : no error popup about rtlsdr_cancel_async.
So that is better.
However, hackrf is not found there neither.

I also tried to build with (updated) Visual Studio's 2017 and 2019 but I get this message:
[image]

Should I use another PothosSDR version than the one in BUILDING.md ?

zuckschwerdt, 2023-01-06:
We have no file `sched.h` but I guess this originates with `THREAD_RETURN` maybe?

Re HackRF: you did install Pothos and e.g. `SoapySDRUtil` can find the SDR?

mdeweerd, 2023-01-06:
This shows where sched.h is located (PothosSDR) :
[image]

Checking the headers, I find:
c:\Program Files\PothosSDR\include\pthread.h:#include <sched.h>

```
C:\Windows\System32>soapysdrutil --find="driver=hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

←[1m←[33m[WARNING] SoapyVOLKConverters: no VOLK config file found. Run volk_profile for best performance.←[0m
Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 a06063c8232a4c5f
  part_id = a000cb3c0053475a
  serial = 0000000000000000a06063c8232a4c5f
  version = 2017.02.1
```

And with the firmware posted in  #2251 :

```
rtl_433.exe -d "" -A -T 10
rtl_433 version 22.11-23-g4c495fd3 branch feat-logtooutputs at 202212091704 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "D:\...\workspace\rtl_433\release2\rtl_433.conf"...
Trying conf file at "C:\Users\...\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
[Protocols] Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
[sdr_open_soapy] Failed to open sdr device matching ''.

D:\...\rtl_433\release2>rtl_433.exe -d "driver=hackrf" -A -T 10
rtl_433 version 22.11-23-g4c495fd3 branch feat-logtooutputs at 202212091704 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "D:\...\workspace\rtl_433\release2\rtl_433.conf"...
Trying conf file at "C:\Users\...\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
[Protocols] Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
[sdr_open_soapy] Failed to open sdr device matching 'driver=hackrf'.
```

zuckschwerdt, 2023-01-06:
Oh, I guess with Pothos properly installed you would need to delete the DLLs bundled with rtl_433? Those DLLs will work but don't point to a folder with e.g. HackRF driver modules.

mdeweerd, 2023-01-06:
The binaries of PothosSDR look in their local directory, do I copied rtl_433.exe over to that bin directory and I also had to copy librtlsdr.dll .

This then worked:
```
rtl_433.exe -d "driver=hackrf" -f 433MHz -A -T 10
```

Thanks for your help - maybe something to add to the documentation?

zuckschwerdt, 2023-01-06:
Thanks! Yes, there should be some notes. Or maybe should not add the Soapy DLL at all.

gdt, 2023-09-27:
A fix was merged, so calling this done.   Please open a fresh issue against current git master if there is still a problem.

closed 2023-09-27 by gdt