merbanan/rtl_433 #2303: Ecowitt WH32/WN32B not decoded by rtl_433

sombrero2003, 2023-01-09:
Hello,
I am running rtl_433 via the HA add-on and my ecowitt WH32 sensor is not recognized.
I noticed that the sensor id listed on the box reads WH32B and not WH32. It seems that's the new name of the [WH32](https://www.wxforum.net/index.php?topic=44754.0). Could the change of name explain the issue with the decoding?
Thanks a lot for this great piece of software

zuckschwerdt, 2023-01-09:
Could be the same problem as #2274
Can you grab a sample and upload as zip here? https://triq.org/rtl_433/ANALYZE.html

sombrero2003, 2023-01-10:
Thanks. I installed rtl_433 on my mac and  ran the following command
rtl_433 -f 915.05M -S unknown

I have attached the first file that is generated. 

[g001_915.05M_1000k.cu8.zip](https://github.com/merbanan/rtl_433/files/10378320/g001_915.05M_1000k.cu8.zip)

Let me know if this what you need.

zuckschwerdt, 2023-01-10:
I assumed a typo there, but you can confirm the name of "WN32B" right? This looks much like the WH32B but the two repeats are not mashed together. Difficult to detect now wether it's 
Fine Offset Electronics WH25, WH32, WH32B, **WN32B** / WH0290 / WH24, WH65B, HP1000 -- they are all mostly the same each with different quirks.

I need to carefully figure this out, if you just want it to work right now apply these changes and recompile.

```diff
--- a/src/devices/fineoffset.c
+++ b/src/devices/fineoffset.c
@@ -474,7 +479,10 @@ static int fineoffset_WH25_callback(r_device *decoder, bitbuffer_t *bitbuffer)
     unsigned bit_offset;

     // Validate package
-    if (bitbuffer->bits_per_row[0] < 190) {
+    if (bitbuffer->bits_per_row[0] > 160 && bitbuffer->bits_per_row[0] < 190) {
+        type = 32; // new WN32B
+    }
+    else if (bitbuffer->bits_per_row[0] < 190) {
         return fineoffset_WH0290_callback(decoder, bitbuffer); // abort and try WH0290
     } else if (bitbuffer->bits_per_row[0] < 440) {  // Nominal size is 488 bit periods
         return fineoffset_WH24_callback(decoder, bitbuffer); // abort and try WH24, WH65B, HP1000
@@ -487,7 +495,7 @@ static int fineoffset_WH25_callback(r_device *decoder, bitbuffer_t *bitbuffer)
     // Find a data package and extract data payload
     // Normal index of WH25 is 367, and 123, 570 for WH32B
     // skip some bytes to find faster
-    bit_offset = bitbuffer_search(bitbuffer, 0, 100, preamble, sizeof(preamble) * 8) + sizeof(preamble) * 8;
+    bit_offset = bitbuffer_search(bitbuffer, 0, 0, preamble, sizeof(preamble) * 8) + sizeof(preamble) * 8;
     if (bit_offset + sizeof(b) * 8 > bitbuffer->bits_per_row[0]) {  // Did not find a big enough package
         decoder_logf_bitbuffer(decoder, 1, __func__, bitbuffer, "short package. Header index: %u", bit_offset);
         return DECODE_ABORT_LENGTH;
```

sombrero2003, 2023-01-11:
Thanks a lot. This works great.
As far as the sensor is concerned, I purchased a WH32 Outdoor coded (https://shop.ecowitt.com/products/wh32-outdoor) but on the back it does read WN32B.
As per this thread https://www.wxforum.net/index.php?topic=44754.0, the N seems to mean "no solar panel"

gdt, 2023-09-26:
A lot has been improved.   Please retest with current git master.

sombrero2003, 2023-10-28:
I don't think these sensors have been included in the master. I have kept using the code changes suggested by [zuckschwerdt](https://github.com/zuckschwerdt) with no issue (https://github.com/sombrero2003/rtl_433/tree/fix_wh32).

Quick67, 2023-12-24:
Hello, no news on when this fix will be present in the release? thanks!

commit 83afc7b4 (2023-12-25): Add support for Fine Offset WN32B (closes #2303)
closed 2023-12-25 by zuckschwerdt