merbanan/rtl_433 #2304: Different Device IDs in regular output vs JSON

infabo, 2023-01-09:
rtl_433 reports different device IDs, when I start in regular stdout and JSON output mode.
I've found another github issue around here, where someone complained about the JSON-schema's `id`-attribute is not declared as `integer`. Though zuckschwerdt argued, that it can be `string` and is perfectly valid to be `integer|string`.
It looks to me, someone silently changed this for JSON-output mode silently and casts/converts any string to integer?

see #2005
see https://github.com/merbanan/rtl_433/blob/master/docs/DATA_FORMAT.md - still mentions the "rarely string".

## Regular

`docker run --device /dev/bus/usb hertzg/rtl_433 -f 868.25M -s 1536k -R 150`

```
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 1 out of 223 device decoding protocols [ 150 ]
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 1536000 S/s.
Tuner gain set to Auto.
Tuned to 868.250MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-01-09 13:28:12
model     : Klimalogg-Pro Id        : 7a3f
Battery   : 1            Temperature: 20.4 C       Humidity  : 56            Sequence Number: 0        Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-01-09 13:28:23
model     : Klimalogg-Pro Id        : 7a3f
Battery   : 1            Temperature: 20.4 C       Humidity  : 56            Sequence Number: 0        Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-01-09 13:28:34
model     : Klimalogg-Pro Id        : 7a3f
Battery   : 1            Temperature: 20.4 C       Humidity  : 56            Sequence Number: 0        Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-01-09 13:28:36
model     : Klimalogg-Pro Id        : 1d16
Battery   : 1            Temperature: 7.3 C        Humidity  : 76            Sequence Number: 0        Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-01-09 13:28:45
model     : Klimalogg-Pro Id        : 7a3f
Battery   : 1            Temperature: 20.4 C       Humidity  : 56            Sequence Number: 0        Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-01-09 13:28:47
model     : Klimalogg-Pro Id        : 1d16
Battery   : 1            Temperature: 7.2 C        Humidity  : 76            Sequence Number: 0        Integrity : CRC
```

## JSON

`docker run --device /dev/bus/usb hertzg/rtl_433 -f 868.25M -s 1536k -R 150 -Fjson`

```
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 1 out of 223 device decoding protocols [ 150 ]
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 1536000 S/s.
Tuner gain set to Auto.
Tuned to 868.250MHz.
Allocating 15 zero-copy buffers
{"time" : "2023-01-09 13:30:31", "model" : "Klimalogg-Pro", "id" : 7446, "battery_ok" : 1, "temperature_C" : 7.200, "humidity" : 76, "sequence_nr" : 0, "mic" : "CRC"}
{"time" : "2023-01-09 13:30:36", "model" : "Klimalogg-Pro", "id" : 31295, "battery_ok" : 1, "temperature_C" : 20.400, "humidity" : 56, "sequence_nr" : 0, "mic" : "CRC"}
{"time" : "2023-01-09 13:30:47", "model" : "Klimalogg-Pro", "id" : 31295, "battery_ok" : 1, "temperature_C" : 20.400, "humidity" : 56, "sequence_nr" : 0, "mic" : "CRC"}
{"time" : "2023-01-09 13:30:58", "model" : "Klimalogg-Pro", "id" : 31295, "battery_ok" : 1, "temperature_C" : 20.400, "humidity" : 56, "sequence_nr" : 0, "mic" : "CRC"}
```

zuckschwerdt, 2023-01-09:
Yes, "generally integer" is what we want. A quick summary show currently there are 225 instances of integer and 32 of string -- and those are likely in error.
This is rooted in the confusion you observe: the KV output has a formatting applied and that can display integer as hex (31295 is 0x7a3f). There is no "hex" literal in e.g. JSON, numbers are decimal. If you want to view the JSON numbers as hex then you need to format them on display, but a quick hack is to output a string -- not ideal.

We should check those 32 cases to see if we can output a number and make users aware of this possible confusion.

gdt, 2023-09-27:
@infabo It would be great if you could retest with current master and then file a PR to fix what you are seeing.

zuckschwerdt, 2023-09-27:
This is linked to #1321
The resolution here specifically: it is confusing that data (an int) is formatted (as hex) for the KV output. Nothing we want to change though.

closed 2023-09-27 by zuckschwerdt