merbanan/rtl_433 #2309: MQTT issue when two different servers with same dongles try to connect to MQTT server

avicarmeli, 2023-01-13:
I am running two different VM servers on Proxmox each with it's own radio dongle. I get that on tyhe log of the MQTT server:

```
2023-01-13 12:01:21: Client rtl_433-a796d7a3 already connected, closing old connection.
2023-01-13 12:01:21: New client connected from 192.168.XXX.XXX:49550 as rtl_433-a796d7a3 (p2, c0, k60, u'username').
```
I think it is similar to [that ](https://github.com/merbanan/rtl_433/issues/1129)issue.
I wonder, how can I change client id  sent to the MQTT server?

merbanan, 2023-01-13:
Use the -d parameter.

closed 2023-01-13 by merbanan

avicarmeli, 2023-01-13:
Well that can't be done since that are two different servers running rtl_433 each sees one dongle as d0.

zuckschwerdt, 2023-01-13:
Like merbanan said, use `-d` to change the client id. Use `rtl_eeprom` to change the serial if needed.
Otherwise make sure they don't use the same hostname (actually not the same `CRC-16(hostname)`)
https://github.com/merbanan/rtl_433/blob/2cd5623c9ac65d49a542420ce91ac33ea645f32d/src/output_mqtt.c#L485-L489

avicarmeli, 2023-01-13:
Changing the serial number to 00000002 would not change anything. Still I can only run with -d 0.  using  -d 1 would yield: 
[Protocols] Registered 2 out of 224 device decoding protocols [ 224 224 ]
[Protocols] Registered 2 out of 224 device decoding protocols [ 224 224 ]
```
[SDR] Found 1 device(s)
[SDR] Found 1 device(s)
[SDR] trying device  1:  n/a, n/a, SN: n/a
[SDR] trying device  1:  n/a, n/a, SN: n/a
[sdr_open_rtl] Failed to open rtlsdr device #1.
[sdr_open_rtl] Failed to open rtlsdr device #1.
[sdr_open_rtl] Unable to open a device
[sdr_open_rtl] Unable to open a device
```
I might not use it the way you intended.

zuckschwerdt, 2023-01-13:
> OK the problem was due to conf file while using full command from command line.

Not sure what that means, but glad it now works!

Note: If you set a serial of `00000002` then use `-d :00000002` (serial is not a number but a string, e.g. for a [Nooelec NESDR SMArt you could use `NESDRSMA`)

Also no `-d` is different from `-d 0`, so that should work too. But really your hostnames clash and that should be repaired.

avicarmeli, 2023-01-14:
Just to clarify, I had translated a command line into a config file and then forgot about the config file and ran again the full command line (I realized that when I pasted the output up here, when I saw that every line of the output repeated twice) .
So actually, I guess, the rtl_433 was running two process doing the same, which end up in reconnecting each time to the MQTT broker. The solution was to add -b NULL to the command line.