merbanan/rtl_433 #2313: LTV-R3 hand random jump in rainfall count every 64mm

nilkemorya, 2023-01-14:
I have an Lacrosse LTV-R3 rain guage sensor which has a odd jump in rainfall every 64mm of received rain. Whenever it hits a 64 mm number the rainfall amount jumps by 16320 mm. Ex. 63.75mm tips over to go to 64mm and instead it goes to 16384mm. Then later when it goes another 64mm it will jump to 32768. Kind of a suggestive number. 
Additionally it has an offset of 10880 from zero whenever you reset the guage. I assumed this was just an onboard chip bug, but now I'm wondering. 
 This seems like some kind of weird internal counter. Unsure if it's just my unit, a measure extra bytes bug or just something that these do for there own ineffable reasons.

zuckschwerdt, 2023-01-14:
What's the value of `rain2_mm` does it always match `rain_mm` ?

Is it truely a LTV-R3? Or could it be some different model/revision/reseller?

Have a look here to see if any of the notes point to the source of the error https://github.com/merbanan/rtl_433/blob/master/src/devices/lacrosse_r1.c

Quick guees: since rain is measured in 1/4mm the 64 you notice is 256 raw, so a byte overflow. Since 256*256/4 is the "step" you see we may have the byte order wrong?

@mbruski did you observe clean overflows in the rain bytes?

nilkemorya, 2023-02-24:
Sorry for the long delay, 
rain2_mm does match rain_mm within 1 minute. 
It is truly an LTV-R3 according to the purchase receipt case and manual. 
I can definitely try poking about in raw_rain decode to see if maybe there is just a byte order issue. 
It definitely feels really suspect that the amount it jumps by in raw rain numbers is exactly 1111111100000000 in binary. 
I have to admit my bitwise math is really weak though. So, poking about a bit more, you are right, it looks like instead of going from 11111111 to 100000000 it goes to 10000000000000000 and so on every 256.

nilkemorya, 2023-02-24:
I should have mentioned. Rain2 does just appear to be the value of rain2 lagged by 2 minutes. I'm guessing the lacrosse weather stations use that for rainfal rate tracking somehow.

zuckschwerdt, 2023-02-24:
LaCrosse LTV-R3 is protocol `175`. Please with `-R 175:v` which turns that decoder verbose and you get the raw codes:
`codes     : {144}70f6a24600aa0000aa00f100000000000000`
Try to capture them before and after an overflow.

nilkemorya, 2023-02-25:
The good news is it's only 256 tips, so that's not so bad. 
Codes before: 
lacrosse_r1_decode: : {144} 71 06 1d 08 00 aa c3 00 aa b7 01 00 00 00 00 00 00 00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-25 14:53:34
model     : LaCrosse-R3  Sensor ID : 71061d
Battery level: 1         Sequence  : 4             Total Rain: 10928.75 mm   Total Rain2: 10925.75 mm
Integrity : CRC
pulse_slicer_pcm(): LaCrosse Technology View LTV-R1, LTV-R3 Rainfall Gauge, LTV-W1/W2 Wind Sensor
bitbuffer:: Number of rows: 1
[00] {234} aa aa aa aa aa aa aa b4 aa 8b 75 1c 41 87 42 00 2a b0 c0 2a ad c0 40 00 00 00 00 00 00 00
Codes after: 
time      : 2023-02-25 14:56:11
model     : LaCrosse-R3  Sensor ID : 71061d
Battery level: 1         Sequence  : 1             Total Rain: 27264.75 mm   Total Rain2: 27264.75 mm
Integrity : CRC
pulse_slicer_pcm(): LaCrosse Technology View LTV-R1, LTV-R3 Rainfall Gauge, LTV-W1/W2 Wind Sensor
bitbuffer:: Number of rows: 1
[00] {233} aa aa aa aa aa aa aa b4 aa 8b 75 1c 41 87 40 80 6a 80 c0 6a 80 d1 80 00 00 00 00 00 00 00
lacrosse_r1_decode: packet length: 265
lacrosse_r1_decode: : {175} 0a 8f ff 06 1f c0 2b 03 10 07 53 00 d2 d2 d2 00 00 00 00 00 00 00

Sorry I didn't get the exact packet before and after the overflow, I think I miscounted. Let me know if that's critical. Here's the packet from 0 as well notice the large initial offset: 
time      : 2023-02-25 14:58:20
model     : LaCrosse-R3  Sensor ID : 71061d
Battery level: 1         Startup   : 1             Sequence  : 1             Total Rain: 10880.00 mm
Total Rain2: 10880.00 mm Integrity : CRC
pulse_slicer_pcm(): LaCrosse Technology View LTV-R1, LTV-R3 Rainfall Gauge, LTV-W1/W2 Wind Sensor
bitbuffer:: Number of rows: 1
[00] {233} aa aa aa aa aa aa aa b4 aa 8b 75 1c 41 87 50 80 2a 80 00 2a 80 31 80 00 00 00 00 00 00 00

Thanks for your help!

zuckschwerdt, 2023-02-25:
Here are those codes [in a BitBench](https://triq.net/bitbench#c=aaaa%20aaaa%20aaaa%20aab4%20aa8b%20751c%204187%20%204200%202ab0%20c02a%20adc0%204000%200000%200000%200000&c=aaaa%20aaaa%20aaaa%20aab4%20aa8b%20751c%204187%20%204080%206a80%20c06a%2080d1%208000%200000%200000%200000&c=aaaa%20aaaa%20aaaa%20aab4%20aa8b%20751c%204187%20%205080%202a80%20002a%208031%208000%200000%200000%200000&f=ID%3A24h%20%3F%3A4b%20SEQ%3A3d%20%3F%3A1b%20RAIN%3A24h%20RAIN%3A24h%20CRC%3A8h%20TRAILER%3A56h&a=Preamble&m=d2aa2dd4&cw=4). We observe the middle byte of the rain fields is `aa` which might be an unused "filler".
It's strange because we did see a value that field: 
https://github.com/merbanan/rtl_433/blob/030d0ab222ef235e2acac81ad1a9363e745de626/src/devices/lacrosse_r1.c#L52-L67

Did I understand correctly: the last code you posted is after a reset to zero?

If you want to quickly test a fix then change these two lines:
[    int raw_rain1 = (b[4] << 16) | (b[5] << 8) | (b[6]);](https://github.com/merbanan/rtl_433/blob/master/src/devices/lacrosse_r1.c#L150-L151)
to
```
    int raw_rain1 = (b[4] << 8) | (b[6]);
    int raw_rain2 = (b[7] << 8) | (b[9]); // only LTV-R3
```
It does not explain the middle byte though. Is it just a fixed value of `aa`, `54` (and perhaps `a8`)?

nilkemorya, 2023-02-26:
Yes, the last code was after a reset to zero. I figured it might help to have it for reference. I'll give that a try later and post back.

nilkemorya, 2023-02-26:
Alright tested with your suggested change! Not only has the weird 0 offset vanished, but I just hand tipped it past 64mm no problem. I ran it all the way past 128mm as well, seems to now be counting up as expected. Both rain1 and rain2. I think we're in business. Doesn't answer the middle byte, but the rest seesms to be working.

zuckschwerdt, 2023-02-26:
To get insight into the middle byte behaviour we would need to tip past 65535 (16383.75 mm). I wouldn't expect you do that by hand. Maybe keep an eye out for that number, if it maybe someday comes close?

closed 2023-02-26 by zuckschwerdt

jbjonesjr, 2023-07-03:
Thanks for this work. I haven't updated my rlt code in a while, but have been experiencing this error recently (welcome to summer strong weather season in the Eastern US), so am happy to see it resolved. I'll pull from `main` and make sure my app is updated. Any debugging I can provide for you in the future, let me know if I can help

commit fc034df5 (2023-02-26): Fix LaCrosse-R1/LaCrosse-R3 rain value (closes #2313)