merbanan/rtl_433 #2315: Ability to decode the partial signal every frame

sindalschmidt, 2023-01-15:
_Previously discussed [here](https://groups.google.com/g/rtl_433/c/8uVm_9eevng/m/QjBNtIKcAQAJ)._

### Use case:
Listening for radio transmission from smoke detectors and other alarms, it would beneficial to receive the output as soon as a valid frame is available. This allows one to act immediately on threats announced by alarms which signals using a long continuous transmission.

### Current behavior
Take for instance Cavius smoke-, heat- and water alarms. These are currently supported by rtl_433. However when these alarms go into pairing, test or alarm state, they will broadcast a radio signal for up 15-20 seconds (Repeatedly if in actual alarm state). This means that rtl_433 will first output event when the radio transmission goes silent after 15-20 second. This is not ideal if the signal are used as an aid in an emergency case, such as turning on all lights using mqtt and openHAB in case of fire.

### Expected behavior
As soon as a valid frame can be decoded, rtl_433 should output it. I do not know how to achieve this, but as discussed [here](https://groups.google.com/g/rtl_433/c/8uVm_9eevng/m/QjBNtIKcAQAJ), Christian suggest adding the ability to decode the partial signal every frame (250ms). This will result in duplicate events, but this should not be an issue, as they can easily be dealt with (openHAB will by default ignore them, based on the configuration).

zuckschwerdt, 2023-01-15:
There is currently no easy way to achieve this, the FSK demod will only (successfully) end when the OOK demod sees a gap, which here is not the case.

We could perhaps try to decode the partial signal every frame (250ms) or such. But this will lead to duplicates if every prefix (partial frame) is valid.

With "duplicates" here I don't mean the repeated events for each subsequent packet but true duplicate outputs for each decoder run over the unfinished signal prefix. I don't see how we could collect feedback from every possible decoder if a portion of the prefix could be discarded as "done".

Likely this will require a new strategy on how to demod. Or a census what the longest expected signal should be (around 1 second?) and then a last-resort cut-off at that point.

Note this can be tested with:
`rtl_433 -M replay tests/cavius/g001_869.67M_250k.cu8`
(this takes 6 seconds to report – we'd like it to be much faster)

sindalschmidt, 2023-01-15:
> Note this can be tested with: `rtl_433 -M replay tests/cavius/g001_869.67M_250k.cu8` (this takes 6 seconds to report – we'd like it to be much faster)

I apologize for my ignorance, but I don't understand this part. Could you please elaborate?

merbanan, 2023-01-15:
The current logic works by segmenting the raw baseband from signal start to stop. After that we demodulate the complete signal and then send of the different bitbuffers to be decoded by the decoders.

What could be done is to segment signals based on the usb buffers we get and then pass of to demodulators and decoders. We could gather blocks from signal start and pass more and more completed buffers as time pass. And if we get a decoded signal we stop and wait for signal stop or restart.

So first we try with buffer 1 then 1 and 2, then 1 and 2 and 3. If we get a decoded signal after 3 buffers we can reset signal start and try to decode again until signal end.

Maybe this could be handled if we ever do a rewrite of the dsp buffer management.

sindalschmidt, 2023-01-27:
Unfortunately, with admittedly uncertainty that it is in fact the root cause, it seems that my current workaround mentioned in the original post has caused my Intel NUC running Debian to stall one time.

Therefore I have looked into implementing an abort condition instead, as suggested by Christian Z [here](https://groups.google.com/g/rtl_433/c/8uVm_9eevng/m/QjBNtIKcAQAJ), but with no C experience, I struggle with understanding how and where to make such change, regardless of how many Google searches I perform 😄

Would it be possible to guide me on how to accomplish this?

sindalschmidt, 2023-02-02:
> Or a census what the longest expected signal should be (around 1 second?)

Perhaps support of an additional argument would be the best solution to this? The argument could add the capability to specify timeout in milliseconds after detecting a signal. If one were to specify "250" the OOK demod should return to FSK demod after a maximum time of 250 milliseconds.

Unfortunately I have no experience in C, but I would love to be of assistance in any other way that I can.

zuckschwerdt, 2023-02-02:
The idea is well defined but there is currently no easy way to achieve this feature.
It would very likely be a major rewrite of the demod code.

sindalschmidt, 2023-02-02:
I appreciate your reply. I took the leap into the "c"-world and managed to hack something together myself, by doing the following:
```
case PD_OOK_STATE_PULSE:
s->pulse_length += 1;
// Custom code
if (s->pulse_length % 100000 == 0) {
//fprintf(stderr, "pulse_length %u\n", s->pulse_length);
s->custom_pulse = 1;
pulses->pulse[pulses->num_pulses] = s->pulse_length;
s->max_pulse = MAX(s->pulse_length, s->max_pulse);
s->pulse_length = 0;
s->ook_state = PD_OOK_STATE_GAP_START;
}
// End of custom code
```
```
case PD_OOK_STATE_GAP_START:    // Beginning of gap - it might be a spurious gap
s->pulse_length += 1;
// Pulse detected again already? (This is a spurious short gap)
if (s->custom_pulse == 0 && am_n > (ook_threshold + ook_hysteresis)) { // Custom code, added statement: s->custom_pulse == 0
```
By doing so every 100000 pulses I get a message almost instantaneous and around 2-3 message outputs per second during the pulse of the alarm.

I have never coded in C before, so do let me know if I'm setting myself up for trouble by doing so. The last thing I want to do is miss this message when needed for real.

zuckschwerdt, 2023-02-03:
That's a neat idea. Thanks for sharing!
Though `pulse_length` here is not the overall signal length but really just the current high-pulse.
So your condition is actually `if (s->pulse_length >= 100000) { // abort on overlong (100us) pulses`

If the problem you are seeing is/has overlong pulses then that's easier to solve then random continuous signals :)
We actually would want that to protect from a "jammed" band.

sindalschmidt, 2023-02-04:
Very interesting. The current high-pulse would seem to go on for the 14-15 seconds or so that the alarm broadcasts it signal. I don't know if that's normal, good or bad, or what it means. I found that the s->pulse_length would reach huge numbers and thought it was simply an signal duration counter, that I could use as a "timer" after figuring out the minimum count that would produce a message.

I suppose the best option would be to query the demod(?) every so often during a pulse to determine if valid message is already available and then output it. I can't seem to figure out how to do this. Would I need to use "fsk_pulses->num_pulses" (hint taken from the comment "Determine if FSK modulation is detected")?

Otherwise I could start a timer with first high-pulse and then an abort condition if timer is >250ms. I'm not sure which approach would be most reliable – a timer or the current way I'm doing it.

Thank!

zuckschwerdt, 2023-02-04:
In this case it can act as a timer: the FSK signal sees a continuous pulse in OOK (varying frequency though).
This must end quickly, no legitimate FSK signal will carry on that long, and repeated signals need a break between packets.

A hard abort on overlong pulses, no matter if it's a OOK high or FSK data, needs to happen.
Your current approach is valid.

We need to census what the longest OOK and FSK transmission we know are, add some safety margin and set that as you did.

bgmcode, 2023-10-14:
This would also be useful to me, since I am using rtl_433 to decode my doorbell signal.