merbanan/rtl_433 #2318: pretty vs raw values

jedi7, 2023-01-16:
Hi,
is there a design plan for pretty versus raw values?

Context:
For example I receive data saying power is "1000 kW". This is nice for user as visualization, but for computer processing (json, mqtt output) it is not good. There we will need raw value 1000000.

One solution is to add both keys. One key with pretty number and second key with raw value. But user will see many redundant information.

Or somehow extend the storage interface to contain these informations:
* pretty value // visible only in console
* pretty measurement unit  // visible only in console
* raw value   // visible only in json,mqtt, databases, etc

Or somehow allow visibility change by configuration parameter.

I'm able to do PR for this, at least for the WMBUS , but I will need your solution proposal :)

zuckschwerdt, 2023-01-16:
We aim to output only a raw value but converted to sensible units.
If the report is in kW or W then that scale. If a majority of devices use a specific one then we should convert. (Currently wind km/h vs m/s is undecided but should be normalized).

It might be possible to offer a conversion, e.g. `-C pretty`, but keys include the scale and unit and we should not changes those dynamically.

merbanan, 2023-01-16:
Is it only power that has this issue? To me the native unit of power is only watts.

jedi7, 2023-01-16:
It is for more units, like J, W, m3/h, etc. (see "unit_names" in m_bus.c)
Problem is, when I convert it to base unit, then I can get very hard to read numbers. 
Some of the numbers can have exponent 9 (or -6). So you will then see 1000000000 J instead of 1 GJ.

zuckschwerdt, 2023-01-16:
Agreed, we might see a kW/h report but not kW, only W (I mean "1000 kW"? is that an industrial application?)
The mbus units are defined by the spec I take it? We are just talking about the KV output then, and only for metric SI keys, and likely only a short hand-picked list? I would rather see consumers implementing this.

What is your specific use-case? Are you looking at the KV output for more than setup/debugging? Or do you want this in another output?

jedi7, 2023-01-16:
better example is with the Joule. When for heaters is normal to use GJ, but for other applications it can be really just J.
The wmbus have some predefined units and possible exponents from -6 to 9.

I like the idea of "-C pretty". So maybe we can store the base measure unit and on a common place recompute it to more readable value?

zuckschwerdt, 2023-01-16:
Ok so, for now, only "J". With J it won't be a negative scale, right? As said it would be good not to change the keys, so this should be located in the KV output at `print_kv_double()` / `print_kv_int()`.
It needs to detect the format to be "...J" and then apply some scale and insert a prefix, right?

zuckschwerdt, 2023-01-16:
If you want to dive into that, it will be accepted if the changes are minimal, i.e. a key match and call to a "metric format number"-function in those 1 or 2 print functions. A default prettified output is okay for KV.

zuckschwerdt, 2023-01-22:
Closing this question, but do PR if you like and mention this issue ID for tracking.

closed 2023-01-22 by zuckschwerdt