merbanan/rtl_433 #2322: Support for EXPLORE SCIENTIFIC Soil Moisture and Soil Temperature Sensor

mr-varga, 2023-01-16:
Hi everyone,

thanks for this amazing project. I have  a weather station from Bresser (NATIONAL GEOGRAPHIC WIFI Colour Weather Station with 7in1 Sensor) and I use this tool to import the data to my MQTT server. The only think I missing and I loke to import is the data from the external soil sensor connected to the same station on Chanal 1. The sensor is the EXPLORE SCIENTIFIC Soil Moisture and Soil Temperature Sensor (https://www.bresser.de/it/Articoli-di-Fine-Serie/Sensore-per-temperatura-e-umidita-del-suolo-EXPLORE-SCIENTIFIC.html). Can someone help me to import also this data to my MQTT server and add this feature to this project please?

merbanan, 2023-01-16:
Hi, follow the instructions here:

https://github.com/merbanan/rtl_433/blob/master/docs/ANALYZE.md

mr-varga, 2023-01-16:
> Hi, follow the instructions here:
> 
> https://github.com/merbanan/rtl_433/blob/master/docs/ANALYZE.md

Wow It's a bit complicated for me... I try but I don't know what I'am doing...

matthias-bs, 2023-01-16:
Hi,

since on product page of the weather station you mentioned, also this sensor https://www.bresser.de/it/Ora-Tempo/Misuratore-di-umidita-e-temperatura-del-suolo-BRESSER.html is listed as accessory, there is a chance that both use the same protocol. The latter uses the Bresser 6-in-1 protocol (decoder: https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_6in1.c).

Good luck!

gdt, 2023-09-27:
This seems like a request for support with no activity - which isn't really useful to have in an issue, since there is nobody picking random devices from issues, buying one, figuring it out, writing code, and filing PRs..    Please feel free to engage if you are going to dig in.

closed 2023-09-27 by gdt