merbanan/rtl_433 #2324: Request: Add Meat Thermometer decoder

lucywinters, 2023-01-18:
Hi There,
I have a BBQ Meat thermometer that I'd love to see added into RTL_433. I've captured the signal and believe I have a working flex decoder and I can decipher what the hex/binary means. Though writing the code to make this happen is a little beyond my comfort levels.

**Model:** Gasmate Remote Digital Thermometer (BA1008)

The Flex Decoder Options are:
-X 'n=name,m=OOK_PPM,s=536,l=1668,r=2000'

Some of my testing:
--------------------------------------------------------
1st 4bits: Always same
2nd 4bits: Negative or Positive (0000=Positive 0100=Negative)
3rd 4bits: temp tenth eg 1 in 10c
4th 4bits: single val eg 0 in 10c
5th 4bits: Always 4x0s?
6th 4bits: ?
7th 4bits: ?
8th 4bits: ?
--------------------------------------------------------

Examples:
-------------------------------
F0230FDC (+23C)
1111 0000 0010 0011 0000 1111 1101 1100

F4040BFB (-04C)
1111 0100 0000 0100 0000 1011 1111 1011

F4100BEF (-10C)
1111 0100 0001 0000 0000 1011 1110 1111

F4060BF9 (-06C)
1111 0100 0000 0110 0000 1011 1111 1001

f4040bfb (-4C)
1111 0100 0000 0100 0000 1011 1111 1011

f0030ffc (+03C)
1111 0000 0000 0011 0000 1111 1111 1100

F0310FCE (+31C)
1111 0000 0011 0001 0000 1111 1100 1110

I hope that the above information is enough for somebody to work with, If there is anything further I can offer please let me know.

Many thanks

Lucy

zuckschwerdt, 2023-01-18:
The last nibble (4-bit) is a checksum: nibble-wide add with carry.
Here is [a BitBench](https://triq.net/bitbench#c=%5BGasmate%20BA1008%20meat%20thermometer%20%232324%5D&c=F4040BFB%5B-04C%5D&c=F4060BF9%5B-06C%5D&c=F4100BEF%5B-10C%5D&c=f0030ffc%5B%2B03C%5D&c=F0230FDC%5B%2B23C%5D&c=F0310FCE%5B%2B31C%5D&f=MODEL%3Ah%20SIGN%3Ah%20TEMP%3AhhC%20%3Fhhh%20CHK%3Ah&cf=ADD&cl=32&cw=4) with the decoding and checksum.

The maximum of `+99` seems a little close to normal operating values, should there be some margin and another digit maybe? Or is this explicitly not for baking, grilling and such? Can you (briefly) expose the device to 100+?

lucywinters, 2023-01-19:
Such a quick response - thanks so much.

I've heated the thermometer up to over 100C. While I can decipher the results, it is not as clean as I had originally thought.

f1810e7e = 181C
1111 0001 1000 0001 0000 1110 0111 1110
 
f1740e8b = 174C
1111 0001 0111 0100 0000 1110 1000 1011

f1690e96 = 169C
1111 0001 0110 1001 0000 1110 1001 0110

f1640e9b = 164C
1111 0001 0110 0100 0000 1110 1001 1011

making the updated legend:
1st 4bits: Always same
2nd 4bits: hundreds eg 2 in 200C OR Negative Positive (0000=Positive 0100=Negative)
3rd 4bits: temp tenth eg 1 in 10c
4th 4bits: single val eg 0 in 10c
5th 4bits: Always 4x0s?
6th 4bits: ?
7th 4bits: ?
8th 4bits: ?

I hope this helps, and really thank you for your time.
Lucy

zuckschwerdt, 2023-01-19:
Yes, this works nicely, [updated BitBench](https://triq.net/bitbench#c=%5BGasmate%20BA1008%20meat%20thermometer%20%232324%5D&c=F4040BFB%5B-04C%5D&c=F4060BF9%5B-06C%5D&c=F4100BEF%5B-10C%5D&c=f0030ffc%5B%2B03C%5D&c=F0230FDC%5B%2B23C%5D&c=F0310FCE%5B%2B31C%5D&c=f1810e7e%5B181C%5D&c=f1740e8b%5B174C%5D&c=f1690e96%5B169C%5D&c=f1640e9b%5B164C%5D&f=MODEL%3Ah%20%3Fb%20SIGN%3Ab%20TEMP%3A2hhhC%20%3Fhhh%20CHK%3Ah&cf=ADD&cl=32&cw=4).
The format is `MODEL:h ?b SIGN:b TEMP:2hhhC ?hhh CHK:h` and that first unknown bit could maybe indicate low-battery. Did you test that? If you want to probe more, then try a (nearly) flat battery or a variable voltage power supply :)

lucywinters, 2023-01-30:
Hi Christian,

Sorry I've been away for a few days.

I've just tested the probe with different battery combinations and it does not seem to make any difference. So I'm led to believe that the unit is not reporting battery status.

For example:
22C and 2x New Batteries = 3.2V
Result: 1111 0000 0010 0010 0000 1111 1101 1101

I got the same result for 2.8V, 2.5V 2.15V and at less than 2V the unit wouldn't power up.

Hope this information is helpful

Thanks
Lucy

zuckschwerdt, 2023-01-30:
Yes, thanks. I guess I can implement and add a decoder module in the next days.

lucywinters, 2023-01-30:
> Yes, thanks. I guess I can implement and add a decoder module in the next days.

That would be absolutely wonderful. I do truly appreciate it and your patience with me.

zuckschwerdt, 2023-01-30:
No trouble, you did all the work. Happy to help.

zuckschwerdt, 2023-01-30:
Oh, did you ever capture a sample file? https://triq.org/rtl_433/ANALYZE.html
Can you upload one as zip here?

lucywinters, 2023-01-31:
Here's one I grabbed a while ago. Hope it helps.
[Sensor-Data-cu8.zip](https://github.com/merbanan/rtl_433/files/10542262/Sensor-Data-cu8.zip)

lucywinters, 2023-02-02:
Looks Great! - Though it is a little cold at my desk today :)
[image]

Thanks for incorporating this into rtl_433

zuckschwerdt, 2023-02-03:
Thanks for testing and the feedback.

If you someday spot a pattern in the "Unknown value" let us know, it's taking up space in the packets so it should be important :)

closed 2023-02-04 by zuckschwerdt