merbanan/rtl_433 #2325: Bresser 6-1 Rain problem

MadMacMad, 2023-01-18:
hello,
well after a few day's i'll got strange numbers for rain every 15 seconds...
Rain      : 232.8 mm <- that's the real value
15 seconds later
Rain      : 69917.6 mm
15 seconds later normal again
Rain      : 232.8
15 seconds later
Rain      : 69917.6 mm

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-01-18 07:29:49+0100
model     : Bresser-6in1 id        : 55879a19
channel   : 0            Sensor type: 1            Wind Gust : 0.0 m/s
Wind Speed: 0.0 m/s      Direction : 198           Rain      : 232.8 mm
Startup   : 1            Flags     : 1             Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-01-18 07:30:01+0100
model     : sencor       count     : 1             num_rows  : 1
rows      : 
len       : 240          data      : 55555555555516ea32862ac3cd0c8c7fffff8cc44b3744fff81f55000000
codes     : {240}55555555555516ea32862ac3cd0c8c7fffff8cc44b3744fff81f55000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-01-18 07:30:01+0100
model     : Bresser-6in1 id        : 55879a19
channel   : 0            Temperature: -3.4 C       Humidity  : 89
Sensor type: 1           Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s
Direction : 198          Rain      : 69917.6 mm    UV        : 0.0
Startup   : 1            Flags     : 0             Integrity : CRC
_

this is my config:
   gain          0
   frequency     868.30M
   hop_interval  600
   ppm_error     0
   sample_rate   1000000
   report_meta   time:tz
   protocol 172 # Bresser Weather Center 6-in-1 (newer 5-in-1)
   protocol 173 # Bresser Weather Center 7-in-1
   output mqtt://192.168.178.142:1883,user=xxx,pass=xxx,retain=0,devices=rtl_433/bresser51

3 day's ago all worked fine.... since the temp has dropped .... i'll get this strange  >Rain      : 69917.6 mm < value ...

someone know what's wrong or what i'll can do?
many thx

MadMacMad, 2023-01-18:
invested more into....
took the outdoor unit indoors...
and voilla.... work normal again when over 0.0°C
took oudoors again... -2.1°C and the error with wrong rain every 10 seconds (Rain : 69917.6 M.:smoker:)
indoor again... and working normal... 
there must be something wrong with the code when it becomes minus degrees

matthias-bs, 2023-01-18:
Hi,
This is a known issue, see https://github.com/merbanan/rtl_433/issues/1214#issuecomment-1346856949 - there is already a proposed fix.

MadMacMad, 2023-03-28:
hello,
is there any update on this issue?
did not found a answer in #1214 
have installed the last and my Station still reports Rain : 69917.6 when Temp is below 0°
thx

gdt, 2023-09-26:
I'm closing this as a dup of #1214.  Please  test with latest git master and post in that bug.

closed 2023-09-26 by gdt