merbanan/rtl_433 #2326: Protmex PT3390A

pjg142000, 2023-01-18:
I purchased an outdoor weather station with nice colour display from aliexpress. It's labeled Protmex PT3390A (see pictures).
rtl_433 appears to identify it as follows, but does not display any other sensor values than the temperature:

'
time: "2023-01-18 08:13:24"
model: "Altronics-X7064"
id: 1397
channel: 4
battery_ok: 1
temperature_C: 20.05556
humidity: 1
mic: "CHECKSUM"
'
So portions of the message structure seam to be understood by rtl_433.
Any chance to have this weatherstation decoded in the future. 

https://de.aliexpress.com/item/1005005043423037.html?spm=a2g0o.order_list.order_list_main.11.6eab5c5fRK8Dhw&gatewayAdapt=glo2deu

merbanan, 2023-01-18:
Hi, follow these instructions and we might be able to help:

https://github.com/merbanan/rtl_433/blob/master/docs/ANALYZE.md

zuckschwerdt, 2023-01-18:
Also #2300 might work, in any case @ProfBoc75 can surely help.

dzejPL, 2023-01-19:
Hi
I have Jula / Marquant Weather Station like [this](https://www.jula.pl/catalog/gospodarstwo-domowe/sprzet-rtv/termometry-i-stacje-pogodowe/stacje-pogodowe/bezprzewodowa-stacja-pogodowa-014331/) The sensor in this station is the same (PT3390A) - so i tried to get data from that. Using Emax decoder from #2300 i got bellow - so You can try it. 

time      : 2023-01-19 15:41:11
model     : Emax-W6      id        : ea5
Channel   : 4            Battery_OK: 1             Temperature_F: 31.2
Humidity  : 90           Wind avg speed: 0.0 km/h  Wind Direction: 21
Total rainfall: 265.2 mm UV Index  : 8             Lux       : 77100
Integrity : CHECKSUM

pjg142000, 2023-01-19:
Oh, that’s cool. I'll give it a try.
Thank you.

dzejPL, 2023-01-20:
Something wrong - yesterday this decoder Emax works - but today not....
I got below info - and decoder cannot extract info... 

[emax_decode] Preamble not found
[emax_decode] Found row: 1
[emax_decode] Length check fail
[emax_decode] Preamble not found
[emax_decode] Preamble not found
[emax_decode] Found row: 4
[emax_decode] Length check fail
[emax_decode] Preamble not found
[emax_decode] Preamble not found
[emax_decode] Found row: 7
[emax_decode] Length check fail
[emax_decode] Preamble not found
[pulse_slicer_pcm] Emax W6, rebrand Altronics x7063/4, Optex 990040/50/51, Orium 13093/13123, Infactory FWS-1200, Newentor Q9, Otio 810025 Weather sensor
codes     : {27}fffc000, {185}aaaaaaaaaacaca54aa04ea5204c45b01010267062f10000, {144}80a0c0e101220222426282a2c2f58595a000, {33}fffff0000, {185}aaaaaaaaaacaca54aa04ea5204c45b01010267062f10000, {144}80a0c0e101220222426282a2c2f58595a000, {33}fffff0000, {185}aaaaaaaaaacaca54aa04ea5204c45b01010267062f10000, {144}80a0c0e101220222426282a2c2f58595a000

ProfBoc75, 2023-01-20:
> Something wrong - yesterday this decoder Emax works - but today not.... I got below info - and decoder cannot extract info...
>
> [emax_decode] Preamble not found [emax_decode] Found row: 1 [...]

Strange data length, too short, should be around 276 , at least 256 to get the 32 bytes of data. And for sure  the checksum can't be found. Did your internal display station show the good values ? May be another sensor sending data at the same Time, this one is exactly every 30 seconds.

dzejPL, 2023-01-23:
Hi

Maybe its only problem with batteries - because today its works again - but today its much more sunny than last few days. So can be that the station have a more energy to send the data. 
But for sure its not summer - so maybe the UV and LUX on that station are meaning something different ? And thats why doesnt fit the params. 

Today received data

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-01-23 09:34:12
model     : Emax-W6      id        : ea5
Channel   : 4            Battery_OK: 1             Temperature_F: 32.3
Humidity  : 90           Wind avg speed: 0.0 km/h  Wind Direction: 19
Total rainfall: 268.2 mm UV Index  : 8             Lux       : 165160
Integrity : CHECKSUM

And data gets from local receiver

[image]

dzejPL, 2023-01-23:
Ohh - the UV index for sure are wrong - check latest readings

time      : 2023-01-23 09:36:12
model     : Emax-W6      id        : ea5
Channel   : 4            Battery_OK: 1             Temperature_F: 32.3
Humidity  : 90           Wind avg speed: 1.0 km/h  Wind Direction: 7
Total rainfall: 268.2 mm UV Index  : 4294967295    Lux       : 164480
Integrity : CHECKSUM

ProfBoc75, 2023-01-24:
> Ohh - the UV index for sure are wrong - check latest readings
> 
> time : 2023-01-23 09:36:12 model : Emax-W6 id : ea5 Channel : 4 Battery_OK: 1 Temperature_F: 32.3 Humidity : 90 Wind avg speed: 1.0 km/h Wind Direction: 7 Total rainfall: 268.2 mm UV Index : 4294967295 Lux : 164480 Integrity : CHECKSUM

Are you using the last official version that is now able to decode this weather station, my last version is now using UV INDEX in 5 bits instead of 4 in the previous version.

ProfBoc75, 2023-01-25:
Hi all,

Based on your results, we are facing to an issue regarding the way how the data are decoded.

Within emax.c , the frame is split into rows by using the gap=1200 setting, and we should expect, 3 rows with the same data.
Unfortunately, if the light is strong, this creating a false positive gap at UV LL positions :

AA 04 II IB 0T TT HH 0W WW 0D DD RR RR UV LL LL 04 05 06 07 08 09 10 11 12 13 14 15 16 17 xx SS 
                                                              ...   10 00 00 04 05 ....
Up to 6 consecutive zeros are considered as a gap by rtl_433 decoder and the data is split into more rows than expected.
By removing the gap setting, the issue is gone, it also reduces the decoding success by 3, since only the first 32 bytes will be decoded, no more 3 rows to be analyzed, but no more gap issue.

I created a new PR to add these updates #2346 , can you please test with this new version and let us know.
Thx.

ProfBoc75, 2023-01-25:
Hi,

PR #2346 created.
Fix Emax LUX decoding, but also some errors at UV Index, Wind Direction,...
This new version is decoding only the first data without any gap, even if the LUX reached the max of 327.6 KLux 
The 2 others data packets, same as first one, will be taken into account later this week to improve the success of decoding this device and it'll replace the lost of gap feature.

To be continued ...

dzejPL, 2023-01-26:
Hi

Thanks - right now its much better :)

[image]

ProfBoc75, 2023-02-06:
Hi all, 
The PR #2346 was merged today, please check and report any issue here since it's a new way to decode the 3 packets.
It's working well for me 👍 and better than previous version :)
Thx.

merbanan, 2023-08-03:
Closing, seems fixed.

closed 2023-08-03 by merbanan