merbanan/rtl_433 #2327: Brandnew Wireless Smoke Detector GS558 does not show any decoded data

riegelbrau, 2023-01-18:
I bought a wireless smoke detector GS558 which is branded as from SITERWELL, because I read in the example .conf that it should be supported by protocol 86. I put the batteries in the detector should be activated. Running rtl_433 -d 1 -v -R 86 -F kv from the Linux shell does not show any data. I tried my two different receivers, where -d 1 is the Nooelec, NESDR SMArt v5 and -d 0 is a cheap blue one.
I the ran this command rtl_433 -d 1 -v -A -R 86 -F kv -w  GS558_out1.cu8 but the zipped version is still 17,9 mb in size.
I cannot find any hints, how to run rtl_433 for ths device. 

What can I do, check or test?

Thank you!
Christoph

zuckschwerdt, 2023-01-18:
The option `-w` writes all received data out. You'd want to only snapshot the actual events with `-S unknown`
Did you trigger the GS558? It will only transmit in programming mode and on alarm. There is no e.g. keep-alive.

riegelbrau, 2023-01-18:
Hello,

thanks for the hints. I was just playing around with the device and read 
the manual again in detail. Then I pressed the button for test mode and now 
I get some decodes via mqtt. I check again during the next days and will 
update or better close this issue.

Gruß Christoph

riegelbrau, 2023-01-19:
Ok, the sensor is sending, so I close this issue.
I will ask in the Google group for some experience from other users how to get reasonable information from the smoke detector like alarm (of course) and batteryevel or battery_ok.

Thanks and regards
Christoph

closed 2023-01-19 by riegelbrau

zuckschwerdt, 2023-01-19:
Glad it worked!

There is no advanced info transmitted. It's just unit and group number (an alarm) or those interleaved with empty packets (learn mode).
S.a. https://github.com/merbanan/rtl_433/blob/master/src/devices/smoke_gs558.c#L13