merbanan/rtl_433 #2332: Cotech 36-7959 / Ragova FT-0310 - Rain Not valid

manuel-held, 2023-01-20:
Hello, 

my Ragova FT-0310 Base Station ist showing no Rain 
[image]

But rtl_433 does:

[image]

I use the Version 22.11-40-g676a4042 branch master at 202301182158

Any hints?

zuckschwerdt, 2023-01-20:
Common pitfall actually ;) The values from rtl_433 are "current total" -- you need to aggregate yourself (in some dashboard software maybe), e.g. hourly, daily, … i.e. save the "midnight" value, subtract from the next "midnight" value and you got "daily".

closed 2023-01-20 by zuckschwerdt

manuel-held, 2023-01-20:
Ahh damn. Thank you :)