merbanan/rtl_433 #2348: rtl_433 does not exit when rtl_tcp host reboots

emilburzo, 2023-01-28:
I'm using `rtl_433` via docker using a `rtl_tcp` instance running on a separate host (due to having better reception there).

However, when the `rtl_tcp` host goes down (e.g. reboot), `rtl_433` doesn't seem to recover.

The last log line is:

```
Async read stalled, exiting!
```
But the process never exits.

Any ideas what might be wrong?

### rtl_433 arguments

```
 -d rtl_tcp://XXXXXXX:1234 \
 -Y autolevel -Y minmax -Y magest -M level -M time:usec \
 -F "mqtt://YYYYYYY:1883,user=XXXX,pass=XXXX,retain=0"
```

gdt, 2023-09-26:
Please test with up-to-date git master.  Use two bare-metal hosts and no docker.   Turn on debugging.   Also see the `-D restart` option.  There might be a bug here, but to make progress you will have to narrow down where it is.

rct, 2023-09-26:
Just saw this.  If you see `Async read stalled, exiting!` but the process hangs and doesn't exit, this might be #2426

zuckschwerdt, 2024-01-18:
This should be resolved.

closed 2024-01-18 by zuckschwerdt