merbanan/rtl_433 #2353: LaCrosse TX145WSDTH

Jefronty, 2023-01-30:
I am able to read partial data from a [LaCrosse TX145WSDTH](https://www.lacrossetechnology.com/products/tx145wsdth), shown as model='LaCrosse-TX141W', but none of the wind data is included.  This sensor reports wind speed and direction along with humidity and temperature.  I am seeing the humidity and temperature, as well as battery status, via rtl_433, but no wind data.

zuckschwerdt, 2023-01-30:
The model is correct. It has alternating packets with either temp or wind.
Please capture some samples: https://triq.org/rtl_433/ANALYZE.html

Jefronty, 2023-01-31:
@zuckschwerdt thanks, I'm seeing it now.  The feed I was looking at was filtering out things with the same time value.

closed 2023-01-31 by Jefronty