merbanan/rtl_433 #2354: Elster TPR6 Water Meter - Wavenis protocol

Toniob, 2023-01-31:
Following my issue here : https://github.com/weetmuts/wmbusmeters/issues/318

I’m trying to get the informations sent by my water meter.

Here are the signals I get this morning.
With the command : `rtl_433 -s 2048k -f 868.5M -Y autolevel -Y magest -M noise -M level -Y classic -S unknown`
[2048k-cu8.zip](https://github.com/merbanan/rtl_433/files/10545820/2048k-cu8.zip)

and later, with : `rtl_433 -s 250k -f 869M -Y autolevel -Y magest -M noise -M level -Y classic -S unknown`
[1000k-cu8.zip](https://github.com/merbanan/rtl_433/files/10545823/1000k-cu8.zip)

Thanks for your help

gdt, 2023-09-26:
Could you retest with git master, and update us on the status of wmbusmeters?   I am not sure if there is something actionable, and someone is going to have to dig in to make progress.

Toniob, 2023-09-27:
Hi. I don’t think someone will work on that on the wmbusmeters side. And I clearly don’t have the skills to do it. It may be better if you close this issue.

gdt, 2023-09-27:
Thanks - feel free to open an issue if there's a way to make progress.  I don't mean to discourage you from participating -- we just have 250 open issues and it's hard to sort through and understand.

closed 2023-09-27 by gdt