merbanan/rtl_433 #2356: Somfy RTS Issue

Ghils1, 2023-02-01:
Hello,

The remote controllers for my Garage Motors are not decoded.
I'am not sure that it fits the protocol described by PushStack.

Attached some captures from 2 different RC. 
On each RC the same button was used for every capture.
[traces_somfy.zip](https://github.com/merbanan/rtl_433/files/10561335/traces_somfy.zip)

Here is a link to the post I opened in the google group: [here](https://groups.google.com/g/rtl_433/c/WmTP1uzEifs)

Many Thanks.
Regards.
_G.

zuckschwerdt, 2023-02-01:
> The captures are good. We get
> 
> {267}e1e1e1e1e1e1e1e1e1e1e1e1fccd3532d532d55552d52d532accccb535555552d200
> {218}e1e1e1e1e1e1fccd3532d532d55552d52d532accccb4d555554d2c00
> {219}e1e1e1e1e1e1fccd3532d532d55552d52d532accccb4b555554b4a00
> {219}e1e1e1e1e1e1fccd3532d532d55552d52d532accccb355555534d200
> {219}e1e1e1e1e1e1fccd3532d532d55552d52d532accccd355555534b200
> 
> shifted by one bit this is
> 
> 70f0f0f0f0f0f0f0f0f0f0f0fe669a996a996aaaa96a96a99566665a9aaaaaa9690
> 70f0f0f0f0f0fe669a996a996aaaa96a96a99566665a6aaaaaa6960
> 70f0f0f0f0f0fe669a996a996aaaa96a96a99566665a5aaaaaa5a50
> 70f0f0f0f0f0fe669a996a996aaaa96a96a995666659aaaaaa9a690
> 70f0f0f0f0f0fe669a996a996aaaa96a96a995666669aaaaaa9a590
> 
> The decoder looks for a fixed sync of  f0f0f0f0f0f0f0ff
> But the best we get here is   0f0f0f0f0f0fe
> 
> So this is just an instance where the first databit is zero instead of the assumed one. Also the sync search is too harsh.

Thanks, I'll take a look to see if it can be easily fixed.

zuckschwerdt, 2023-02-02:
The command is
`rtl_433 -X "n=somfy-test,m=OOK_PCM,s=604,l=604,t=40,g=3000,r=10000" file.cu8`

The problem here is not a change in protocol. It looks somewhat different, but really just the preamble is shorter and the bitrate slightly off.

closed 2023-02-02 by zuckschwerdt

zuckschwerdt, 2023-02-02:
Should be fixed now.

Ghils1, 2023-02-03:
Hello Christian,

I compiled last commit and it works now, many Thanks!!!

```
rtl_433 version 22.11-76-g425726f8 branch master at 202302022018 inputs file rtl_tcp RTL-SDR with TL
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.163684s
model     : Somfy-RTS    id        : 4C2A31
Control   : ? (4)        Counter   : 619           Retransmission: 1
Integrity : CHECKSUM
[wam@Opium:~ $

```

After digging to the code, PushStack's wordpress, and do some additional tests I've got several questions:

1. Do you agree that this impementation of RTS protocol, differs in PushStack description with:

- Half Symbol Width is longer: aprrox 640 us, instead 604
- Hardware sync part, if we consider it starting after the first long silence;  can be wrote as: 
`
^^^____^^^^____^^^^____^^^^____^^^^____^^^^____^^^^____^^^^____^^^^____^^^^____^^^^____^^^^____`
(only 3 for last one)
`
- Software sync.:  approx 4800 us high, followed by 1/2 symbol width low(~640 us).
- Retransmit pattern is:
`
^^^^____^^^^____^^^^____^^^^____^^^^____^^^^____
`
- There are 2 extra bytes in the payload versus the "standard protocol".
- No others changes in the protocol, scrumbling and checksum calculation are still the same.

2. About theses extra 2 bytes, are they displayed, by the new version of decoder? 
    My feeling is not (4C2A31), could you confirm?

Thanks to you again.
Regards.
_G.

zuckschwerdt, 2023-02-03:
>     1. Do you agree that this impementation of RTS protocol, differs in PushStack description with:

Note that I have not read "PushStack" and I'm not the author of this decoder.

>     * Half Symbol Width is longer: aprrox 640 us, instead 604

Around 621 µs I would say.

>     * Hardware sync part, if we consider it starting after the first long silence;  can be wrote as:

The original preamble was 65 half-bits of:
`^^^^....^^^^....^^^^....^^^^....^^^^....^^^^....^^^^....^^^^^^^^.`

The "new" preamble is shorter (one `^^^^....` missing) and the first is `^^^....`, so 56 half-bits of:
`^^^....^^^^....^^^^....^^^^....^^^^....^^^^....^^^^^^^^.`

>     * Software sync.:  approx 4800 us high, followed by 1/2 symbol width low(~640 us).

Yes, that's `^^^^^^^^.` seen above.

>     * Retransmit pattern is:

No, the above is the retransmit. The first transmit is:
`^^^....^^^^....^^^^....^^^^....^^^^....^^^^....^^^^....^^^^....^^^^....^^^^....^^^^....^^^^....^^^^^^^^.`

Drop a cu8 on https://triq.org/pdv/ -- zoom with scroll and inspect.

>     * There are 2 extra bytes in the payload versus the "standard protocol".

2 or 3, not sure. It was 56 bit before, it looks like 80 bit now.

>     * No others changes in the protocol, scrumbling and checksum calculation are still the same.

No idea.

>     2. About theses extra 2 bytes, are they displayed, by the new version of decoder?

No, there is no output.

Velocet, 2023-05-01:
Maybe i can help with some info i got from reversing io-homecontrol. There are at least 4 different protocols:
RTD (Domis RTD - Actuator > seems related to Somfy Protect)
RTDS (Domis RTD - Sensor = Siren)
RTN (seems like a mix of io-homecontrol and RTS?)
RTS (Radio Technology Somfy)
RTW (associated to "My Higgins ASA"?)

When you glance over the patents you will find two protocols and they mention two different types: one with 56 bits and one with 80 bits.

@Ghils1 / @zuckschwerdt 
You can also help reverse the Somfy RTS Kernel driver:
[Somfy-RTS-KernelModule-4.14.223.zip](https://github.com/merbanan/rtl_433/files/11367214/Somfy-RTS-KernelModule-4.14.223.zip)

Ghils1, 2023-05-04:
Hello Velocet

First sorry for the delay.

Thanks a lot for this info, I'll have a look to this patents.

About helping, to begin, I can furnish more traces, and also try to resumed what I noticed and where I'm stuck.
Note that I have 2 kind of Remote Controllers, and it seems the implementation of protocol is not exactly the same in terms of restransmission between both RCs. Both of RC starts the motors...

My goad is to is to be able to reproduce the signal, based on arduino and a RFM69 RF transmitter.
I'am able to generate trafic that is recognized by the version committed by Christian.
But unfortunately this dont open my gate...
My main issue is  that I didn't find the logic beetween frames in a restransmission.

Regards.

MarkGodwin, 2023-10-27:
> My goad is to is to be able to reproduce the signal, based on arduino and a RFM69 RF transmitter. I'am able to generate trafic that is recognized by the version committed by Christian. But [...]

Sorry, I know this is not the right place for me to post this, but @Ghils1 if you're still looking for an example of generating Somfy codes with an RFM69HCW, I recently managed this (with the 7-byte protocol, but this can easily be extended) [here](https://github.com/MarkGodwin/somfy_pico/blob/master/firmware/) in `radio.cpp` and `commandQueue.cpp`. It's using a Pi Pico, but translating to arduino libraries shouldn't be hard.

I'm working on adding receive support which lead me here.

Ghils1, 2023-11-05:
Hello
@MarkGodwin : thank for you help, i looked to your project and it seems nice.
Thanks again, because your comment pushed me to take time to go on...

About Tx part I use an Arduino + RFM69HCW + [the Etimou Somfy RTS Project](https://github.com/etimou/SomfyRTS).
It's only 7 Bytes protocol version, but it works good, and correctly decoded by the rtl_433 decoder.
Based on this, I develloped a little ruby script to analyse output of the flex decoder. 
And happily I found same results than rtl_433 somfy_rts decoder.

About the 10 Bytes protocol, I did new tests with a [Keypop 2 RTS](https://boutique.somfy.fr/keypop-2-rts.html). 

1.  With this RC, there is no silence between restransmits. So the rtl_433 decoder gets first frame, correctly decode it, but retransmissions are not seens.
Here is the output for the decoder:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.324092s
model     : somfy-test   count     : 1             num_rows  : 1             rows      :
len       : 9            data      : f80
codes     : {9}f80
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.363628s
model     : somfy-test   count     : 1             num_rows  : 1             rows      :
len       : 488          data      : e1e1e1e1e1e1e1e1e1e1e1e1fe669a99a999a9a6655a669a95aaaa6a9aaaaaa9591e1e1e1e1e1e1fe669a99a999a9a6655a669a95aaaa6a9aaaaaa9590
codes     : {488}e1e1e1e1e1e1e1e1e1e1e1e1fe669a99a999a9a6655a669a95aaaa6a9aaaaaa9591e1e1e1e1e1e1fe669a99a999a9a6655a669a95aaaa6a9aaaaaa9590
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.363628s
model     : Somfy-RTS    id        : 478DE1
Control   : ? (4)        Counter   : 122           Retransmission: 1         Integrity : CHECKSUM

And now the output for the same cu8 file, of my script where retransmit (only one here) is displayed:

a4 f5 0 7a e1 8d 47 84 84 1d aa 15 CheckSum: 12	  
a4 f5 0 7a e1 8d 47 84 84 1d CheckSum: 12

So my decoder seems to work but dont know what to do with 3 extra bytes.
Also the checksum caculated by my decoder gives 12.

2. In order to try understanding the 3 extra-bytes I made 100 successive press on the Keypop.
It seems that 2 last extra-bytes are fix, but fist one change every 16 push.
But I dont understood how it changes...

a4 f6 0 79 e1 8d 47 84 84 1d CheckSum: 12
a4 f5 0 7a e1 8d 47 84 84 1d CheckSum: 12
a4 f4 0 7b e1 8d 47 84 84 1d CheckSum: 12
a4 f3 0 7c e1 8d 47 84 84 1d CheckSum: 12
a4 f2 0 7d e1 8d 47 84 84 1d CheckSum: 12
a4 f1 0 7e e1 8d 47 84 84 1d CheckSum: 12
a4 f0 0 7f e1 8d 47 84 84 1d CheckSum: 12
a4 f0 0 80 e1 8d 47 7b 84 1d CheckSum: 12
a4 f1 0 81 e1 8d 47 7b 84 1d CheckSum: 12
a4 f2 0 82 e1 8d 47 7b 84 1d CheckSum: 12
a4 f3 0 83 e1 8d 47 7b 84 1d CheckSum: 12
a4 f4 0 84 e1 8d 47 7b 84 1d CheckSum: 12
a4 f5 0 85 e1 8d 47 7b 84 1d CheckSum: 12
a4 f6 0 86 e1 8d 47 7b 84 1d CheckSum: 12
a4 f7 0 87 e1 8d 47 7b 84 1d CheckSum: 12
a4 f8 0 88 e1 8d 47 7b 84 1d CheckSum: 12
a4 f9 0 89 e1 8d 47 7b 84 1d CheckSum: 12
a4 fa 0 8a e1 8d 47 7b 84 1d CheckSum: 12
a4 fb 0 8b e1 8d 47 7b 84 1d CheckSum: 12
a4 fc 0 8c e1 8d 47 7b 84 1d CheckSum: 12
a4 fd 0 8d e1 8d 47 7b 84 1d CheckSum: 12
a4 fe 0 8e e1 8d 47 7b 84 1d CheckSum: 12
a4 ff 0 8f e1 8d 47 7b 84 1d CheckSum: 12
a4 f1 0 90 e1 8d 47 6a 84 1d CheckSum: 12
a4 f0 0 91 e1 8d 47 6a 84 1d CheckSum: 12
a4 f3 0 92 e1 8d 47 6a 84 1d CheckSum: 12
a4 f2 0 93 e1 8d 47 6a 84 1d CheckSum: 12
a4 f5 0 94 e1 8d 47 6a 84 1d CheckSum: 12
a4 f4 0 95 e1 8d 47 6a 84 1d CheckSum: 12
a4 f7 0 96 e1 8d 47 6a 84 1d CheckSum: 12
[... 51 lines cut ...]
a4 fe 0 ca e1 8d 47 3f 84 1d CheckSum: 12
a4 ff 0 cb e1 8d 47 3f 84 1d CheckSum: 12
a4 f8 0 cc e1 8d 47 3f 84 1d CheckSum: 12
a4 f9 0 cd e1 8d 47 3f 84 1d CheckSum: 12
a4 fa 0 ce e1 8d 47 3f 84 1d CheckSum: 12
a4 fb 0 cf e1 8d 47 3f 84 1d CheckSum: 12
a4 f5 0 d0 e1 8d 47 2e 84 1d CheckSum: 12
a4 f4 0 d1 e1 8d 47 2e 84 1d CheckSum: 12
a4 f7 0 d2 e1 8d 47 2e 84 1d CheckSum: 12
a4 f6 0 d3 e1 8d 47 2e 84 1d CheckSum: 12
a4 f1 0 d4 e1 8d 47 2e 84 1d CheckSum: 12
a4 f0 0 d5 e1 8d 47 2e 84 1d CheckSum: 12
a4 f3 0 d6 e1 8d 47 2e 84 1d CheckSum: 12
a4 f2 0 d7 e1 8d 47 2e 84 1d CheckSum: 12
a4 fd 0 d8 e1 8d 47 2e 84 1d CheckSum: 12
a4 fc 0 d9 e1 8d 47 2e 84 1d CheckSum: 12
a4 ff 0 da e1 8d 47 2e 84 1d CheckSum: 12
a4 fe 0 db e1 8d 47 2e 84 1d CheckSum: 12
a4 f9 0 dc e1 8d 47 2e 84 1d CheckSum: 12

Now I have to adapt the Etimou's Project to generate same 10-Bytes frames, and do some tests.
The issue is that without understanding the first one of the  "extra 3 bytes", it will probably not work more than 16 times.
Also I'am reading the code of the somfy rts, it should be interesting to have the "3 Extra Bytes" and also the the restranmission for 10 Bytes protocol version. If somebody is interested, i can share my traces and devlopment.

Regards

commit 425726f8 (2023-02-02): Fix Somfy-RTS for wrong bitrate (closes #2356)