merbanan/rtl_433 #2364: Govee H5054

sombrero2003, 2023-02-04:
Hello,
I think there may still be issues with the decoding of Govee H5054
I am running the following rtl_433 version 22.11-78-g8286df53 branch master at 202302041443

Pressing the button on the sensor shows the following 

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:14:18
model     : Govee-Water  id        : 34004
event     : Button Press Raw Code  : 7b2b0505075a  Integrity : PARITY
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:14:18
model     : Vauno-EN8822C ID        : 0
Channel   : 1            Battery   : 1             Temperature: 0.0 C        Humidity  : 0 %           Integrity : CHECKSUM
*** Saving signal to file g004_433.92M_250k.cu8 (246376 samples, 524288 bytes)
```

When the sensor is alarming, I am getting what looks like the same response but repeating itself
```
time      : 2023-02-04 09:17:37
model     : Govee-Water  id        : 34004
event     : Water Leak   Raw Code  : 7b2b0404075a  Integrity : PARITY
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:37
model     : Vauno-EN8822C ID        : 0
Channel   : 1            Battery   : 1             Temperature: 0.0 C        Humidity  : 0 %           Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:38
model     : Govee-Water  id        : 34004
event     : Water Leak   Raw Code  : 7b2b0404075a  Integrity : PARITY
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:38
model     : Vauno-EN8822C ID        : 0
Channel   : 1            Battery   : 1             Temperature: 0.0 C        Humidity  : 0 %           Integrity : CHECKSUM
*** Saving signal to file g005_433.92M_250k.cu8 (517520 samples, 1048576 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:42
model     : Govee-Water  id        : 34004
event     : Water Leak   Raw Code  : 7b2b0404075a  Integrity : PARITY
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:42
model     : Vauno-EN8822C ID        : 0
Channel   : 1            Battery   : 1             Temperature: 0.0 C        Humidity  : 0 %           Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:43
model     : Govee-Water  id        : 34004
event     : Water Leak   Raw Code  : 7b2b0404075a  Integrity : PARITY
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:43
model     : Vauno-EN8822C ID        : 0
Channel   : 1            Battery   : 1             Temperature: 0.0 C        Humidity  : 0 %           Integrity : CHECKSUM
*** Saving signal to file g006_433.92M_250k.cu8 (517387 samples, 1048576 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:47
model     : Govee-Water  id        : 34004
event     : Water Leak   Raw Code  : 7b2b0404075a  Integrity : PARITY
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:47
model     : Vauno-EN8822C ID        : 0
Channel   : 1            Battery   : 1             Temperature: 0.0 C        Humidity  : 0 %           Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:48
model     : Govee-Water  id        : 34004
event     : Water Leak   Raw Code  : 7b2b0404075a  Integrity : PARITY
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-04 09:17:48
model     : Vauno-EN8822C ID        : 0
Channel   : 1            Battery   : 1             Temperature: 0.0 C        Humidity  : 0 %           Integrity : CHECKSUM
*** Saving signal to file g007_433.92M_250k.cu8 (517487 samples, 1048576 bytes)
```

I have attached two relevant cu8 files.

[Archive.zip](https://github.com/merbanan/rtl_433/files/10608853/Archive.zip)

zuckschwerdt, 2023-02-04:
Is this releated to any of the many Govee issues we already have? https://github.com/merbanan/rtl_433/issues?q=Govee

What is wrong with the output above and what did you expect?

sombrero2003, 2023-02-04:
Thanks
I thought that the issue had been resolved by commit 4b6dc283824970b744d3f6e69a0d1172f68b33ae.

I was expecting the output to indicate that a water leak is detected when the sensor is alarming (see this [post](https://github.com/pbkhrv/rtl_433-hass-addons/issues/124)). 

Also I am not sure were the Vauno sensor is coming from, is rtl_433 getting confused between the two?
Thanks a lot.

zuckschwerdt, 2023-02-04:
Let's ping @aav7fl and @jasonmichalski from #2273 -- can you help?

sombrero2003, 2023-02-04:
Somehow I managed to miss the fact that the water leak was properly reported. 
This just leaves the battery unreported and the possible confusion with the Vauno sensor. Sorry for the confusion.

aav7fl, 2023-02-04:
> Let's ping @aav7fl and @jasonmichalski from #2273 -- can you help?

I'm afraid I probably won't be able to help too much.

I have a lot of stuff that is keeping me away from development right now.

Is there a chance that more codes needed to be added to the recent change to the new sensor revision, or perhaps certain signal types were missed? Or is this yet another additional revision?

I'm not sure if it's the best practice, but when I had fixed mine, I made sure to include a bunch of sample signals in the file comments for others to test against in the future if there were additional changes made.

aav7fl, 2023-02-04:
> Somehow I managed to miss the fact that the water leak was properly reported. 
> This just leaves the battery unreported and the possible confusion with the Vauno sensor. Sorry for the confusion.

No worries. Yea, the battery events are sent separately. That meant I needed to adjust some of my automations because if I get a water leak alert, it would wipe out the old battery data when I overwrote the entire payload.

zuckschwerdt, 2023-02-04:
> and the possible confusion with the Vauno sensor.

That is indeed strange. The Govee has PWM 440/940 and the Vauno PPM 2000/4000 which is a major difference and should never match?

Can you grab a sample that shows this on reading back? https://triq.org/rtl_433/ANALYZE.html

jasonmichalski, 2023-02-04:
Can you try popping the batteries out and plugging them back in?  The battery reports are not sent out regularly like they were in earlier revisions of this sensor.

sombrero2003, 2023-02-04:
The battery is indeed read properly after I played with the battery. Good to know and very sorry about the confusion.

If I restrict the protocols read to 192,  only the govee water leak sensor shows up. If I don't then pressing the button shows both Govee and Vauno. I have attached a sample
[g001_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/10609527/g001_433.92M_250k.cu8.zip)

It is being decoded as 
```
[Input] Test mode active. Reading samples from file: g001_433.92M_250k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.118568s
model     : Govee-Water  id        : 31523
event     : Button Press Raw Code  : 84dc0505074a  Integrity : PARITY
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.118568s
model     : Vauno-EN8822C ID        : 0
Channel   : 1            Battery   : 1             Temperature: 0.0 C
Humidity  : 0 %          Integrity : CHECKSUM
```

zuckschwerdt, 2023-02-04:
There are now two checks for false positives in Vauno-EN8822C, so that should be fixed.

commit 31dc48ca (2023-02-04): Fix Vauno-EN8822C false positives (closes #2364)
closed 2023-02-04 by zuckschwerdt