merbanan/rtl_433 #2367: "House" code missing from JSON output?!

sheilbronn, 2023-02-05:
I noticed that a decoded house is missing from the JSON output (at least for the Generic-Remote sensor):

```
$ rtl_433 -M house
rtl_433 version 22.11-21-gdc95819b branch master at 202212100004 inputs file rtl_tcp RTL-SDR
...
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-05 16:30:45
model     : Generic-Remote                         House Code: 35974
Command   : 131          Tri-State : X010X0ZXX001
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-05 16:30:45
model     : Generic-Remote                         House Code: 35974
Command   : 131          Tri-State : X010X0ZXX001
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
...
```

as opposed to the output delivered as JSON:

```
$ rtl_433 -M house -F json
rtl_433 version 22.11-21-gdc95819b branch master at 202212100004 inputs file rtl_tcp RTL-SDR
...
{"time" : "2023-02-05 16:33:19", "model" : "Generic-Remote", "id" : 35974, "cmd" : 131, "tristate" : "X010X0ZXX001"}
{"time" : "2023-02-05 16:33:19", "model" : "Generic-Remote", "id" : 35974, "cmd" : 131, "tristate" : "X010X0ZXX001"}
...
```

Is there some option I'm missing and/or is this on purpose?

(I've built an extended MQTT bridge with auto-discovery around rtl_433 and noticed this during testing.)

zuckschwerdt, 2023-02-05:
What you see in the KV output are "pretty names". "House Code" is `id`.

closed 2023-02-05 by zuckschwerdt

merbanan, 2023-02-05:
House Code in console output is the same as id in the json output. And this is by design.

sheilbronn, 2023-02-05:
Oh I feel sorry! I overlooked the obvious. Offcourse, you're right. Thanks for your good work!