merbanan/rtl_433 #2368: Temperature output for Klimalogg

ckoring, 2023-02-06:
Hi,

I just started using rtl_433 to grab the data from several Klimalogg sensors. As output I'm using a connection to my Influx database. I've now recognised that the temperature values are transfered as strings (e.g. "20.6 C") and not as floats in contrast to humidity values that are transfered as integers. For later using e.g. visualisation it is better not to mix up numerical value and the unit letter.

My suggestion is to change the definition of `static char *output_fields[]` in `klimalogg.c` (line 97) from "temperature_C" to just "temperature". According my local test this seems to influence the (influx) output as wished.

cheers ckoring

merbanan, 2023-02-06:
The key naming is part of the api so that can not be changed internally. But can you post your command line? We might need to filter the units in the use case you are using.

ckoring, 2023-02-06:
Hi,

I run rtl_433 via with this command:

`rtl_433 -R 150 -d 0 -f 868M -s 2400k -F "influx://URL:30818/api/v2/write?org=influxdata&bucket=test1,token=jdkcIQu4.....RK0aew=="`

I just changed the source code at one point (klimalogg.c at line 97), please have a look on the added screenshot. Red marking as it is on Github, data not pretty processible, Influx data explorer can show it as raw data only. Green marking with this minor change, data in Influx as double, so the dashboard can easily create simple diagrams. 

My little tweak works for me, but perhaps it can help others as well. As I'm not a programming expert unfortuneatly my contribution ends here.

Before I forgot, many thanks for this useful piece of software!

[rtl_433_influx_output](https://user-images.githubusercontent.com/124563375/217083031-62178a3f-5421-41fc-9bc1-1f41f49fbbd7.png)

merbanan, 2023-02-06:
As I said the key can not change, it is correct because we defined it that way. The possible issue might be in the influx output. I dont understand the code well enough to say that it is an issue.

zuckschwerdt, 2023-02-06:
A few points: the "output_fields" are for the first line (heading) CSV only and not used for anything else.
The Influx output does not use any format but "raw" `%f`: https://github.com/merbanan/rtl_433/blob/master/src/output_influx.c#L401
The model key in your screenshot is not what we output. 

You must have used some other (post-)processing or forwarding software/script to feed InfluxDB.

ckoring, 2023-02-22:
Sorry to get back to this topic again. I still belive that the output for Klimalogg devices is not correctly implemented because the temperatur is output as string e.g. `"21.1 C"` and not as float. I freshly clone the sources from Github and compiled it as explained in `BUILDING.md`. For evidence see CSV file ([klimalogg.csv](https://github.com/merbanan/rtl_433/files/10777512/klimalogg.csv)). RTL_433 run command was `rtl_433 -d 0 -f 868M -s 2400k -R 150 -T 30 -F csv:/home/pi/klimalogg.csv"`. No post-processing done! Output in JSON format and Influx looks pretty the same. As mentioned before my programming skills are too limited to make a sensefull suggestion for a change in the source code. 

Meanwile I found a workaround for me, but on the Influx side. It just convert the string value to float by using the internal Influx functions. Here's the relevant part of the flux query:
`|> filter(fn: (r) => r["_measurement"] == "Klimalogg_Pro")`
`|> filter(fn: (r) => r["_field"] == "temperature_C" or r["_field"] == "humidity")`
`|> map(`
`        fn: (r) =>`
`            ({r with _value:`
`                    if r._field == "temperature_C" then`
`                        float(v: strings.split(v: string(v: r._value), t: " C")[0])`
`                    else`
`                        float(v: string(v: r._value)),`
`            }),`
`    )`

regards

zuckschwerdt, 2023-02-23:
You are not using our source. We can't help you. Contact wherever you are getting your fork from please. Also tell them to stop directing people at our rtl_433 -- this is a nuisance.