merbanan/rtl_433 #2369: rtl_433 MQTT Auto Discovery not working

mebitek, 2023-02-06:
Hi,
I have this configuration:

```
frequency 868300000
frequency 433920000
hop_interval 60
protocol 172
protocol 12
output mqtt=http://core-mosquitto:1883,user=*****,pass=*****,retain=1,devices=rtl_433[/model][/id]
convert si
report_meta time:iso:tz:local
output json
```

and I can see in the logs my devices:

```
[devices] {"time" : "2023-02-06T14:14:07+0100", "model" : "Oregon-THGR810", "id" : 139, "channel" : 1, "battery_ok" : 1, "temperature_C" : 19.100, "humidity" : 75}
[devices] Tuned to 868.300MHz.
[devices] Allocating 15 zero-copy buffers
[devices] {"center_frequency" : 868300000, "frequencies" : [868300000, 433920000], "hop_times" : [60]}
[devices] {"time" : "2023-02-06T14:14:16+0100", "model" : "Bresser-6in1", "id" : 338715378, "channel" : 0, "battery_ok" : 1, "temperature_C" : 10.400, "humidity" : 74, "sensor_type" : 1, "wind_max_m_s" : 3.400, "wind_avg_m_s" : 3.200, "wind_dir_deg" : 22, "uv" : 0.000, "startup" : 1, "flags" : 0, "mic" : "CRC"}
[devices] {"time" : "2023-02-06T14:14:28+0100", "model" : "Bresser-6in1", "id" : 338715378, "channel" : 0, "sensor_type" : 1, "wind_max_m_s" : 3.400, "wind_avg_m_s" : 3.200, "wind_dir_deg" : 248, "rain_mm" : 360.400, "startup" : 1, "flags" : 1, "mic" : "CRC"}
```

however hte auto discovery is not creating anything with this config:

```
mqtt_host: core-mosquitto
mqtt_port: 1883
mqtt_user: ****
mqtt_password: ****
mqtt_retain: true
rtl_topic: "/rtt_433/+/events"
discovery_prefix: "homeassistant"
discovery_interval: 600
force_update: false
log_level: default
```

if I manually add the sensor in yaml file it works:

```
    - name: Bresser51 Temperature
      state_topic: rtl_433/Bresser-6in1/temperature_C
      unique_id: bresser51_temperature_C
      value_template: "{{ value | float }}"
      unit_of_measurement: "°C"
      device_class: temperature
```
could you please help me in the configuration process?

closed 2023-02-06 by mebitek

zuckschwerdt, 2023-02-06:
You have only activated MQTT "devices" output, not MQTT "events" output. The discovery script has nothing to work with.

mebitek, 2023-02-06:
I changed the configuration and now I can see the events, but not devices or entiteis created in hass