merbanan/rtl_433 #2375: Optex 990051

Mikayex, 2023-02-10:
I've an [Optex 990051](https://www.groupe-normand.com/shop/990051-station-meteo-professionnelle-2185#attr=80090,80093,80092,80091), the little brother of the Optex 990040 (which is the Emax EM3390W6 under the hood).
It is correctly recognized as an Emax weather station as the protocol is *almost* the same, everything is decoded correctly except UV and Lux sensor, as it doesn't have any! Also the internal reference, according to the documentation, is EM3551H-DCF.

Here is some decoded data:
```
model     : Emax-W6
id        : dde
Channel   : 4
Battery_OK: 1
Temperature_F: 16.7
Humidity  : 33
Wind avg speed: 0.0 km/h
Wind Direction: 99
Total rainfall: 60.2 mm
UV Index  : 15
Lux       : 0
Integrity : CHECKSUM
```

And here is some raw data:
```
ffff8000aaaaaaaaaacaca54aa04c52b064c34010101d4019e4001010004050607080910111213141516ec74ed0000
ffff8000aaaaaaaaaacaca54aa04c52b064c34010101d4019e400101000405060708091011121314151662ea630000
ffff8000aaaaaaaaaacaca54aa04c52b065348010101d4019e400101000405060708091011121314151613b6140000
ffff8000aaaaaaaaaacaca54aa04c52b065133010101d3019e40010100040506070809101112131415163ac53b0000
ffff8000aaaaaaaaaacaca54aa04c52b064e33010101ca019e40010100040506070809101112131415169c1b9d0000
ffff8000aaaaaaaaaacaca54aa04c52b065034010101cb019e40010100040506070809101112131415161194120000
ffff8000aaaaaaaaaacaca54aa04c52b065533010101d3019e400101000405060708091011121314151630bf310000
ffff8000aaaaaaaaaacaca54aa042143065532010101e601013001010004050607080910111213141516e54de60000
ffff8000aaaaaaaaaacaca54aa042143065732010101d80101300101000405060708091011121314151647a3480000
ffff8000aaaaaaaaaacaca54aa04214b065732010101da010130010100040506070809101112131415166ed46f0000
ffff8000aaaaaaaaaacaca54aa042143065932010101f10101300101000405060708091011121314151681f8820000
ffff8000aaaaaaaaaacaca54aa04df80063533010101d501014001010004050607080910111213141516d114d20000
ffff8000aaaaaaaaaacaca54aa04df80063133010101d9010140010100040506070809101112131415161f62200000
ffff8000aaaaaaaaaacaca54aa04df80062f340101014e0101400101000405060708091011121314151632e9330000
ffff8000aaaaaaaaaacaca54aa040f610645320101013701014001010004050607080910111213141516854a860000
ffff8000aaaaaaaaaacaca54aa046cea061634010101b601034001010004050607080910111213141516acabad0000
ffff8000aaaaaaaaaacaca54aa046cea060b35010101b6010340010100040506070809101112131415165d525e0000
ffff8000aaaaaaaaaacaca54aa046cea059651010101b50103400101000405060708091011121314151613ad140000
ffff8000aaaaaaaaaacaca54aa04df800631340130010101014001010204050607080910111213141516e17ee20000
ffff8000aaaaaaaaaacaca54aa04df80063334012e015c01014001010f04050607080910111213141516575c580000
ffff8000aaaaaaaaaacaca54aa04df80063534013601cf01014001010f040506070809101112131415166aec6b0000
ffff8000aaaaaaaaaacaca54aa04df80063633011601ed0101400101070405060708091011121314151645bd460000
ffff8000aaaaaaaaaacaca54aa04df80063534010101ed010140010100040506070809101112131415161d791e0000
ffff8000aaaaaaaaaacaca54aa04df80063533013801ed01014001010f0405060708091011121314151644e5450000
ffff8000aaaaaaaaaacaca54aa040f610648330123015e01014001010a040506070809101112131415160a260b0000
ffff8000aaaaaaaaaacaca54aa040f610648320101015e01014001010f040506070809101112131415162b292c0000
ffff8000aaaaaaaaaacaca54aa040f61064a320101015e01014001010e04050607080910111213141516b4b3b50000
ffff8000aaaaaaaaaacaca54aa040f610648320101015e010140010119040506070809101112131415169fa7a00000
ffff8000aaaaaaaaaacaca54aa040f61064a320101015e01014001011b04050607080910111213141516d9e5da0000
ffff8000aaaaaaaaaacaca54aa040f610648320101015e01014001010b0405060708091011121314151689838a0000
ffff8000aaaaaaaaaacaca54aa040f610648320101015e01014001010304050607080910111213141516ebddec0000
ffff8000aaaaaaaaaacaca54aa040f61064832010101d9010140010116040506070809101112131415164ccc4d0000
```

Before the integration of the Emax decoder, I was actually trying to decode my station (independently). Most of my findings matches more or less the W6 ones, except for the data after the rain bits :
`4 UV: 0 LUX: 0101 G: 0b CONSTANT: 04050607080910111213141516 CHECKSUM: 89838a` (my data)
`0 UV: 1 LUX: 0101 CONSTANT: 0405060708091011121314151617 CHECKSUM: 9dad9e` (data on W6 notes)

I found out that the `G` byte correspond to wind gust*1.5. In my example, 0x0b/1.5 = 11/1.5 = 7.3333... and the station displayed 7.3 km/h gust speed. It worked for every test I made.

The constant part is not the same between the 2 models (the W6 has one more byte, to compensate for the lack of `G` ?) Is it a good idea to differentiate the 2 models by that (in which case, I can open PR for that)

Also, a bit unrelated but for documentation purposes, the battery low flag triggers when voltage <=3.7 V

(cc @ProfBoc75 you may be interested ?)

ProfBoc75, 2023-02-12:
Hi @Mikayex , 

I'm looking to your raw samples and let you know later my findings.
I'm not surprise, this kind of weather station can be found with several options, with or without the Wifi for the display and with or without the UV/Lux on the weather station.
I was already facing that situation with an older Optex Weather Station, 990018 with UV & Lux ( a rebrand from AOK ref 5056 ), it was very close the Holman ws5029 (without UV and Lux) already decoded by rtl_433 .
I have rewritten a device driver for this older Optex/AOK , but I never shared it here, since the CRC / Checksum algorithm was never found, so lot of errors.

ProfBoc75, 2023-02-12:
My findings from your samples : 
- 
- before UV = 0  , you have "4" or "3" values , not always the same value, and not sure was it is related to ? --> any idea ?
- The number of bytes are the same between the both models, just the fixed values are shifted by one byte and the last fixed value if 0x16 for your model, and 0x17 for mine.
This means that an easy check at this byte position before the crc/cheksum can help to identify the model with / without the UV/Lux information.

About the Gust, I trust you, I can't compare with mine which does not provide this information.
I will update the emax.c driver accordingly.

Bytes | Preamble | Type | Channel | ID | Bat/Pairing | Temp C | Hum % | Wind KmH | Dir ° | Rain mm | UV | Lux | 04 | Gust | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 17 | xx | crc | yy | 0000
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
aa04c52b064c34010101d4019e4001010004050607080910111213141516ec74ed0000 | aa | 0 | 4 | c52 | b | 21.8 | 52% | 0 | 211 | 31.4 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | ec | 74 | ed | 0000
aa04c52b064c34010101d4019e400101000405060708091011121314151662ea630000 | aa | 0 | 4 | c52 | b | 21.8 | 52% | 0 | 211 | 31.4 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 62 | ea | 63 | 0000
aa04c52b065348010101d4019e400101000405060708091011121314151613b6140000 | aa | 0 | 4 | c52 | b | 22.2 | 72% | 0 | 211 | 31.4 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 13 | b6 | 14 | 0000
aa04c52b065133010101d3019e40010100040506070809101112131415163ac53b0000 | aa | 0 | 4 | c52 | b | 22.1 | 51% | 0 | 210 | 31.4 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 3a | c5 | 3b | 0000
aa04c52b064e33010101ca019e40010100040506070809101112131415169c1b9d0000 | aa | 0 | 4 | c52 | b | 21.9 | 51% | 0 | 201 | 31.4 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 9c | 1b | 9d | 0000
aa04c52b065034010101cb019e40010100040506070809101112131415161194120000 | aa | 0 | 4 | c52 | b | 22.0 | 52% | 0 | 202 | 31.4 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 11 | 94 | 12 | 0000
aa04c52b065533010101d3019e400101000405060708091011121314151630bf310000 | aa | 0 | 4 | c52 | b | 22.3 | 51% | 0 | 210 | 31.4 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 30 | bf | 31 | 0000
aa042143065532010101e601013001010004050607080910111213141516e54de60000 | aa | 0 | 4 | 214 | 3 | 22.3 | 50% | 0 | 229 | 0 | 30 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | e5 | 4d | e6 | 0000
aa042143065732010101d80101300101000405060708091011121314151647a3480000 | aa | 0 | 4 | 214 | 3 | 22.4 | 50% | 0 | 215 | 0 | 30 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 47 | a3 | 48 | 0000
aa04214b065732010101da010130010100040506070809101112131415166ed46f0000 | aa | 0 | 4 | 214 | b | 22.4 | 50% | 0 | 217 | 0 | 30 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 6e | d4 | 6f | 0000
aa042143065932010101f10101300101000405060708091011121314151681f8820000 | aa | 0 | 4 | 214 | 3 | 22.5 | 50% | 0 | 240 | 0 | 30 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 81 | f8 | 82 | 0000
aa04df80063533010101d501014001010004050607080910111213141516d114d20000 | aa | 0 | 4 | df8 | 0 | 20.5 | 51% | 0 | 212 | 0 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | d1 | 14 | d2 | 0000
aa04df80063133010101d9010140010100040506070809101112131415161f62200000 | aa | 0 | 4 | df8 | 0 | 20.3 | 51% | 0 | 216 | 0 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 1f | 62 | 20 | 0000
aa04df80062f340101014e0101400101000405060708091011121314151632e9330000 | aa | 0 | 4 | df8 | 0 | 20.2 | 52% | 0 | 77 | 0 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 32 | e9 | 33 | 0000
aa040f610645320101013701014001010004050607080910111213141516854a860000 | aa | 0 | 4 | 0f6 | 1 | 21.4 | 50% | 0 | 54 | 0 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 85 | 4a | 86 | 0000
aa046cea061634010101b601034001010004050607080910111213141516acabad0000 | aa | 0 | 4 | 6ce | a | 18.8 | 52% | 0 | 181 | 0.4 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | ac | ab | ad | 0000
aa046cea060b35010101b6010340010100040506070809101112131415165d525e0000 | aa | 0 | 4 | 6ce | a | 18.2 | 53% | 0 | 181 | 0.4 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 5d | 52 | 5e | 0000
aa046cea059651010101b50103400101000405060708091011121314151613ad140000 | aa | 0 | 4 | 6ce | a | 11.7 | 81% | 0 | 180 | 0.4 | 40 | 0101 | 00 | 0.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 13 | ad | 14 | 0000
aa04df800631340130010101014001010204050607080910111213141516e17ee20000 | aa | 0 | 4 | df8 | 0 | 20.3 | 52% | 9.4 | 0 | 0 | 40 | 0101 | 02 | 1.3 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | e1 | 7e | e2 | 0000
[... cut ...]
aa040f610648320101015e01014001010b0405060708091011121314151689838a0000 | aa | 0 | 4 | 0f6 | 1 | 21.6 | 50% | 0 | 93 | 0 | 40 | 0101 | 0b | 7.3 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 89 | 83 | 8a | 0000
aa040f610648320101015e01014001010304050607080910111213141516ebddec0000 | aa | 0 | 4 | 0f6 | 1 | 21.6 | 50% | 0 | 93 | 0 | 40 | 0101 | 03 | 2.0 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | eb | dd | ec | 0000
aa040f61064832010101d9010140010116040506070809101112131415164ccc4d0000 | aa | 0 | 4 | 0f6 | 1 | 21.6 | 50% | 0 | 216 | 0 | 40 | 0101 | 16 | 14.7 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 4c | cc | 4d | 0000

ProfBoc75, 2023-02-15:
@Mikayex : did you try my PR ? Let us know please to move forward and merge it if ok. 😉

Mikayex, 2023-02-15:
Sorry for the delay, I couldn't test sooner... The PR works great !

I've teardown the station to see the PCB and I suspect the 3/4 value (now back at 3) is because the UV port is unpopulated and the floating pins are sending random data... 

[image]
[image]

ProfBoc75, 2023-02-15:
@Mikayex , thanks for the feedback and the nice photos.
Yes ,it's the same EM3390 board without the UV sensor, upgrade may be possible I guess but need to find the good ref of UV/Lux component.
Some other details here : https://fccid.io/WEC-W6

Mikayex, 2023-02-15:
I guess this issue can be closed now...
Thanks for the help !

closed 2023-02-15 by Mikayex