merbanan/rtl_433 #2377: Help decoding Rainpoint Sensor using RXB6

garudaonekh, 2023-02-14:
Hi,
I am trying to decode the RainPoint sensor ( https://github.com/merbanan/rtl_433/blob/master/src/devices/rainpoint.c ) using low cost RXB6. I can get data from it but it's not correctly decoded. The result I get on my rtl_433 is `rainpoint_decode: : {96} aa 81 06 03 9f 04 02 20 07 00 4d 15 `

I use this library:
https://github.com/robwlakes/433MHz_Tx_Rx/blob/master/Receiver1.ino

I think the problem is in figure out this params:
```
// Variables for Manchester Receiver Logic:
word    sDelay     = 250;  //Small Delay about 1/4 of bit duration  try like 250 to 500
word    lDelay     = 500;  //Long Delay about 1/2 of bit duration  try like 500 to 1000, 1/4 + 1/2 = 3/4
byte    polarity   = 0;    //0 for lo->hi==1 or 1 for hi->lo==1 for Polarity, sets tempBit at start
byte    tempBit    = 1;    //Reflects the required transition polarity
byte    discards   = 1;    //how many leading "bits" need to be dumped, usually just a zero if anything eg discards=1
byte    discNos    = 0;    //counter of how many discarded
boolean firstZero  = false;//has it processed the first zero yet?  This a "sync" bit.
boolean noErrors   = true; //flags if signal does not follow Manchester conventions
//variables for Header detection
byte    headerBits = 10;   //The number of ones expected to make a valid header
byte    headerHits = 0;    //Counts the number of "1"s to determine a header
//Variables for Byte storage
byte    dataByte   = 0;    //Accumulates the bit information
byte    nosBits    = 0;    //Counts to 8 bits within a dataByte
byte    maxBytes   = 9;    //Set the bytes collected after each header. NB if set too high, any end noise will cause an error
byte    nosBytes   = 0;    //Counter stays within 0 -> maxBytes
//Variables for multiple packets
byte    bank       = 0;    //Points to the array of 0 to 3 banks of results from up to 4 last data downloads 
byte    nosRepeats = 0;    //Number of times the header/data is fetched at least once or up to 4 times
//Banks for multiple packets if required (at least one will be needed)
byte  manchester[4][20];   //Stores 4 banks of manchester pattern decoded on the fly
```

Thanks;

zuckschwerdt, 2023-02-14:
"RXB6" -> Superheterodyne Receiver. We can't really help with those "1-bit" OOK modules. The signal quality is aweful and the range very limited.
Use a CC1101 or similar (e.g. Si4432). Be sure to read the programming guide.

closed 2023-02-14 by zuckschwerdt

garudaonekh, 2023-02-15:
Thanks,
I have used CC1101 with spring antenna (https://github.com/NorthernMan54/rtl_433_ESP) for my weather station. But the range is too bad(5 meters) and successful decoding just 1 packet after 2 hours. I change to use RXB6(same spring antenna) instead and the range is far better.

I have no idea why too.

zuckschwerdt, 2023-02-15:
How did you determine that the Superheterodyne receiver has better range? Did you already receive good packets?
The Superheterodyne receiver would output garbage where the CC1101 knows there is no sensible data and is just silent. It should work for far more than 5m.

garudaonekh, 2023-02-15:
Sorry, it's RXB7 not RXB6.

I have used CC1101 with https://github.com/NorthernMan54/rtl_433_ESP and tested with my WH1080 weather station. I can only successfully decode 1 packet after two hours and if I put the weather station outside the room, it's 0. 

Recently, I use RXB7 and test it along side with your code(RTL_433 using my RTL_SDR), I run for two hours, RXB6 can successfully decode 90% of what RTL_433 decode.

garudaonekh, 2023-02-15:
I see some people also facing same issue like me. At first, I want to try to use SX1278 instead but somehow, it look similar.

https://hackaday.com/2023/01/13/arduino-library-brings-rtl_433-to-the-esp32/#comment-6584152