merbanan/rtl_433 #2381: No Data shown for ecowitt wh51 433mhz device

owingst, 2023-02-15:
I am unable to see any decoded data from the ecowitt wh51 Soil Moisture Sensor. I am using the command line: 
rtl_433 -vvv -R142 and place the device close to the sdr (rtl-sdr). I have several other devices working fine using rtl_433. Any ideas? 
Here is the log file:
[ecowitt.log](https://github.com/merbanan/rtl_433/files/10747533/ecowitt.log)

zuckschwerdt, 2023-02-15:
Try to grab a sample and upload here as zip. https://triq.org/rtl_433/ANALYZE.html

owingst, 2023-02-15:
Here they are....

[ecowittsamples.zip](https://github.com/merbanan/rtl_433/files/10749238/ecowittsamples.zip)

ProfBoc75, 2023-02-15:
@owingst : Hi, are you sure of the frequency ? Should be 915 Mhz for you in Texas, if yes add -f 915M

From the fineoffset device : 
>Test decoding with: rtl_433 -f 433920000  -X "n=soil_sensor,m=FSK_PCM,s=58,l=58,t=5,r=5000,g=4000,preamble=aa2dd4"
>Note: for WH51 at 915MHz: try also "-Y classic" i.e. : rtl_433 -f 915M -Y classic -- see https://github.com/merbanan/rtl_433/issues/2235

owingst, 2023-02-16:
My device is 433MHZ... I chose 433 version (there are 3 I think) assuming rtl_433 would read it.  Here is pic of my device.

[image]

Running with new setting you gave, I get the following with a segmentation fault after showing 2 readings.

`
rtl_433 version -128-NOTFOUND branch master at 202204011242 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 125 out of 153 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-104 108-116 119 121 124-128 131-153 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-15 20:13:15
model     : SwitchDoc Labs FT020T AIO              Device    : 12            Serial Number: 205
Battery Low: 0           Ave Wind Speed: 11        Gust      : 17            Wind Direction: 125       Cum Rain  : 300
Temperature: 1089        Humidity  : 84            Light     : 0             UV Index  : 0             Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-15 20:13:17
model     : soil_sensor  count     : 1             num_rows  : 1             rows      : 
len       : 124          data      : 510d431e127f000027000000bd34204
codes     : {124}510d431e127f000027000000bd34204
./rtl433.sh: line 15:  1539 Segmentation fault      rtl_433 -f 433920000 -X "n=soil_sensor,m=FSK_PCM,s=58,l=58,t=5,r=5000,g=4000,preamble=aa2dd4"
`

owingst, 2023-02-16:
Modified run parameters as follows and it appears to finally get readings for wh51 consistently, albeit the data is a little funky:

` 
rtl_433 -f 433920000 -X "n=soil_sensor,m=FSK_PCM,s=58,l=58,t=5,r=5000,g=4000,preamble=aa2dd4" -R23 -R146 -R147 -C customary -F json | mosquitto_pub -l -t Sensor -h $IP
`
Not sure what the data means. Any ideas on how to fix it  or understand the data?

ProfBoc75, 2023-02-16:
@owingst  : Hi,  Thanks for the confirmation.

The -X option is to get the raw data in hexa byte in order to build a device driver to decode this data. The [fineoffset.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset.c) is concern here and already manages the devices with such preamble aa2dd4.

Looking at the fineoffset.c device line 614, a test is made on byte 0:  if = 0x51 then "Msg familly unknown 0x51", and DECODE_ABORT_EARLY, this is your case unfortunatly.

This means that your device is not decoded by rtl_433 with the last version of rtl_433.

I noticed that you're using an older one from april last year.
Are you able to test with a most recent version and procure more results as you made.
For each , are you able to get the real data from the ecowitt gateway/ application, this is required to decode the hexa raw data and may be update the device driver with that findings.

gdt, 2023-02-16:
Are you building git master from source when testing?  I have used rtl_433 with 3 WH51s on 915 just fine.

owingst, 2023-02-16:
@ProfBoc75/gdt: I am not using the gateway. I am trying to get data from these devices and integrate into my python/IOS apps. I was not aware the gateway played a part in decoding the data ( assumed incorrectly that all decoding was done by rtl_433).

I will try a new version of rtl_433 (w/Switchdoc support). Let me know if this makes sense. Thanks!

zuckschwerdt, 2023-02-16:
@ProfBoc75, like me, didn't remember that we already have WH51 support. The request for original gateway readings was just to analyze the (thought to be) unknown hex data. He did some great work on other sensors using that method.

But @gdt is right, it should just work. Please try a recent rtl_433 version.

owingst, 2023-02-16:
That worked when using new build (git clone https://github.com/merbanan/rtl_433.git) but I am missing switchdoc labs drivers. How do I get latest that includes ecowitt and switchdocs lab?

owingst, 2023-02-16:
When moving to use switchdocs labs build (https://github.com/switchdoclabs/rtl_433) to get both wh51 and switchdoc, I get the following segementation fault:

`
rtl_433 version -128-NOTFOUND branch master at 202212290929 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 4 out of 155 device decoding protocols [ 23 142 146-147 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
mySecondFlags = 0 0
mySecondFlags LB = 0 0
myLight before = 00008eb1 36529
myLight = 00008eb1 36529
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-16 11:33:32
model     : SwitchDoc Labs FT020T AIO              Device    : 12            Serial Number: 205
Battery Low: 0           Ave Wind Speed: 17        Gust      : 20            Wind Direction: 2         Cum Rain  : 318           Temperature: 851
Humidity  : 46           Light     : 36529         UV Index  : 28            Integrity : CRC
./rtl433.sh: line 18:  3300 Segmentation fault      rtl_433 -R23 -R142 -R146 -R147
`
Any ideas?

gdt, 2023-02-16:
It seems obvious that bugs when running switchdoclabs code should be reported in the switchdoclabs issue tracker.   I do not recall them having introduced themselves and the purpose/status of their fork on the list.  If it's just someone working on new features and they aren't quite baked enough for a pull request yet, that sounds good.

merbanan, 2023-02-16:
@gdt They sell some products that uses rtl_433 in the back end.

@owingst if there is a specific sensor you need support for you can follow https://triq.org/rtl_433/ANALYZE.html (again). And we can not support the Switchdock fork but we can probably add the sensors.

ProfBoc75, 2023-02-16:
agree with @gdt , the fork is very old, 4 or 3 years ago and logically only the switchdoclabs drivers are updated ... very difficult to merge both version. I hand over ...

owingst, 2023-02-16:
I opened an issue on switchdoc issue tracker. There latest build has some compile issues. I currently run rtl_433 with the following sensors:

`
SwitchDoc Labs FT020T AIO 
SwitchDoc Labs F016TH Thermo-Hygrometer
DSC Contact
` 
Be great to get a build that has those three plus wh51. Thanks!

merbanan, 2023-02-16:
Follow https://triq.org/rtl_433/ANALYZE.html for each and every sensor and we'll get there eventually.

But this issue can be closed as there support is already there.

closed 2023-02-16 by merbanan

zuckschwerdt, 2023-02-16:
I did already merge the interesting bits from the switchdoc fork. But since all "switchdoc" devices are rebrands we are using the original names.
E.g. SwitchDocLabs FT020T -> Cotech 36-7959 Weatherstation
SwitchDocLabs F016TH -> Ambient Weather F007TH 
SwitchDocLabs SM23 -> Fine Offset / Ecowitt WH51

owingst, 2023-02-16:
Where can I download that from? Main branch?

zuckschwerdt, 2023-02-16:
All of that is included since 2021-10.

owingst, 2023-02-16:
Using latest build from main branch, switchdoc stuff is there but no data from wh51 using the following parameters:

rtl_433 -R23 -R142 -R146 -R147 -C customary -F json | mosquitto_pub -l -t Sensor -h $IP

owingst, 2023-02-16:
downloaded again from git clone https://github.com/merbanan/rtl_433.git. Now only get wh51 data...no idea what I am doing wrong.

owingst, 2023-02-16:
This is what I get with main branch build using:  rtl_433 -R23 -R142 -R146 -R147
`
rtl_433 version nightly branch master at 202302162137 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
[Protocols] Registered 4 out of 238 device decoding protocols [ 23 142 146-147 ]
[SDR] Found 1 device(s)
[SDR] trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-16 14:59:26
model     : Fineoffset-WH51                        ID        : 0d431e
Battery level: 1.222     Battery   : 1800 mV       Moisture  : 0 %           Transmission boost: 0     AD raw    : 40
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-16 14:59:26
model     : Fineoffset-WH51                        ID        : 0d431e
Battery level: 1.222     Battery   : 1800 mV       Moisture  : 0 %           Transmission boost: 0     AD raw    : 40
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-16 15:00:36
model     : Fineoffset-WH51                        ID        : 0d431e
Battery level: 1.222     Battery   : 1800 mV       Moisture  : 0 %           Transmission boost: 0     AD raw    : 40
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-16 15:01:46
model     : Fineoffset-WH51                        ID        : 0d431e
Battery level: 1.222     Battery   : 1800 mV       Moisture  : 0 %           Transmission boost: 0     AD raw    : 40
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-16 15:01:46
model     : Fineoffset-WH51                        ID        : 0d431e
Battery level: 1.222     Battery   : 1800 mV       Moisture  : 0 %           Transmission boost: 0     AD raw    : 40
Integrity : CRC
`

ProfBoc75, 2023-02-16:
@owingst : can you add  -R 20 and -R 153 , to be able to decode the Swichdoc sensors as I found them into your samples :
```
  [20]  Ambient Weather F007TH, TFA 30.3208.02, SwitchDocLabs F016TH temperature sensor
 ...
 [153]  Cotech 36-7959, SwitchDocLabs FT020T wireless weather station with USB
```
Notice that without -R and numbers, it will decode all the supported device protocols.  with -R only it will list all the protocols supported by rtl_433.

owingst, 2023-02-17:
That worked! Thanks for all your help.

owingst, 2023-02-17:
After changing to use Cotech-367959 from SwitchDocLabs using same weather station, I no longer get UV/Light values from rtl_433, although cotech_36_7959.c shows those values. Any ideas on what is going on?

`
time      : 2023-02-17 11:51:26
model     : Cotech-367959 ID        : 205
Battery   : 1            Temperature: 47.2 F       Humidity  : 38 %
Rain      : 31.8 mm      Wind direction: 307       Wind      : 1.6 m/s
Gust      : 2.6 m/s      Integrity : CRC
`

Here is data I got under switchdoc code:
`{'time': '2022-12-13 10:36:36', 'model': 'SwitchDoc Labs FT020T AIO', 'device': 12, 'id': 222, 'batterylow': 0, 'avewindspeed': 4, 'gustwindspeed': 7, 'winddirection': 234, 'cumulativerain': 666, 'temperature': 1100, 'humidity': 91, 'light': 20819, 'uv': 12, 'mic': 'CRC'}
`

zuckschwerdt, 2023-02-17:
The values must have been invalid or out of bounds.
Use `rtl_433 -R 153:vv` to show the raw codes and paste one here, then we can see what the raw uv/lux value was.

owingst, 2023-02-17:
Here is what I got....

`
time      : 2023-02-17 12:53:34
model     : Cotech-367959 ID        : 205
Battery   : 1            Temperature: 49.3 F       Humidity  : 34 %          Rain      : 31.8 mm
Wind direction: 101      Wind      : 2.0 m/s       Gust      : 2.3 m/s       Integrity : CRC
[pulse_slicer_manchester_zerobit] Cotech 36-7959, SwitchDocLabs FT020T wireless weather station with USB
codes     : {127}002999a0282eca027d06fa44380a5340
1

zuckschwerdt, 2023-02-17:
[This BitBench](https://triq.net/bitbench#c=002999a0282eca027d06fa44380a5340&f=h%20ID%3A8d%204b%208h%208h%208h%20h%2012h%20h%2012h%208h%20UV%3A16d%20INDEX%3A8h%20CRC%3A8h%20&a=Preamble&m=014&cw=4) shows the fields. The UV-Index is 0x29 (41) which is not valid. No idea why this might be and if something could be fixed?

owingst, 2023-02-17:
Here is log info from same test but using switchdocslab build.

`time      : 2023-02-17 14:40:14
model     : SwitchDoc Labs FT020T AIO              Device    : 12            Serial Number: 205
Battery Low: 0           Ave Wind Speed: 8         Gust      : 10            Wind Direction: 302       Cum Rain  : 318           Temperature: 939          Humidity  : 28
Light     : 97677        UV Index  : 25            Integrity : CRC
pulse_demod_manchester_zerobit(): SwitchDoc Labs Weather FT020T Sensors 
bitbuffer:: Number of rows: 1 
[00] {127} 00 29 99 a8 10 14 5c 02 7d 07 56 38 fb 1a 33 ca 
`

zuckschwerdt, 2023-02-17:
Yeah, but we figured a valid UV Index range of 0-15. So 25 seems like bad data and is thus ignored. There is nothing *wrong*, it's an explicit filtering of invalid data.

@obones can you look at this, involving #2305

obones, 2023-02-17:
Well, the light value is completely wrong as well.
I went looking for the FT020T model and I found this page:
https://shop.switchdoc.com/products/wireless-weatherrack2

It says rtl_433 supports it but most importantly, there is a table that says UV index is between 1 and 15 :

[image]

So, to me, it is perfectly expected that an out of bound UV index (and light intensity) be ignored. Note that I also found this page:
https://uctechnologyltd.com/product/ft020t-weather-station-sensor-no-uv/

And this one is a declination of the FT020T specifically with no UV sensor so maybe you have this variant.

owingst, 2023-02-17:
My weather station has uv/light. I don't see any check in the code for a valid uv range in switchdoclabs_FT020T.c. So I guess I will have to open an issue on those folks....

zuckschwerdt, 2023-02-17:
To be clear: the values you captured as code were 7.173 kLux and UV-Index 29, so pretty shaded with chance of severe burn in mere minutes.
Or the values you posted 97.677 kLux UV-Index 25, very bright day but still UV death rays.
Those values don't mean anything sensible, the sensor could be broken perhaps.

The switchdoclabs fork is old and unmaintained I guess, otherwise both decoders are basically the same (though different on sanity checks).

owingst, 2023-02-17:
I really appreciate the help. Thanks!