merbanan/rtl_433 #2382: Request more detail about the RainPoint Soil Sensor

garudaonekh, 2023-02-16:
Hi,
I am trying to decode this using normal superheterodyne receiver. I can read but get the incorrect binary data.
I see it's mentioned in the doc of this devices:
**1. A Transmission contains three packets with Manchester coded data, note that the pause is a constant pulse, strangely.**
 **.short_width = 500,
 .long_width  = 500,**

https://github.com/merbanan/rtl_433/blob/master/src/devices/rainpoint.c

Can you tell me more about this config.?

Thanks;

Thanks;

zuckschwerdt, 2023-02-16:
You already asked this in #2377
We can't help you with a superheterodyne (1-bit) receiver, it always ends in frustration.

Closing as this is not something to change or fix in rtl_433, feel free to discuss further though.

closed 2023-02-16 by zuckschwerdt

merbanan, 2023-02-16:
@zuckschwerdt do we have any sample signals for this decoder? I cant seem to find any in the tests repo.

zuckschwerdt, 2023-02-16:
Yes, I have them. No surprises but the level as packet gap.
Must have been a zip upload. Not sure, dating from 2021-08-31. I really need to get around to update the tests repo some day.
[image]

garudaonekh, 2023-02-20:
Thanks for your input. Much appreciated.
I am trying to figure out and learning in the process.

garudaonekh, 2023-02-23:
> Yes, I have them. No surprises but the level as packet gap. Must have been a zip upload. Not sure, dating from 2021-08-31. I really need to get around to update the tests repo some day. [image]
> 
> _edit:_ that's three packets and the broken "gaps" between. A 1-bit receiver will go mad there.

After a few days of struggling, I got it working on RXB7. Thanks

By the way, what do you mean by "1-bit receiver will go mad..."?  You mean it'll interrupt other transmitter due to its long transmission if I use multiple transmitter?

zuckschwerdt, 2023-02-23:
The long weird pause with a carrier signal will likely cause 1-bit receivers to hallucinate data.

garudaonekh, 2023-02-28:
No idea about this. But currently, it can decode wh1080, this soil sensor and rc-switches.